sinara-hw/Booster #91: Channel doesn't work unless it's lid is screwed on

bnichol, 2018-09-10:
One of the channels doesn't turn on when it's individual lid is removed. 
Nothing appeared to be loose on the main PCB board or supplementary control panel. However the channel did not appear to work until it's lid had been screwed back on.

gkasprow, 2018-09-10:
please place tiny piece of Kapton tape in the cavities where connector pins are. If that helps it means that the pins were not trimmed enough.

hartytp, 2018-09-10:
> please place tiny piece of Kapton tape in the cavities where connector pins are. If that helps it means that the pins were not trimmed enough

I don't understand.

The symptom here is that this channel is not detected at all by the firmware when the lid is off. When the lid is on it works fine.