sinara-hw/Booster #92: Imperfections on the Plating

bnichol, 2018-09-10:
cloudy white marks started appearing on the plating as shown in the picture
[image]

gkasprow, 2018-09-10:
It looks like the plating process was faulty. This should be possible to fix. Let's wait and observe what happens. Later on we will replace the modules and ask the company who did plating for repair

hartytp, 2018-09-10:
Sounds good. Shouldn't this have been picked up by QC at TechnoSystems?

gkasprow, 2018-09-10:
It looks like it was not immediate effect. Probably they didn't wash it before plating, so electrochemical process continues and such marks will grow and cover whole surface after a few days.

hartytp, 2018-09-10:
Okay, I didn't know that cold happen. @bnichol if you get the chance, could you look at this again in a week or two to see if the problem is spreading?

hartytp, 2018-09-11:
I'll close this issue now, as there is no immediate action to be taken.

@gkasprow will you report this to TechnoSystem, or shall I?

closed 2018-09-11 by hartytp