sinara-hw/Booster #93: switch glitches

hartytp, 2018-09-10:
We observe glitches [image] when the interlock trips or is reset. These glitches are about 45Vpp (the measurement in the picture had a 30dB power attenuator), so quite big.

I don't think they will damage our AOMs, which have narrow-band impedance matching, but they're bigger than I would expect. The analogous glitches from an external MCL switch are about an order of magnitude smaller. Also, I thought that switching to a reflective switch was supposed to get rid of these switches.

@gkasprow do you think that there could be a cause of these glitches other than the RF switch itself? e.g. could they be due to coupling between the TTL and the RF line, or some resonance in the decoupling network?

gkasprow, 2018-09-10:
It looks like step response of some resonant circuit.
One can add small, let's say 100pF capacitors to the RF switch inputs to GND and see if that helps.

hartytp, 2018-09-10:
@gkasprow which switch is used now? The schematic PDF I have still shows HMC284, but I thought we changed to a reflective switch, so maybe it's out of date?

hartytp, 2018-09-10:
Also, not sure if this is just a schematic rendering issue, but I don't see a pin for the EP on the switch. However, the data sheet says that it *must* be grounded.

hartytp, 2018-09-10:
oops, never mind, I must have been looking at an old version of the schematic. It's now [HMC5544A](http://www.analog.com/media/en/technical-documentation/data-sheets/hmc544ae.pdf) which is reflective

gkasprow, 2018-09-10:
The HHMC284 has pin 9 grounded. It is hidden pin that has GND value assigned by default.
New version of schematic uses HMC544A. Releases / Booster v1.2, file Booster_PA_preamp.PDF

gkasprow, 2018-09-10:
Yes, I mean pull-up/down resistors. Potentiometer + resistor would be probably better. I'd start with 1k trimpot + 100k resistor

gkasprow, 2018-09-10:
where do you see these 1k resistors? I cannot find them in the [datasheet](http://www.analog.com/media/en/technical-documentation/data-sheets/hmc544ae.pdf)

gkasprow, 2018-09-10:
Oh, I looked at datasheet of HMC544E which don't have such resistors..

hartytp, 2018-09-10:
> Glitch can also be caused by charge injection. It can be mitigated by simultaneous control of A and B inputs with opposite edges. In our case B is slightly delayed by negator Tpd and can cause such effect. 

True, but this is exactly how they do it in the reference schematic shown on the data sheet, so I'm surprised the effect is that large. Also, I would have thought that the propagation delay of that inverter is only a few ns, whereas this glitch is over of hundred ns so the timescales don't seem right for that.

hartytp, 2018-09-10:
> where do you see these 1k resistors? I cannot find them in the datasheet

Table on the final page of the data sheet.

gkasprow, 2018-09-10:
the resistors are on the board layout but no on the schematic...

gkasprow, 2018-09-10:
We use series resistors, but they are 100Ohm ones.

gkasprow, 2018-09-10:
One can replace them with 10k and see if that helps. Input current is 1uA so 10k won't affect the threshold but will slow down the edges.

hartytp, 2018-09-10:
I'm tempted to buy an evaluation board and have a play with it.

gkasprow, 2018-09-11:
@hartytp true, the negatr propagation is a few ns, but to compensate the charge injection, both pulses must be in the same moment. If they are shifted by a few ns, a few ns long pulse may cause oscillations.
My opinion is that offset difference causes that effect, not charge injection.

hartytp, 2018-09-11:
Okay, I'll order the evaluation board now and see if I can reproduce this. Then we can consider what the correct solution is (presumably either increase the resistor values or add a capacitor between the two control pins or both).

Jochenwolf3, 2018-09-11:
Another test: resetting the interlock while the channel input is still high enough to trip the interlock.
The result is that the interlock takes ~700 us to re-trip, i.e. quite long. However, as the interlock button only triggers on release, I could not observe any 'bouncing' of the interlock, i.e. one press & release always only resulted in one reset.
As per @hartytp suggestion, can you please tweak C10 and R1 on the supplemental control board to get this time down to approximately <10 us?

dhslichter, 2018-09-11:
There are a number of switch chips that have integrated CMOS electronics onboard such that the chip only requires a single logic line instead of two operated in tandem.  That might fix it.  Why not use the same switch as on Urukul?  Is this switch after the PA stage, and thus needs to be high-power-handling?

dhslichter, 2018-09-11:
For example: http://www.analog.com/en/products/adg901.html

Has a 17 dBm max input power rating, ~15 dBm P1dB, 5 ns off time (!!), internal 50 ohm terminations (absorptive).  
Costs $3 each, same as HMC544A.

dhslichter, 2018-09-11:
Here for $4/per, can take a 5V rail:
http://www.analog.com/en/products/hmc1055.html
http://www.analog.com/en/products/hmc349ams8g.html

Slightly pricier (~$6-7) but better power handling:
http://www.analog.com/en/products/hmc8038.html

hartytp, 2018-09-12:
> Why not use the same switch as on Urukul? Is this switch after the PA stage, and thus needs to be high-power-handling?

No, it's the first element in the signal path (it has to be since one aspect of its functionality is to protect Booster from an excessive input level).

I were designing this from scratch, I probably would use the same switch as Urukul. However, apart from these glitches the switch that's in there works really well and is very low cost. So, given that we now have a few of these units produced, I'd prefer to have a quick go at fixing the glitches by changing the resistors and maybe adding a capacitor on the logic inputs. If that doesn't work we can consider a different switch.

gkasprow, 2018-09-12:
Did someone checked if the switch in Urukul produces glitches as well ?

hartytp, 2018-09-12:
Yes. @jordens took some nice data on that when he originally tested Urukul. See here https://github.com/sinara-hw/Urukul/issues/3#issuecomment-396951212

gkasprow, 2018-09-12:
The difference is that in Booster, the switch transient is amplified by further stages. In Urukul it is not, so it can be even worse and we don't see it.

hartytp, 2018-09-12:
@gkasprow The signal is amplified just as much as the transients. So, the important thing isn't the overall size of the glitches at the output, but the ratio of the glitch to the maximum signal. In Booster, the glitch is much larger than the maximum signal level, in Urukul the glitch cannot easily be seen at all compared with the signal.

So, I think we can safely say that the glitches are much worse on Booster.

gkasprow, 2018-09-12:
@hartytp you are right. The transient has 45Vpp on the Booster output. So at the input it is order of 0.4Vpp. Such value would be clearly seen at the Urukul output.

dhslichter, 2018-09-12:
@hartytp I'm fine with the quick fix idea if it can be made to work, but since there will be future revisions of these boards anyway because of other issues, it seems like it would be worth keeping the new switch idea in the back pocket in case the fixes don't work as much as one would like.  Certainly this kind of transient is unacceptable in the final product...

hartytp, 2018-09-12:
@dhslichter yes, it needs to be fixed. If appropriate RCs don't fix the issue we will use the same switch as Urukul.

hartytp, 2018-09-13:
@gkasprow we could also replace the two inverters with two NAND gates (one with the second pin pulled high, one with it pulled low). That way the propagation delay for the two signals should be very well matched...

gkasprow, 2018-09-13:
True. Providing that time difference causes the issue. Let's make sure that
it is not caused by input offest. Did U measure it?

hartytp, 2018-09-13:
I'm setting up to measure this now. The epoxy is setting on an SN74AHC04 that I added to the switch eval board.

What do you mean by input offset?

gkasprow, 2018-09-13:
DC voltage between RF inputs of the mux caused by internal biasing

hartytp, 2018-09-13:
I see. Well if that's the problem then we'll need a new switch. I'll let you know how I get on with the eval board.

gkasprow, 2018-09-13:
Or we can add external pullup or pulldown resistor plus trimpot to
compensate it

hartytp, 2018-09-13:
We could do, but that adds more production time complexity, which I'd prefer to avoid if possible.

hartytp, 2018-09-14:
Okay, good news: this isn't a fundamental property of the switch.
- took a switch eval board and glued on a SN74AHC04 (hex inverter)
- Input of first inverter driven by a 3.3pk-pk square wave at 1kHz
- output of first inverter connects to input of second inverter as well as the switch "A" input
- output of second inverter connects to switch B input
- apply RF at something like -11dBm 200MHz to switch RF2
- terminate switch RF1
- common port goes to booster
- booster output goes to a fast scope (via a 30dB attenuator, but I normalize that out in post processing)

Output is:
[image]

so no glitch

hartytp, 2018-09-14:
Well, this isn't totally fair because in Booster the switch comes after 6dB of attenuation. But, still, the eval board seems better.

This is what it looks like when I swap the input and output of the switch:
[image]

hartytp, 2018-09-14:
This is the switch with RFC connected as an input (same as last plot) but no RF applied.
- the normalisation here is a bit odd, but attempting to mimic [this](https://github.com/sinara-hw/Booster/issues/93#issue-358751736) measurement. I do not correct for the 30dB attenuator, but I do scale by 8dB to correct for the attenuator inside booster.
[image]

hartytp, 2018-09-14:
So, that's actually pretty close to what we measured before.

With the switch connected so that RF1 is the input, I do not see any glitch without RF applied.

So, that seems to be the answer: we have the switch connected the wrong way round!

hartytp, 2018-09-14:
Nope, sorry for the noise, I do see a glitch with RF2 used as an input, although it's half the size of the glitch with the common pin used as an input.

Here is the glitch with RF2 used as an input, with the same normalisation as the plot above.
[image]

hartytp, 2018-09-14:
So, that corresponds to something like 8Vpp. However, that is effectively doubled due to the attenuation inside booster (the gain is higher from the switche's perspective)...

hartytp, 2018-09-14:
@gkasprow unless you have a better idea, I'll order an eval board for the HMC349ALP4CE used in Urukul and repeat this measurement.

gkasprow, 2018-09-15:
Interesting thing is that we see such behaviour with two switch types. So
this could be related with inverter delay.I would try to increase the
resistor values or add capacitors at the switch control inputs.

hartytp, 2018-09-15:
> I would try to increase the resistor values 

I was using the eval board, which has 1k resistors fitted (see table at bottom of data sheet). So, it's already an order of magnitude larger than the resistors on Booster, but doesn't help.

> or add capacitors at the switch control inputs.

Between the control inputs? Or between the control inputs and ground?

We could also replace the inverters with NAND gates to give ensure that both control inputs switch at the same time.

However, rather than experimenting, it might be better to as HMC.

gkasprow, 2018-09-15:
I mean between inputs and GND. Then timing of the inverter won't matter.

hartytp, 2018-09-16:
Okay I'll try that this week. What capacitance do you want me to try?

gkasprow, 2018-09-16:
100p

Jochenwolf3, 2018-09-21:
Further to @hartytp's measurement using inverters, I have tried the same with the following filters attached to the input of the switch:
1) Inverters plus an RR filter: 1kOhm - 1kOhm
2) Inverters plus an RCR filter, 1kOhm - 100 pF - 1kOhm
3) Inverters plus an RCR filter, 1kOhm - 560 pF - 1kOhm

With 1 of those R's being the 1kOhm resistor that is already installed on the evaluation board.

4) Furthermore, I have tried using two XOR gates instead of inverters to minimize the timing difference; this time with no filtering, only the already installed 1kOhm resistor.
In all graphs, blue is the TTL signal going into the XOR or inverters to trigger the switch. Yellow and green are the two outputs of the switch. RF input is -20 dBm at 200 MHz.

Inverters plus an RR filter: 1kOhm - 1kOhm:
[image]

Inverters plus an RCR filter, 1kOhm - 100 pF - 1kOhm:
[image]

Inverters plus an RCR filter, 1kOhm - 560 pF - 1kOhm:
[image]

XOR gates, no filtering:
[image]

Conclusions: 
- Type of gate logic (two chained inverters vs two parallel XOR gates) made no difference
- Using an RCR filter greatly decreases the glitch: from a 300 mV spike (0 pF), over 150 mV spike (100 pF), to 50 mV (560 pF). It also both delays the response from the switch and lengthens the spike duration. 
Therefore, using a 560 pF capacitor seems like a good size to use. Next thing to try: sending this smaller glitch through booster to check if booster's filtering has any effect on this glitch.

Jochenwolf3, 2018-09-21:
As the RCR filter with 560 pF decreasesd the size of the glitch, I have sent it through booster. I am using -20 dBm at 200 Mhz at the input of the switch, and a 30 dB attenuator on my oscilloscope.

Switching RF OFF:
[image]

Switching RF ON:
[image]

As you can see, the glitch is gone when using a 560 pF capacitor on the switch inputs!

hartytp, 2018-09-21:
Thanks @Jochenwolf3 some comments about that:
- Swapping the inverters for the XOR ensures that both control pins switch at the same time. This makes no difference to the glitches, suggesting that the delay between switching is not the cause of these glitches
- Adding the filtering doesn't seem to change the glitch energy, but just slows it down. 
- With a 500pF + 1k filter we are able to make the glitches slow enough that they are filtered out by Booster's high-pass response, ensuring that no glitch reaches the output
- With 500pF, the time from the interlock tripping to the switch tripping is 400us. That's a bit on the slow side but we can live with it.

Unless anyone objects (@cjbe) I suggest we stick with the switch we have and add an RC filter on the supplemental control board (the cheap 2-layer FR4 PCB that connects the RF PCB to the D-type).
- This switch is about $3 cheaper than the switch on Urukul. That's not a huge amount, but it all adds up. 
- It also saves us from having to do rework on the RF PCB

hartytp, 2018-09-21:
NB this was running the inverters from 5V logic. We will need to double check this with the 3V3 logic used in Booster

closed 2018-09-21 by hartytp