sinara-hw/Booster #96: Internal power meter accuracy testing

Jochenwolf3, 2018-09-11:
[image]
To check the accuracy of the internal power meter, I am comparing the readings of the internal power meter to an external one across different frequencies, at a set input power of -10 dBm. Above is the result of a 50 MHz to 750 MHz sweep. 
The peak to peak difference in readings is 0.8 dB for the range 50 - 500 MHz.

Jochenwolf3, 2018-09-11:
[image]
Here is the same but with a sweep of the power at a fixed frequency of 200 MHz. 
The peak to peak difference in readings is ~0.25 dB above 20 dBm output power.

hartytp, 2018-09-11:
Thanks @Jochenwolf3!

Some comments:
- the peak-peak deviation in the frequency measurement is approx 1dB. That seems consistent with the accuracy of the [directional coupler](https://ww2.minicircuits.com/pdfs/SYDC-20-62HP+.pdf)
- it's interesting to note that the accuracy of the RMS output versus frequency is something like twice as good as for the interlock measurement that @Jochenwolf3 posted yesterday. That is another reason we might prefer to use the external comparator in the next hw revision.
- the accuracy over power is really quite good, which shows that the 2-point calibration implemented by @wizath works nicely.

Over all, I'm very happy with this performance -- well done all!

Final measurement: check the deviation versus temperature...time to get out the hairdryer

Jochenwolf3, 2018-09-12:
Here is the channel-channel diviation of the internal power meter:
[image]

[image]
The frequency dependence is very similar, so there's just a 0.75 dB offset between the two channel's internal power meters.

Jochenwolf3, 2018-09-12:
see https://github.com/sinara-hw/Booster/issues/73#issuecomment-420682798 for temperature coefficient measurement. tl;dr the internal power detector works really well!

closed 2018-09-12 by Jochenwolf3