sinara-hw/Booster #98: Up to date firmware

wizath, 2018-09-11:
Due to repo cleanup, I'll post most recent firmware here for a moment.

[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/2371280/eth-scpi.dfu.zip)
- Updated HW Int calibration procedure to modify values in memory and EEPROM

hartytp, 2018-09-12:
Do I need to re-run the interlock calibration after flashing this firmware?

cjbe, 2018-12-06:
@wizath can you please post the current firmware source and build instructions? The firmware bugs are currently blocking us making any progress.

wizath, 2019-03-12:
[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/2958188/eth-scpi.dfu.zip)

- software interlock for v1.3
- fixed SCPI int:cle <chid>

wizath, 2019-03-19:
[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/2982877/eth-scpi.dfu.zip)

- Changed ADS7924 measurement interval to be a bit longer
- Added clearing of one-time OVC error counters that occurs during power-up

wizath, 2019-03-19:
[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/2983120/eth-scpi.dfu.zip)

- unified interlock set&get commands between VCP console and SCPI interface

wizath, 2019-03-19:
[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/2985101/eth-scpi.dfu.zip)

- added `intparams <ch> <a> <b>` command for setting interlock parameters from VCP
- watchdog enabled with more feeding inside interlock task

wizath, 2019-03-21:
[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/2991987/eth-scpi.dfu.zip)
- fixed `intparams` eeprom conversion vefore saving

wizath, 2019-03-21:
[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/2992857/eth-scpi.dfu.zip)

- disabled SW OVC procedure
- fixed `int:stat?` command

wizath, 2019-03-22:
[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/2996395/eth-scpi.dfu.zip)

- added automatic channel enable post power-up

wizath, 2019-03-22:
[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/2996848/eth-scpi.dfu.zip)

- fixed channel auto enable :cry: 
- fixed channel output interlock calibration
- fixed channel input interlock calibration

wizath, 2019-03-25:
[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/3003488/eth-scpi.dfu.zip)

- fixed dhcp timings

wizath, 2019-03-29:
[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/3023289/eth-scpi.dfu.zip)

- added support for multiple TCP connections (4)
- fixed MAC address loading
- added default MAC address loaded from on-board EEPROM

cjbe, 2019-04-09:
@wizath is [this](https://github.com/sinara-hw/Booster/issues/98#issuecomment-478037725) the latest firmware? Or, is there a more recent one with additional fixes?

cjbe, 2019-04-09:
e.g. interlock calibration fixes?

wizath, 2019-04-09:
Oh, forgot to upload interlock calibration fixes, I'll do it tomorrow. I'll also re-enable auto enable after power up since I commented it out by a mistake. But you can test multiple connection handling straight away.

hartytp, 2019-04-09:
Thanks. Was the unit that Creotech shipped to us calibrated after fixing the rounding issues? i.e. should I expect the interlock inaccuracies we measured before to be gone now?

wizath, 2019-04-09:
That was after fixing floats I believe. You should see it in status table - HWIS HWIO values

hartytp, 2019-04-09:
Thanks. I'll check that and finish testing the Creotech unit once you post the new fw.

hartytp, 2019-04-10:
@wizath ready to test once you post new fw.

hartytp, 2019-04-10:
ping @wizath can I have the latest fw please?

wizath, 2019-04-10:
[eth-scpi.elf.zip](https://github.com/sinara-hw/Booster/files/3063418/eth-scpi.elf.zip)

- fixed floats in most of the VCP commands
- reenabled automatic channel power-up

wizath, 2019-04-10:
There are still non-float values in power readings coefficients but in order to convert them to floats I need to change EEPROM addressing which require some more testing (shortly I need any working device).

hartytp, 2019-04-10:
@wizath @gkasprow what's the plan for getting @wizath a working Booster to finish firmware development on? Do we need to ship you an older unit, or can TechnoSystem/Creotech supply one? I really want to finish this project so let's make sure that @wizath's progress isn't blocked by lack of hardware!

hartytp, 2019-04-10:
@wizath did you mean to post a `.dfu` instead of an elf?

wizath, 2019-04-10:
[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/3063536/eth-scpi.dfu.zip)
Yep, sorry

hartytp, 2019-04-10:
@wizath 

```
PGOOD: 1
FAN SPEED: 30 %
AVG TEMP: 0.00 CURRENT: 0.00
CHANNELS INFO
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	1	1	1	1	1	
HWID		E8:D9	02:F8	02:9A	8E:8B	02:4C	AE:00	DE:A7	3A:FB	
TXPWR [V]	0.01	0.01	0.01	0.03	0.02	0.02	0.01	0.01	
RFLPWR [V]	0.22	0.01	0.01	0.08	0.01	0.02	0.00	0.11	
TXPWR [dBm]	-4.86	-5.60	-4.96	-5.70	-5.09	-5.01	-7.07	-6.14	
RFLPWR [dBm]	-8.89	-3.65	-2.18	-5.24	-2.24	-4.99	-4.28	-5.88	
I30V [A]	0.050	0.049	0.050	0.050	0.050	0.047	0.053	0.050	
I6V0 [A]	0.222	0.229	0.252	0.226	0.213	0.230	0.230	0.205	
5V0MP [V]	4.976	4.974	4.989	4.988	4.973	4.988	4.977	4.980	
ON		1	1	1	1	1	1	1	1	
SON		1	1	1	1	1	1	1	1	
IINT		0	0	0	0	0	0	0	0	
OINT		0	0	0	0	0	0	0	0	
SINT		0	0	0	0	0	0	0	0	
OVC		0	0	0	0	0	0	0	0	
ADC1		20	19	19	49	34	38	22	18	
ADC2		368	13	12	132	16	34	7	173	
DAC1		2386	2386	2386	2386	2386	2386	2386	2386	
DAC2		3572	3640	3577	3548	3686	3636	3766	3730	
SCALE1		83	83	85	83	86	86	84	85	
OFFSET1		423	484	442	521	472	470	616	540	
BIASCAL		1249	1399	1363	1573	1471	1177	1125	1419	
HWIS		84.00	83.42	84.08	82.75	85.50	84.08	84.33	85.92	
HWIO		548.00	637.42	550.08	569.75	608.50	609.08	730.33	637.92	
LTEMP		28.50	28.50	29.00	28.00	29.00	30.00	30.00	29.00	
RTEMP		28.50	28.25	28.00	29.00	29.25	29.00	30.25	31.50	
==============================================================================
```

Confirmed, `HWIS` and `WHIO` are now floats.

FYI, look at the bias current accuracy! Lovely...

hartytp, 2019-04-13:
Closing as not really an issue...

closed 2019-04-13 by hartytp