sinara-hw/Booster #101: temperature reading error

hartytp, 2018-09-12:
We heated channel 2 of Booster up with a hot air gun while monitoring the temperature over ethernet via `MEAS:TEMP? 2` command. The temperature was okay until about 40C. Then it suddenly jumped to 150C.

After that, we looked on the VCP. All channels report a temperature of 0.0 for LTEMP and 1.0 for RTEMP. So there is some issue with the I2C bus I guess...

hartytp, 2018-09-12:
Power cycling fixed this issue. Channel 2 now reports 34/36C for L and R TEMP.

hartytp, 2018-09-12:
We just reproduced this a second time. Heat the module up with a hot air gun. When it hits about 40C (as reported via ethernet) the temperature reading goes to 1.00C.

Any ideas?

Can you try reproducing this at WUT?

wizath, 2018-09-12:
I'll check that tomorrow. Another question - I'm curious if channel went down with error (red) led, since that protection was not tested.

hartytp, 2018-09-12:
> I'll check that tomorrow. Another question - I'm curious if channel went down with error (red) led, since that protection was not tested.

Thank you for the reminder, I forgot to comment about that. @jochenwolf3  correct me if I'm wrong, but IIRC all LEDs were still green. However the output power dropped by 30dB indicating that the interlock had tripped. So, I do not think that the LEDs were accurately representing the device state.

I can post the UART output tomorrow if that helps?

wizath, 2018-09-12:
Yes, you can post it - if channel was disabled that's a good sign that protection for odd temperature readings worked.

hartytp, 2018-09-12:
Okay, I'll try to reproduce this and post the VCP log in the morning.

Jochenwolf3, 2018-09-13:
@wizath have a look at this:

```
PGOOD: 1
FAN SPEED: 0 %
AVG TEMP: 34.01 CURRENT: 37.00
CHANNELS INFO
==============================================================================
                #0      #1      #2      #3      #4      #5      #6      #7
DETECTED        1       1       1       1       1       1       1       1
TXPWR [V]       1.02    0.04    0.04    0.04    0.05    0.04    0.06    0.04
RFLPWR [V]      0.33    0.04    0.04    0.05    0.05    0.04    0.04    0.06
TXPWR [dB]      27.02   0.02    -1.14   -1.06   -2.02   -1.02   -1.87   -0.56
RFLPWR [dB]     30.94   0.00    -0.04   -0.37   -0.64   -0.67   0.78    -2.20
I30V [A]        0.208   0.167   0.153   0.157   0.145   0.142   0.163   0.165
I6V0 [A]        0.042   0.033   0.031   0.031   0.029   0.028   0.033   0.033
IN8V0 [mA]      0.886   0.709   0.652   0.668   0.618   0.600   0.691   0.699
ON              1       1       1       1       1       1       1       1
SON             1       1       1       1       1       1       1       1
IINT            0       0       0       0       0       0       0       0
OINT            0       0       0       0       0       0       0       0
SINT            0       0       0       0       0       0       0       0
SINTE           1       1       1       1       1       1       1       1
SINTV           38.0    38.0    38.0    38.0    38.0    38.0    38.0    38.0
OVC             0       0       0       0       0       0       0       0
ADC1            1664    66      69      70      77      70      105     72
ADC2            533     68      73      78      75      71      71      99
DAC1            92      94      83      93      90      90      82      95
DAC2            250     244     240     261     166     254     253     166
SCALE1          52      57      56      54      53      54      53      54
OFFSET1         258     65      134     127     183     125     204     102
BIASCAL         3870    2130    285     2475    540     1785    3570    1395
HWIS            9.03    8.42    8.63    10.30   10.52   9.64    9.26    8.65
HWIO            -13.55  -9.94   -10.92  -20.68  -21.56  -16.90  -14.83  -11.25
LTEMP           32.25   31.00   35.50   35.00   34.25   34.25   32.00   32.00
RTEMP           33.00   33.50   36.50   37.00   33.00   35.00   31.00   35.00
==============================================================================

PGOOD: 1
FAN SPEED: 0 %
AVG TEMP: 34.02 CURRENT: 37.50
CHANNELS INFO
==============================================================================
                #0      #1      #2      #3      #4      #5      #6      #7
DETECTED        1       1       1       1       1       1       1       1
TXPWR [V]       1.02    0.04    0.04    0.04    0.05    0.04    0.06    0.04
RFLPWR [V]      0.33    0.04    0.04    0.05    0.05    0.04    0.04    0.06
TXPWR [dB]      27.04   0.02    -1.18   -1.06   -2.00   -1.02   -1.87   -0.56
RFLPWR [dB]     30.96   0.00    -0.08   -0.37   -0.62   -0.67   0.78    -2.20
I30V [A]        0.208   0.167   0.157   0.157   0.145   0.142   0.162   0.164
I6V0 [A]        0.042   0.033   0.031   0.032   0.029   0.028   0.032   0.033
IN8V0 [mA]      0.886   0.712   0.668   0.670   0.616   0.603   0.691   0.699
ON              1       1       1       1       1       1       1       1
SON             1       1       1       1       1       1       1       1
IINT            0       0       0       0       0       0       0       0
OINT            0       0       0       0       0       0       0       0
SINT            0       0       0       0       0       0       0       0
SINTE           1       1       1       1       1       1       1       1
SINTV           38.0    38.0    38.0    38.0    38.0    38.0    38.0    38.0
OVC             0       0       0       0       0       0       0       0
ADC1            1664    66      68      70      77      70      105     72
ADC2            532     68      72      78      74      70      71      98
DAC1            92      94      83      93      90      90      82      95
DAC2            250     244     240     261     166     254     253     166
SCALE1          52      57      56      54      53      54      53      54
OFFSET1         258     65      134     127     183     125     204     102
BIASCAL         3870    2130    285     2475    540     1785    3570    1395
HWIS            9.03    8.42    8.63    10.30   10.52   9.64    9.26    8.65
HWIO            -13.55  -9.94   -10.92  -20.68  -21.56  -16.90  -14.83  -11.25
LTEMP           32.25   33.00   35.00   35.00   34.50   34.50   32.00   33.50
[... 84 lines cut ...]
I30V [A]        0.020   0.020   0.020   0.020   0.020   0.020   0.020   0.020
I6V0 [A]        0.004   0.004   0.004   0.004   0.004   0.004   0.004   0.004
IN8V0 [mA]      0.083   0.083   0.083   0.083   0.083   0.083   0.083   0.083
ON              1       1       1       1       1       1       1       1
SON             1       0       0       1       1       1       1       1
IINT            0       0       0       0       0       0       0       0
OINT            0       0       0       0       0       0       0       0
SINT            0       1       1       0       0       0       0       0
SINTE           1       1       1       1       1       1       1       1
SINTV           38.0    38.0    38.0    38.0    38.0    38.0    38.0    38.0
OVC             0       0       0       0       0       0       0       0
[... cut ...]
OFFSET1         258     65      134     127     183     125     204     102
BIASCAL         3870    2130    285     2475    540     1785    3570    1395
HWIS            9.03    8.42    8.63    10.30   10.52   9.64    9.26    8.65
HWIO            -13.55  -9.94   -10.92  -20.68  -21.56  -16.90  -14.83  -11.25
LTEMP           0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00
RTEMP           1.00    1.00    1.00    1.00    1.00    1.00    1.00    1.00
==============================================================================
```

Jochenwolf3, 2018-09-13:
We are heating channel 2 with a hot air gun. The temperature reading increases normally until about 38C. Then it briefly reads 150C. Next, all channels read LTEMP=0,RTEMP=1. This seems to be very reproducible (although we've only tried it on one channel)

wizath, 2018-09-13:
Ok, time to get a hot air gun

Jochenwolf3, 2018-09-13:
```
PGOOD: 1
FAN SPEED: 29 %
AVG TEMP: 39.16 CURRENT: 36.50
CHANNELS INFO
==============================================================================
                #0      #1      #2      #3      #4      #5      #6      #7
DETECTED        1       1       1       1       1       1       1       1
TXPWR [V]       1.02    0.04    0.04    0.04    1.90    0.04    0.06    0.04
RFLPWR [V]      0.33    0.04    0.04    0.05    1.90    0.04    0.04    0.06
TXPWR [dB]      27.06   0.02    -1.14   -1.07   55.49   -1.02   -1.94   -0.56
RFLPWR [dB]     30.98   0.00    -0.04   -0.39   60.32   -0.67   0.69    -2.20
I30V [A]        0.203   0.165   0.155   0.154   0.151   0.149   0.164   0.164
I6V0 [A]        0.041   0.031   0.031   0.031   0.030   0.030   0.033   0.033
IN8V0 [mA]      0.862   0.662   0.660   0.655   0.642   0.631   0.699   0.699
ON              1       1       1       1       1       1       1       1
SON             1       1       1       1       0       1       1       1
IINT            0       0       0       0       0       0       0       0
OINT            0       0       0       0       0       0       0       0
SINT            0       0       0       0       1       0       0       0
SINTE           1       1       1       1       1       1       1       1
SINTV           38.0    38.0    38.0    38.0    38.0    38.0    38.0    38.0
OVC             0       0       0       0       0       0       0       0
ADC1            1664    66      69      69      3099    69      101     72
ADC2            534     68      73      79      3099    70      70      99
DAC1            92      94      83      93      90      90      82      95
DAC2            250     244     240     261     166     254     253     166
SCALE1          52      57      56      54      53      54      53      54
OFFSET1         258     65      134     127     183     125     204     102
BIASCAL         3870    2130    285     2475    540     1785    3570    1395
HWIS            9.03    8.42    8.63    10.30   10.52   9.64    9.26    8.65
HWIO            -13.55  -9.94   -10.92  -20.68  -21.56  -16.90  -14.83  -11.25
LTEMP           31.25   32.25   34.25   34.25   34.00   32.00   31.25   32.00
RTEMP           32.00   32.00   36.50   35.00   33.00   34.50   30.00   34.50
==============================================================================

PGOOD: 1
FAN SPEED: 29 %
AVG TEMP: 39.14 CURRENT: 35.00
CHANNELS INFO
==============================================================================
                #0      #1      #2      #3      #4      #5      #6      #7
DETECTED        1       1       1       1       1       1       1       1
TXPWR [V]       1.02    0.04    0.04    0.04    1.84    0.04    0.06    0.04
RFLPWR [V]      0.33    0.04    0.05    0.05    1.84    0.04    0.04    0.06
TXPWR [dB]      27.06   0.02    -1.16   -1.07   53.64   -1.04   -1.92   -0.56
RFLPWR [dB]     30.98   0.00    -0.06   -0.39   58.36   -0.69   0.71    -2.20
I30V [A]        0.203   0.156   0.155   0.154   0.151   0.150   0.164   0.164
I6V0 [A]        0.041   0.031   0.031   0.031   0.030   0.030   0.033   0.033
IN8V0 [mA]      0.862   0.662   0.660   0.655   0.642   0.636   0.699   0.699
ON              1       1       1       1       1       1       1       1
SON             1       1       1       1       0       1       1       1
IINT            0       0       0       0       0       0       0       0
OINT            0       0       0       0       0       0       0       0
SINT            0       0       0       0       1       0       0       0
SINTE           1       1       1       1       1       1       1       1
SINTV           38.0    38.0    38.0    38.0    38.0    38.0    38.0    38.0
OVC             0       0       0       0       0       0       0       0
ADC1            1665    66      70      69      3013    69      102     72
ADC2            535     69      74      80      3014    70      71      98
DAC1            92      94      83      93      90      90      82      95
DAC2            250     244     240     261     166     254     253     166
SCALE1          52      57      56      54      53      54      53      54
OFFSET1         258     65      134     127     183     125     204     102
BIASCAL         3870    2130    285     2475    540     1785    3570    1395
HWIS            9.03    8.42    8.63    10.30   10.52   9.64    9.26    8.65
HWIO            -13.55  -9.94   -10.92  -20.68  -21.56  -16.90  -14.83  -11.25
LTEMP           31.25   32.25   34.25   34.25   150.25  33.00   31.25   32.00
RTEMP           32.00   32.00   36.50   35.00   150.25  33.00   30.00   34.50
==============================================================================
[... 116 lines cut ...]
RFLPWR [dB]     31.00   0.00    -0.04   -0.37   21.16   -0.67   0.73    -2.17
I30V [A]        0.020   0.020   0.020   0.020   0.020   0.020   0.020   0.020
I6V0 [A]        0.004   0.004   0.004   0.004   0.004   0.004   0.004   0.004
IN8V0 [mA]      0.083   0.083   0.083   0.083   0.083   0.083   0.083   0.083
ON              1       1       1       1       1       1       1       1
SON             1       1       1       1       0       1       1       1
IINT            0       0       0       0       0       0       0       0
OINT            0       0       0       0       0       0       0       0
SINT            0       0       0       0       1       0       0       0
SINTE           1       1       1       1       1       1       1       1
[... cut ...]
OFFSET1         258     65      134     127     183     125     204     102
BIASCAL         3870    2130    285     2475    540     1785    3570    1395
HWIS            9.03    8.42    8.63    10.30   10.52   9.64    9.26    8.65
HWIO            -13.55  -9.94   -10.92  -20.68  -21.56  -16.90  -14.83  -11.25
LTEMP           0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00
RTEMP           1.00    1.00    1.00    1.00    1.00    1.00    1.00    1.00
==============================================================================
```

Jochenwolf3, 2018-09-13:
```PGOOD: 1
FAN SPEED: 0 %
AVG TEMP: 0.00 CURRENT: 0.00
CHANNELS INFO
==============================================================================
                #0      #1      #2      #3      #4      #5      #6      #7
DETECTED        0       0       0       0       0       0       0       0
TXPWR [V]       0.77    0.69    0.71    0.70    0.70    0.70    0.69    0.70
RFLPWR [V]      0.71    0.71    0.71    0.71    0.70    0.69    0.68    0.70
TXPWR [dB]      0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00
RFLPWR [dB]     0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00
I30V [A]        0.000   0.000   0.000   0.000   0.000   0.000   0.000   0.000
I6V0 [A]        0.000   0.000   0.000   0.000   0.000   0.000   0.000   0.000
IN8V0 [mA]      0.000   0.000   0.000   0.000   0.000   0.000   0.000   0.000
ON              0       0       0       0       0       0       0       0
SON             0       0       0       0       0       0       0       0
IINT            0       0       0       0       0       0       0       0
OINT            0       0       0       0       0       0       0       0
SINT            0       0       0       0       0       0       0       0
SINTE           0       0       0       0       0       0       0       0
SINTV           0.0     0.0     0.0     0.0     0.0     0.0     0.0     0.0
OVC             0       0       0       0       0       0       0       0
ADC1            1267    1132    1160    1147    1141    1141    1136    1144
ADC2            1166    1157    1160    1155    1143    1136    1118    1146
DAC1            0       0       0       0       0       0       0       0
DAC2            0       0       0       0       0       0       0       0
SCALE1          0       0       0       0       0       0       0       0
OFFSET1         0       0       0       0       0       0       0       0
BIASCAL         0       0       0       0       0       0       0       0
HWIS            0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00
HWIO            0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00
LTEMP           0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00
RTEMP           0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00
```

@wizath Our booster seems to have died now...

Jochenwolf3, 2018-09-13:
Oh shit! That was the last test I wanted to perform and now the Booster is dead. No lights. Even after a power cycle. Any suggestions????

wizath, 2018-09-13:
Could you show me photo how you managed to heat them up? Because I've heated module up to 50 degrees and everything works as expected

hartytp, 2018-09-13:
Looking at IC11 on the main board (the I2C switch). Pull up to 3V3 works fine, but all SC/SD output lines are at 1.2V

wizath, 2018-09-13:
All (8) channels are at 1.2V?

hartytp, 2018-09-13:
> Could you show me photo how you managed to heat them up? Because I've heated module up to 50 degrees and everything works as expected

As I said, with my v1.0 unit it all worked fine to 80C+.

All I did was take the lid off and wave a hot air gun over one of the modules.

hartytp, 2018-09-13:
> All (8) channels are at 1.2V?

Yes

wizath, 2018-09-13:
If you take one module off, something changes?

hartytp, 2018-09-13:
after removing all modules, the outputs of the I2C switch are all 1.2V

wizath, 2018-09-13:
Do you have good P3V3 rail?

hartytp, 2018-09-13:
No, I take that back. 3V3 after removing all channels. There must have been one dead channel...

wizath, 2018-09-13:
Can you track dead channel? Most likely temperature sensor failed and shorted I2C bus (that happened to me a while ago). Weird, that my protection didn't worked - need to revise that.

hartytp, 2018-09-13:
I'm pretty sure this is a mechanical issue.

After taking all the channels out and putting them back in it works fine.

The technosystem mechanics are beautiful but the assembly is very over-defined. I suspect that this is the same issue as the channel that only started to work when we fitted the lid (which presses against the front panel and changes the alignment of the D-sub pins).

hartytp, 2018-09-13:
@wizath it might be worth simulating an I2C bus error by shorting the SD and SC lines for one channel and checking that your firmware correctly flags the error and disables all channels

wizath, 2018-09-13:
Yes, I was thinking about the same thing

hartytp, 2018-09-13:
oh, ffs, after doing the screws all up no channels work any more...

wizath, 2018-09-13:
Ok, I2C fail protection is fixed

hartytp, 2018-09-13:
This turned into a massive headache.

As I said, we took all the modules out and replaced them without adding the screws and it worked. Then we added the screws and it stopped working. After a few cycles of taking things out, wiggling them, redoing the screws we finally got it assembled with the lid on and all channels working. Then we put it in the rack and it stopped working again. After a few more rounds of moving things and adjusting screws we finally got it in the rack and working, but it took quite some time.

Since there is clearly some bad mechanical issue here and we don't have any spare units, I don't want to do any more thermal testing on this unit or any other tests that require me to open it up!

closed 2018-09-13 by hartytp

hartytp, 2018-10-23:
This is definitely some kind of mechanical issue. After removing the screws that clamp the module bar down to the main pcb I am no longer able to reproduce this issue.