sinara-hw/Booster #103: post simulation files

hartytp, 2018-09-12:
If there are any simulation files for the design (e.g. microwave office) then please post them to the repository.

Booster is a really lovely platform for building amplifiers and I can imagine wanting to use it to build microwave amps etc. It would be much easier to do that starting from the current design if we have the simulation files.

gkasprow, 2018-09-13:
Unfortunately Tomasz did not want to share his simulations. This is one of
many reasons I finished cooperation with him. He is not related with WUT
anymore.
Hopefully Maciej will be so kind and share his files.

hartytp, 2018-09-13:
> Unfortunately Tomasz did not want to share his simulations. 

:(

> Hopefully Maciej will be so kind and share his files.

That would be great!

Some time ago I put together a quick simulation of most of an old version of Booster using microwave office and the manufacturer's S-parameter models. I'm sure that Maciej did a much better job than me, but we can use my model as a backup.

hartytp, 2019-04-29:
ping @gkasprow Can you ask Maciej to post any simulation/design files he has for this project please?

gkasprow, 2019-04-29:
I asked him again

gkasprow, 2019-04-29:
Unfortunately, Maciej does not have simulation files - he simulated very first prototype with QUCS but cannot track the files anymore. Tomasz did not agree to share his files, I don't have contact with him anymore. He left WUT and does not respond to any form of contact.

hartytp, 2019-04-29:
Shame but ok

closed 2019-04-29 by hartytp