sinara-hw/Booster #108: firmware update questions

hartytp, 2018-09-26:
@wizath re https://github.com/sinara-hw/Booster/issues/98#issuecomment-424757122

Thanks a lot for doing that!

> Added I2C restart procedure after start/TX/RX/ACQ errors

Do you think that will fix the issue we saw where one module not being correctly inserted could cause Booster not to detect any modules? were you able to reproduce that issue in the end?

> Calibration fixes

What did you change in the calibration? Do we need to recalibrate anything after updating the firmware?

> Dropped SW Interlock function

Great. What are the interlock commands now?

wizath, 2018-09-26:
> Do you think that will fix the issue we saw where one module not being correctly inserted could cause Booster not to detect any modules? were you able to reproduce that issue in the end?

As I was testing I2C protections I've discovered following issues:
- If we short SDA line to GND, that causes little errors during readouts, temperature are going off the charts (down 1 degree) and channel is disabled for protection
- If we short SCL line to GND, whole I2C bus is going crazy (probably glitch in I2C state machine or something related to clock stretching feature) and then reading next channel is not possible even after restarting I2C MUX - so the only way to continue working is to reset whole I2C peripherial (which works fine and even after glitch on SCL remaining channels are able to work)

> What did you change in the calibration? Do we need to recalibrate anything after updating the firmware?

At this moment there is no need to recalibrate anything, just some fixed to calibration commands to make them work faster. Although I was wondering of using float coefficients for power readouts to boost accuracy.

> Great. What are the interlock commands now?

for SCPI `int:pow ch,val` - I'll add querying for interlock level tomorrow (since I forgot about it).

hartytp, 2018-09-26:
> As I was testing I2C protections I've discovered following issues:

> - If we short SDA line to GND, that causes little errors during readouts, temperature are going off the charts (down 1 degree) and channel is disabled for protection
> - If we short SCL line to GND, [...]

Nice! Good work. So, maybe that explains what we saw. Out of curiosity, were you able to reproduce the issue by mechanically stressing the unit, or could you only do it by deliberately shorting the lines?

Unfortunately, I can't try to reproduce the issue with the new firmware because both the Boosters we currently have are in active use and can't be taken off line to deliberately try to break them 

> At this moment there is no need to recalibrate anything, just some fixed to calibration commands to make them work faster. Although I was wondering of using float coefficients for power readouts to boost accuracy.

Aah, nice.

> for SCPI int:pow ch,val - I'll add querying for interlock level tomorrow (since I forgot about it).

Cool, thanks. I'll add that to our driver.

closed 2018-09-26 by hartytp