sinara-hw/Booster #110: Booster 1.2 review

hartytp, 2018-10-21:
Thanks for all the hard work on this Greg and team.

I'll review this by end of Monday and get back with any comments.

Anyone else interested in Booster, now is your last chance to have your say before we move from prototypes to "production".

hartytp, 2018-10-22:
- [ ] The "Booster characteristics" section on the schematics is good, but should probably be on the RF module schematic not on the main board schematic. The reason for this is that Booster may be used with a different RF amplifier module in the future (it's basically a really nice platform for building PAs).

hartytp, 2018-10-22:
- [ ] in the power sequencing list (both on the RF channel connection pages and on the power supply page) is there a mistake in the ON sequence? Should "increase 5V to required Icc1" be "decrease OUT_BIAS to give required I28V0"?
- [ ]  add an annotation for what range of `OUT_BIAS` can be supplied by the OpAmp. Make sure that this cannot destroy the output FET.
- [ ] why is R89_1 needed?

hartytp, 2018-10-22:
NB I haven't checked things like the configuration of the 28V0 regulator (e.g. fold back limiting) or the microprocessor. AFACIT, those have been checked already in the V1.1 prototypes so don't need rechecking (but if they do, please can you make sure you've done that @gkasprow)

hartytp, 2018-10-22:
- [ ] main board and supplemental control board annotation for the d-type. Should `P28VT` be `P28V0`

hartytp, 2018-10-22:
- [ ] add bias pot hold in supplemental control board. We probably won't use the pot, but might as well have an access hole for it just in case

hartytp, 2018-10-22:
- [ ] on the supplemental control board is there any reason not to have a ferrite bead on the output overdrive connection?

hartytp, 2018-10-22:
- [ ] on the supplemental control board why is the signal_gnd pin ac coupled to ground instead of being dc connected to ground?

hartytp, 2018-10-22:
- [ ] should there be a capacitor between R2 and r3 on the supplemental control board to remove the glitches?

hartytp, 2018-10-22:
- [ ] on the supplemental control board what does the comment "disable ADL output refer to"?

hartytp, 2018-10-22:
- [ ] since the power detector comparator sinks quite a bit of current should we add 0R resistors to the supplemental control board to allow them to be disconnected? Or do we think that the OpAmp sources so much current that this isn't an issue?

hartytp, 2018-10-22:
Is the power budget on the supplemental control board correct? IIRC Booster draws approx 70W, but each module should only draw 5.5W so 44W total. Is the difference just due to SMPS losses, fans etc?

hartytp, 2018-10-22:
- [ ] Comment on the supplemental control board should read something like `IC1A holds interlocks in reset for ~10us to avoid tripping due to transients during enable`

hartytp, 2018-10-22:
- [ ] on the preamp schematic, what does `pin max -5dBm` mean? Is that `Psat` or absolute max without damage? I assume it's Psat/P3dB or something like that.

hartytp, 2018-10-22:
pre-amp schematic:
- [ ] c16 has resistor symbol not capacitor
- [ ] why do we need C16? Why do we want a filter on power detector reset line?
- [ ] R? and L? are missing part numbers L? is missing a value (what is BLM?)
- [ ] IIRC the gain of the OpAmps after the power detectors used to be 2x. Now it's 1x. Is that okay for the resolution of the ADCs on the microprocessor? 1dB is 35mV, so we need the ADCs on the microprocessor to have resolution + noise that's << 35mV
- [ ] why is the OpAmp powered from 5V and not 3v3?
- [ ] not that it really matters, but that's a slightly expensive OpAmp isn't it? It doesn't need to be particularly low noise or anything special. Just fast enough to allow the interlock to settle to 10% in a few us

hartytp, 2018-10-22:
power amp
- [ ] please translate annotations from polish to english
- [ ] please make signals consistent with other boards. e.g. shouldn't VCC_PWR_AMP_IN be labelled OUT_BIAS to be consistent with other schematics?

hartytp, 2018-10-22:
@gkasprow that's my review complete.

The schematics are really beautiful, quite meticulous. Thanks for that.

Tomorrow I will do a thorough post-mortem of the dangerous booster unit and understand what is wrong with it. That will need careful addressing before we can move forwards.

hartytp, 2018-10-23:
@gkasprow as always, feel free to use some discretion here. Any design can be tweaked and improved forever, but at some point the value in making changes is not worth the effort. Overall, I'm very happy with Booster, and will leave it to you to decide which of these issues should be fixed.

hartytp, 2018-10-23:
@gkasprow one other thing that occurred to me is that the attenuators between the directional coupler in the PA stage and the power detector are too large. See figure 19 on the ADl5904 data sheet. The best accuracy for the power detector is between approx 12.5dBm and -15dBm. So, it makes sense to adjust the attenuators so that the highest power we wish to apply results in 12.5dBm at the power detector. Assuming the highest useful power for Booster is 38dBm, we have 17dBm after the directional coupler. So the attenuation should only be approx 5-6dB, maybe let's go for 8dB to be safe (we don't want to saturate the detector). We currently have 15dB.

This results in lower accuracy for power measurements/interlock when booster is operated at lower output powers. Of course, it's not likely that people will use Booster much below 0.25W (there are much better solutions for lower powers) so this isn't crucial, but it's a quick change so we might as well do it properly.

hartytp, 2018-10-24:
- [ ] Can we have some silk on the back of the D-type on the suplemental control board that labels pin numbers (e.g. 5, 15, 11, 6).
- [ ] can we have a tp on the supplemental control board for p5V0MP?

hartytp, 2018-10-24:
- [ ]  on the main board schematic, the adc channel that measures the P5VMP rail is labelled meas_P3V3MP

gkasprow, 2018-10-30:
@hartytp R89_1 on the main board is needed to make sure that the most negative voltage will arrive at the BIAS_OUT before DAC gets initialised. Just in case somebody does sth stupid with firware.

gkasprow, 2018-10-30:
> should there be a capacitor between R2 and r3 on the supplemental control board to remove the glitches?
This is return signal for FWD/RFL measurements to mitigate DC drop on the GND caused by the main supply rails.

gkasprow, 2018-10-30:
LT6220 is 1.59@100pcs. I can have a look for lower cost rail to rail OA. I had this on in my libs.

gkasprow, 2018-10-30:
The RF board was designed by Maciej. He used his own libraries. I didn't wanted to touch his part but I feel that  it is necessary to clean it and standardise a little.

hartytp, 2018-10-31:
> This is return signal for FWD/RFL measurements to mitigate DC drop on the GND caused by the main supply rails.

I don't understand this. R2 and R3 are RFA and RFB, which are the TTL control lines for the RF switch. They should have a filter capacitor to remove switch glitches.

hartytp, 2018-10-31:
> LT6220 is 1.59@100pcs. I can have a look for lower cost rail to rail OA. I had this one in my libs.

Okay, maybe I'm just being a bit cheap, but 1.60 for an OpAmp in quantity seems quite expensive. The power detector output is 35mV per dB and we only need noise/stability below 0.1dB, so a few mV which is pretty easy.

Still, this isn't a lot of money in the grand scheme so if you don't have another part readily available then don't bother changing it.

gkasprow, 2018-10-31:
Oh, I replied to wrong issue. You asked why SIGNAL_GND is not connected to GND but via 1nF capacitor.  And I replied that this is return signal for FWD/RFL measurements to mitigate DC drop on the GND caused by the main supply rails.

hartytp, 2018-10-31:
aah, ok, that makes more sense. Still, doesn't it need to be tied to ground somewhere?

gkasprow, 2018-10-31:
It is. Close to CPU and at the RF board. But I'm not sure if that really helps.

hartytp, 2018-10-31:
> and at the RF board

Aah, ok! I'd missed that somehow. Makes sense, thanks for clarifying.

gkasprow, 2018-11-01:
@hartytp Let's use [MCP6291](http://ww1.microchip.com/downloads/en/devicedoc/21812c.pdf). It's 0.4EUR@100pcs

gkasprow, 2018-11-01:
One booster is 8*3*1.2 = 28Eur. You own me a lot of beer :)

hartytp, 2018-11-01:
:)

gkasprow, 2018-11-04:
> why do we need C16? Why do we want a filter on power detector reset line?

it was added by Thomas during debugging stage. I don't want to remove it. I believe he did it on purpose.

>  IIRC the gain of the OpAmps after the power detectors used to be 2x. Now it's 1x. Is that okay for the resolution of the ADCs on the microprocessor? 1dB is 35mV, so we need the ADCs on the microprocessor to have resolution + noise that's << 35mV

We had to reduce the gain because ADC range is 2.5V. The DAC range on SUP board is also 2.5V
So I will use gain of 1.5x instead to match the ranges.

gkasprow, 2018-11-04:
The attenuators are now 8dB 
[image]

hartytp, 2018-11-04:
> I think I addressed all issues, closing.

🎆  whoo! Thanks. I'll check over the design this eve to confirm and so we can move on to manufacture on Monday.