sinara-hw/Booster #111: Electrically unsafe

hartytp, 2018-10-22:
@gkasprow there is a serious safety issue with Booster v1.1. One of our students got a nasty shock and  electrical burn from touching the rear exterior of Booster. It seems like there may be some ungrounded metal that is not correctly grounded and became exposed to mains.

We will investigate this tomorrow and try to locate the cause.

gkasprow, 2018-10-23:
Was it correctly grounded? The ungrounded enclosure will get 110V AC if not grounded because of EMC filter capacitors. It is not dangerous but not very pleasant when somebody touches it. Every higher power mains supplied device have such feature and needs grounding. If you use cheap Chinese power cord, it's possible that it does not have PE wire at all.

hartytp, 2018-10-23:
@gkasprow I've done a careful inspection of Booster from a safety point of view. Overall I have to say that I am impressed with the quality of the construction (both the design and TechnoSystem's assembly). All mains wiring is neat and well strain relieved.

I've checked every piece of exposed metal for grounding. Everything, including the anodised screws, is correctly safety grounded *apart* from the screws that hold the fans in. The holes for those screws were annodised after drilling and they are not tapped. As a result, those screws are floating. That must be fixed in the next version (maybe use a start washer between the screw and the chassis, or use plastic screws). 

Having said that, I cannot see how any of those screws can accumulate enough charge/voltage to become dangerous. I'll look at the overall setup next and see if I can understand it. As you say, it's possible that the mains power cord is at fault here and not Booster.

gkasprow, 2018-10-23:
Oh, in the prototype the fan screws were tapped.

hartytp, 2018-10-23:
The fan screws are self tapping, but they screw into the plastic fans, not the chassis itself.

hartytp, 2018-10-23:
@gkasprow I've completed a thorough examination of Booster and it all looks great apart from those screws. However, I can't see any way they could have come live since the mains wiring inside is very good.

We must fix those screws in the next version. But, other than that my conclusion is that the safety issue must lie somewhere else in the setup. Sorry for the wrongful accusation! I'll change the title of the issue accordingly.

gkasprow, 2018-10-24:
I was thinking bout using star washer but I'm not sure if it is aesthetically acceptable :)

hartytp, 2018-10-24:
It's fine by me (it's only the back anyway). Anyway, let's just do this the cheapest way possible that doesn't leave unearthed metal exposed.

hartytp, 2018-11-05:
@gkasprow is this fixed? If so, please close.

hartytp, 2019-03-06:
I can confirm this is fixed in the TechnoSystems v1.3 mechanics I have.

hartytp, 2019-03-06:
@gkasprow do we need to keep this (and other issues) open until the Creotech mechanics has been tested, or will they do their own testing?

gkasprow, 2019-03-06:
let's keep it.

hartytp, 2019-03-06:
Okay. Will someone from Creotech close this issue once they've checked it with their system? Should I reopen other issues like the SMA cutouts?

gkasprow, 2019-03-06:
Yes, please. Assign it to  @mmroz88 who is responsible for Booster but he did not accept the invitation yet.

hartytp, 2019-03-07:
Okay. When @mmroz88 I will assign him.

hartytp, 2019-03-07:
@gkasprow will you chase him to join, or should I?

gkasprow, 2019-03-07:
I will.

closed 2019-03-07 by hartytp