sinara-hw/Booster #112: ERROR LEDs lit when RF modules screwed in

hartytp, 2018-10-23:
As reported before, we're having an issue with one Booster. Now that we have a spare Booster to use in our experiments I can investigate this.

When I undo some of the screws holding the RF modules into the enclosure there is not issue.

When I do up the screws using the old firmware no modules are detected.

With the new firmware, all RED leds are lit. VCP reporst

```
PGOOD: 1
FAN SPEED: 0 %
AVG TEMP: 0.00 CURRENT: 0.00
CHANNELS INFO
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	0	0	0	0	0	0	0	0	
HWID		00:00	00:00	00:00	00:00	00:00	00:00	00:00	00:00	
TXPWR [V]	0.80	0.73	0.73	0.72	0.72	0.73	0.74	0.71	
RFLPWR [V]	0.72	0.73	0.73	0.75	0.72	0.71	0.71	0.73	
TXPWR [dB]	0.00	0.00	0.00	0.00	0.00	0.00	0.00	0.00	
RFLPWR [dB]	0.00	0.00	0.00	0.00	0.00	0.00	0.00	0.00	
I30V [A]	0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000	
I6V0 [A]	0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000	
IN8V0 [mA]	0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000	
ON		0	0	0	0	0	0	0	0	
SON		0	0	0	0	0	0	0	0	
IINT		0	0	0	0	0	0	0	0	
OINT		0	0	0	0	0	0	0	0	
SINT		0	0	0	0	0	0	0	0	
OVC		0	0	0	0	0	0	0	0	
ADC1		1305	1192	1200	1172	1184	1193	1204	1161	
ADC2		1178	1193	1188	1221	1186	1166	1168	1190	
DAC1		0	0	0	0	0	0	0	0	
DAC2		0	0	0	0	0	0	0	0	
SCALE1		0	0	0	0	0	0	0	0	
OFFSET1		0	0	0	0	0	0	0	0	
BIASCAL		0	0	0	0	0	0	0	0	
HWIS		0.00	0.00	0.00	0.00	0.00	0.00	0.00	0.00	
HWIO		0.00	0.00	0.00	0.00	0.00	0.00	0.00	0.00	
LTEMP		0.00	0.00	0.00	0.00	0.00	0.00	0.00	0.00	
RTEMP		0.00	0.00	0.00	0.00	0.00	0.00	0.00	0.00	
==============================================================================
```

Looking on the main PCB I see that the LEDs around the modules are not lit. The main power LEDs are all lit apart from P5V0.

Measuring on the supplemental control board, I see that all power supplies are close to 0. The 3v3 rail on the supplemental control board is currently at approx 0.5V.

@gkasprow any ideas what is wrong with this unit?

hartytp, 2018-10-23:
@gkasprow apologies for not reporting this sooner -- I couldn't look into it until the new booster unit arrived since the unit with the issue was being used in experiments.

If we're confident this is just some mechanical issue, which will be fixed in the next version then maybe we don't need to understand it. However, it seems worth understanding this now, so that it doesn't come back to bite us later...

wizath, 2018-10-23:
Probably there is a short in I2C bus, because software can't detect any channel at all

hartytp, 2018-10-23:
What kind of short? If a single channel has a short to ground, that won't bring down the entire bus will it? Wouldn't that need to be a short before IC11 (the I2C switch).

hartytp, 2018-10-23:
@wizath even if i2c is shorted shouldn't there be some supply active on the supplemental control board? I measured 0.5V on the 3v3 line. Won't that prevent the i2c chips on the rf modules from working? Isn't this more likely to be a supply short than an i2c issue?

wizath, 2018-10-23:
Supply short is possible too. The fact is that something prevented software from discovering ICs on supplemental board (because that's what SW is looking for when detecting channels).

hartytp, 2018-10-23:
What supplies should be on during init? If there is an i2c error does it have to be the 3v3 rail?

hartytp, 2018-10-24:
@gkasprow okay, the issue here is that the P5V0 rail is shorted to ground.
- measuring on the supplemental control board for channel 7 P5V0MP I see approx 20Ohms to ground
- measuring on the main board on the resistor that drives the LED (i.e. directly on the P5V0 rail that drives P5V0MP) I see approx 1Ohm to ground
- I think that the 0.5V on the module P3V0 rail probably comes from the I2C bus driving the LDO via a protection diode or something like that

So, there appears to be a mechanical issue with this unit that means that when all bolts are tightened the PV50 rail becomes shorted to ground so no modules are detected (the module 3V3 rail is derived from the P5V0MP rail, which is in turn driven by P5V0 via a ferrite + fuse).

hartytp, 2018-10-24:
hmmm...all modules removed and the P5V0 rail is still shorted to ground

hartytp, 2018-10-24:
The fuses on each channel have about a 10-20Ohm resistance, which allows me to verify that the short was not after any of the fuses, so it must be on the main rail.

The P6V5 rail seems fine (I measure 6V0 on the big capacitor, after a Schottky diode, so approx 6V5 on the regulator).

gkasprow, 2018-10-24:
I used 6mm pads for mounting holes. So the washers are simply to big. I can also use bigger pads.

hartytp, 2018-10-24:
Maybe TechnoSystems changed the bolts when they changed the mechanics?

@gkasprow I think this really highlights the fact that we need to ensure that Booster has a complete mechanical BOM including PNs for every bolt and also full CAD models for the mechanics. Otherwise issues can easily creep in when suppliers produce the design.

>  I can also use bigger pads.

:+1: That sounds like a good idea.

hartytp, 2018-11-05:
@gkasprow is this fixed? If so, please can we close the issue.

hartytp, 2019-01-31:
@gkasprow ping...if this is fixed? If so, please close. Thanks!

closed 2019-01-31 by gkasprow