sinara-hw/Booster #113: v1.1 rework question

hartytp, 2018-10-23:
@gkasprow a question for you: when I was inspecting our booster v1.1 units I noticed that one of them has a surface mount resistor soldered onto the LMZ regulator, but another one does not. Presumably this is to fix the EN over voltage issue. Should the resistor be there on all units?

gkasprow, 2018-10-23:
yes, it should

hartytp, 2018-10-23:
@gkasprow thanks for confirming. In that case, I'll add it to the unit that is missing it. Can you remind me what the value should be please?

gkasprow, 2018-10-23:
1k5...2k0

hartytp, 2018-10-24:
@gkasprow FWI, this rework was only applied to one of the three Boosters that were shipped to us. Is it actually necessary?

Do I need to cut any traces before adding the resistor?

gkasprow, 2018-10-24:
no.

hartytp, 2018-10-25:
"no" it's not actually necessary, or "no" I don't need to cut any traces?

gkasprow, 2018-10-25:
It's better to install this resistor.
You don't need to cut traces.

hartytp, 2018-10-25:
Okay, thanks. I'll do that then.

Are there any other important pieces of rework that I should check for while I'm at it?

gkasprow, 2018-10-25:
I went through issues and it looks like it was the only rework.

closed 2018-10-25 by hartytp