sinara-hw/Booster #115: gain fluctuactions on one channel

hartytp, 2018-10-24:
On the newest booster we have been sent, the gain for channel 5 seems to fluctuate between approx 30dB and 40dB. This effect disappeared for a while and then came back. I'll investigate, but it looks like either a mechanical or supply issue.

Driving the channel from a synth at 200MHz, approx -10dBm, I measure an output power of approx 30dBm. That's consistent with 40dB gain, so currently the channel is working again...grrr...

VCP trace is:
```
PGOOD: 1
FAN SPEED: 0 %
AVG TEMP: 49.50 CURRENT: 49.50
CHANNELS INFO
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	1	1	1	1	1	
HWID		60:8B	CE:DC	9E:60	E6:4F	91:0D	25:CD	AF:03	D4:60	
TXPWR [V]	0.04	0.05	0.04	0.04	0.06	1.13	0.05	0.05	
RFLPWR [V]	0.04	0.04	0.04	0.07	0.05	0.17	0.05	0.14	
TXPWR [dB]	-0.89	-0.02	0.00	-0.09	-2.19	30.78	-0.56	-0.39	
RFLPWR [dB]	-2.51	-0.77	-0.96	-1.27	-2.83	36.50	-4.11	-6.56	
I30V [A]	0.156	0.150	0.159	0.152	0.149	0.228	0.129	0.167	
I6V0 [A]	0.029	0.030	0.032	0.031	0.030	0.046	0.026	0.033	
IN8V0 [mA]	0.662	0.642	0.678	0.647	0.634	0.972	0.546	0.709	
ON		1	1	1	1	1	1	1	1	
SON		1	1	1	1	1	1	1	1	
IINT		0	0	0	0	0	0	0	0	
OINT		0	0	0	0	0	0	0	0	
SINT		0	0	0	0	0	0	0	0	
OVC		0	0	0	0	0	0	0	0	
ADC1		69	75	72	73	101	1845	74	75	
ADC2		73	72	72	107	87	281	75	226	
DAC1		84	96	90	85	89	81	89	1500	
DAC2		126	132	118	116	130	259	110	130	
SCALE1		55	57	55	54	52	54	55	57	
OFFSET1		118	76	72	78	215	183	105	97	
BIASCAL		3810	2490	2115	1800	2445	2145	2160	1920	
HWIS		9.65	10.28	9.01	9.13	10.01	8.61	8.27	10.21	
HWIO		-16.75	-20.23	-13.02	-13.44	-18.73	-10.85	-8.91	-19.74	
LTEMP		45.25	47.00	46.25	48.25	48.25	46.00	45.00	43.00	
RTEMP		44.00	48.25	46.25	49.50	49.25	49.25	48.50	46.50	
==============================================================================
```

@wizath any idea why the reflected power is showing up as being so high? Was that calibrated correctly on this unit?

hartytp, 2018-10-24:
Aah, it's come back again: still driving with -10dBm at 200MHz, but output power is not only 22dBm. VCP trace is

```
PGOOD: 1
FAN SPEED: 0 %
AVG TEMP: 50.50 CURRENT: 50.50
CHANNELS INFO
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	1	1	1	1	1	
HWID		60:8B	CE:DC	9E:60	E6:4F	91:0D	25:CD	AF:03	D4:60	
TXPWR [V]	0.04	0.05	0.04	0.04	0.06	0.86	0.05	0.05	
RFLPWR [V]	0.05	0.04	0.04	0.07	0.05	0.15	0.05	0.14	
TXPWR [dB]	-0.89	-0.04	0.00	-0.09	-2.15	22.69	-0.55	-0.39	
RFLPWR [dB]	-2.51	-0.79	-0.96	-1.27	-2.78	26.10	-4.09	-6.56	
I30V [A]	0.156	0.150	0.160	0.159	0.149	0.161	0.141	0.167	
I6V0 [A]	0.031	0.030	0.032	0.032	0.030	0.032	0.028	0.034	
IN8V0 [mA]	0.662	0.639	0.678	0.673	0.634	0.683	0.600	0.714	
ON		1	1	1	1	1	1	1	1	
SON		1	1	1	1	1	1	1	1	
IINT		0	0	0	0	0	0	0	0	
OINT		0	0	0	0	0	0	0	0	
SINT		0	0	0	0	0	0	0	0	
OVC		0	0	0	0	0	0	0	0	
ADC1		69	74	72	73	103	1408	75	75	
ADC2		73	73	73	107	87	250	75	225	
DAC1		84	96	90	85	89	81	89	1500	
DAC2		126	132	118	116	130	259	110	130	
SCALE1		55	57	55	54	52	54	55	57	
OFFSET1		118	76	72	78	215	183	105	97	
BIASCAL		3810	2490	2115	1800	2445	2145	2160	1920	
HWIS		9.65	10.28	9.01	9.13	10.01	8.61	8.27	10.21	
HWIO		-16.75	-20.23	-13.02	-13.44	-18.73	-10.85	-8.91	-19.74	
LTEMP		44.25	47.00	46.25	47.00	48.25	47.00	47.00	44.00	
RTEMP		45.50	48.00	46.25	50.50	49.25	49.00	47.00	45.00	
==============================================================================
```

hartytp, 2018-10-24:
I can restore the signal by putting a very firm pressure on the input SMA connector with my finger. So, this seems like it has to be a bad cable. I'll remove the cable and confirm. 

I've seen issues like this on more than one channel, so there may be something wrong with the entire batch of cables used by TechnoSystems. I'll investigate further and report back.

hartytp, 2018-10-24:
@gkasprow this is odd. Maybe it's not the coax. Tapping on the lid seems to cause the output power to jump by 10dB!

wizath, 2018-10-24:
> @wizath any idea why the reflected power is showing up as being so high? Was that calibrated correctly on this unit?

That's odd - even with disconnected cable, RFL power should match TX, not exceed it. Maybe there was some kind of error in calibration procedure..

hartytp, 2018-10-24:
Do you know who calibrated this unit and how carefully/if it was tested before shipping?

hartytp, 2018-10-24:
@gkasprow you can see the incorrectly soldered 0402 100nF capacitor

[image]

I replaced this capacitor with a new 0402 100nF capacitor and also glued the coax cable further from the FET to see if that fixed the issue. Unfortunately, the channel has gone from partially working (gain changing by 10dB) to dead (no 29V current draw). I don't know what's wrong with it.

Since this only affects one channel I do not think that we need to delay production of the next batch of Boosters while we figure it out.

However, if it's okay with you, I'll post it back to you for inspection/repair.

FWIW, I have the impression that the quality control was slightly worse on this booster than the other ones. In particular the channel-channel gain variation between the gains is higher (more than +-2dB), so maybe the tuning was note done as carefully. It's not the end of the world, but the first two units we received seemed to be better!

gkasprow, 2018-10-24:
It is blocking capacitor. It should not influence the gain.
Such case would be easily picked by AOI.
If you touch the input of FET with soldering iron which is not properly grounded, you can easily kill it.
We didn't open the modules once we got them from Maciej.
No DC current looks strange. Didn't you break L3 during soldering?

gkasprow, 2018-10-24:
The modules were tuned by Maciej, then @wizath tested and calibrated it.
It looks like we have to specify strict rules to follow.

gkasprow, 2018-10-24:
If you tap the lid and observe gain jump, look at the discrete resistor attenuators. They were tuned by hand and there may be some cold joint.

hartytp, 2018-10-25:
> If you tap the lid and observe gain jump, look at the discrete resistor attenuators. They were tuned by hand and there may be some cold joint.

@gkasprow before reaching for the soldering iron I did a visual inspection, but didn't see anything obviously wrong apart from the incorrectly soldered decoupling capacitor.

My thinking is that there are two likely possible causes of this issue: (1) a bad joint, for example in an attenuator (2) a parasitic oscillation somewhere (depending on where the oscillation is, this may be hard to detect since a high-frequency oscillation could be filtered out by subsequent matching/gain stages). Given the way this came and went when I tapped the lid, I think that a parasitic oscillation is more likely, but I could be wrong.

If it's a bad joint then I'm not worried, since the QC at Creotech should spot these things in future.

If it's an oscillation, then the question is what the cause is. Again, if it's a bad connection due to some QC issue then it's fine as it shouldn't be a problem with the Creotech units, but if it's some design issue like some piece of decoupling being a bit marginal or some traces being too close together then we may need to tweak the design a little. The worst-case scenario would be if the design is only marginally stable and component tolerances then make a small number of boards oscillate sometimes. I do not think that's the case, but I would be more confident if we could understand what was wrong with this unit. So, I'll post it back to you for analysis.

closed 2018-11-05 by hartytp