sinara-hw/Booster #117: USB jack

hartytp, 2018-10-24:
@gkasprow something that would be nice to fix for the next version if possible: the micro USB on the front panel is quite far recessed due to the thick front panel, so USB cables don't mate very well and are prone to falling out. Is there a longer connector we can use? If there is no easy fix then don't worry (this port isn't used for much).

gkasprow, 2018-10-24:
There are plugs with longer shell available. One can also use normal USB plug but remove ~1mm of plastic mold with sharp blade.
I cannot move it more to the board edge. We can make 1mm deep pit around the connector and this seems to be easiest way.

hartytp, 2018-10-24:
> I cannot move it more to the board edge. We can make 1mm deep pit around the connector and this seems to be easiest way. 

Assuming the machining costs associated with that are really negligible then it's fine with me. Whatever you think is best.

gkasprow, 2018-10-24:
we are machining the front panel anyway from this side so there is no additional cost.

hartytp, 2018-10-24:
Sounds like a good plan then. The current situation can be a bit annoying since the VCP connection is lost very easily if the cable is disturbed which is annoying during debugging.

gkasprow, 2018-11-04:
After discussion we decided to make the cut-out for whole boads from internal side of the panel. So the Ethernet connector and switches will be shifted by 1mm.
Cut-out around the USB connector won't really work because it would require precise machining with very small tools and thus increase the price.