sinara-hw/Booster #118: rc1.3 review

hartytp, 2018-11-05:
- [ ] main board: the power sequencing annotation was fixed on the channel pages but not on the power supply page. 
- [ ] **supplemental control board: there is still no capacitor to fix the switch glitches**
- [ ] supplemental control board: the power budget is not correct since the FET current draw increases with load by something like a factor of two. At the very least we should add a note saying that this is the current consumption with no RF applied. If this power budget was correct then Booster would be >100% efficient at 5W RF output! Also, this power budget does not seem consistent with the annotation on the preamp schematic which says that module power consumption without RF applied is 6W.
- [ ] supplemental control board: please remove the old annotation "IC1A buffers the delay circuit...". That annotation doesn't make sense and there is a new annotation that explains this better.
- [ ] preamp: L1 is still labelled BLM rather than a proper inductance
- [ ] pa: what are "opt1 620 ohm and150pF" and "opt2"? Please add an annotation explaining what these options are for. Also, why do the population options in the schematic not match either opt1 or opt2?
- [ ] pa: what does the annotation "ew. C37 0.7pF, C36 2.2pF, C9 220pF L6 6.8 nH, C37 2.2p" mean?
- [ ] pa: annotations "+3 dB" after the coupler are now incorrect since the attenuators have changed
- [ ] pa: the annotation "R13,R16 =115R || 85R and forms 50R load together with ADL chip" isn't correct any more. (Please make sure that the current power detector termination is correct and will not result in the detector being driven beyond 10dBm input).
- [ ] fix Crms value for the three power detectors

hartytp, 2018-11-05:
@gkasprow thanks for the new rc. All looks good apart from the issues above. Mainly cosmetics apart from the switch capacitor and the Crms.

hartytp, 2018-11-05:
Actually, @gkasprow one final though, but can you increase the 8dB attenuators on the output forward/reverse power detectors to 10dB. Otherwise, we may end up with a bit too much power at the detector due to component tolerances.

hartytp, 2018-11-06:
@gkasprow ping...are you okay to make those changes so we can send the boards off to manufacture, or do you want me to?

Also, still waiting for confirmation that the following are fixed and can be closed: #112 #111 #81 #57

gkasprow, 2018-11-06:
I will do it immediately today and send files for production. I have a
meeting right now.

gkasprow, 2018-11-06:
I used these values
[image]

hartytp, 2018-11-06:
Looks good, but don't forget to either add a termination resistor for the power detector or to adjust the value of one resistor to terminate correctly.

gkasprow, 2018-11-06:
sure :)

gkasprow, 2018-11-06:
all issues fixed, I placed new release
https://github.com/sinara-hw/Booster/releases/tag/v1.3
I will commit files later on

hartytp, 2018-11-06:
Thanks Greg. A few comments:
- [ ] 96.2||85=45R, so why is the attenuator output resistor 39.5Ohm and not 43Ohm?
- [ ] the +10dB annotation after the attenuator is still incorrect (18dBm - 10dB = 8dBm)
- [ ] the annotation "ew. C37 0.7pF, C36 2.2pF, C9 220pF L6 6.8 nH, C37 2.2p" is still present on the PA schematic and is still unintelligible (and does not match the components on the schematic)
- [ ] Please confirm that these issues are fixed and close: #112 #111 #81 #57

gkasprow, 2018-11-06:
>     96.2||85=45R, so why is the attenuator output resistor 39.5Ohm and not 43Ohm?

OK, I changed the PN (ERJ-PA3F46R4V) but not the value displayed..
In this particular design made by Maciej, non-CERN libraries were used so one has to edit all component fields manually.

>     the +10dB annotation after the attenuator is still incorrect (18dBm - 10dB = 8dBm)

Ok, I mixed up power levels with attenuation.
I think Maciej meant attenuation so he used dB not dBm. Let's stay with dBm then.

>     the annotation "ew. C37 0.7pF, C36 2.2pF, C9 220pF L6 6.8 nH, C37 2.2p" is still present on the PA schematic and is still unintelligible (and does not match the components on the schematic)

OK, I didn't notice that. I removed another ones.

>     Please confirm that these issues are fixed and close: #112 #111 #81 #57

They were passed to mechanical team. I want to close them once I get final CAD files.

hartytp, 2018-11-06:
> OK, I changed the PN (ERJ-PA3F46R4V) but not the value displayed..
> In this particular design made by Maciej, non-CERN libraries were used so one has to edit all component fields manually.

Aah, makes sense! Thanks for fixing.

> Ok, I mixed up power levels with attenuation.
I think Maciej meant attenuation so he used dB not dBm. Let's stay with dBm then.

Well, it's a bit confusing since he has dB all over the design. The way I interpret it is that there is a fixed input power at the start of the pre-amp stage and then the dB values track that power through the schematic using the small-signal gain of each stage. So, the dB value of each stage is not the dB gain/loss of that stage, but rather the total gain/loss to the system after that stage, including all prior points. That's why the PA stage starts with +23dB and ends at +38dB.

To be consistent with this, the dB value after the attenuator should be 38dB (output power) - 20dB (coupler) - 10dB (attenuator) = 8dB.

hartytp, 2018-11-06:
> They were passed to mechanical team. I want to close them once I get final CAD files.

Good idea.

hartytp, 2018-11-06:
So, it looks like we can close this issue and begin manufacture. Many thanks Greg!!!

closed 2018-11-06 by hartytp