sinara-hw/Booster #119: SCPI interface not available

ljstephenson, 2018-11-29:
We have three Boosters running the #98 firmware, of which two do not have the SCPI interface available over ethernet (connection is refused).
Using task-stats on the USB-serial interface lists the following tasks in all cases:

```
Task            State   Priority        Stack   #
**************************************************
CLI             R       1               364     5
IDLE            R       0               488     8
DHCP            B       1               320     3
FAN             B       0               685     1
CH MEAS         B       2               679     9
RF Prot         B       4               352     7
UDPServer       S       2               2226    4
Ext             S       2               458     6
ADC             S       3               720     2
CH INFO         S       1               757     10
````

It's not something weird on our network, as I've checked this with the boosters using the same MAC and IP (with only one on at any one time obviously) and the issue persists.

Power-cycling the booster does not resolve this.

The issue spontaneously resolved on _one_ Booster itself after running an nmap check to see if port 5000 was listening or not.
Self-resolving Booster nmap output:

```
$ nmap -A xxx.xxx.xxx.xxx/32 -p 5000
Nmap scan report for <nat> <ip>
Host is up (0.00046s latency).
PORT     STATE SERVICE VERSION
5000/tcp open  upnp?
```

Working nmap output:

```
$ nmap -A xxx.xxx.xxx.xxx/32 -p 5000
Nmap scan report for <nat> <ip>
Host is up (0.00043s latency).
PORT     STATE SERVICE    VERSION
5000/tcp open  tcpwrapped
```

Non-working nmap output:

```
$ nmap -A xxx.xxx.xxx.xxx/32 -p 5000
Nmap scan report for <nat> <ip>
Host is up (0.00049s latency).
PORT     STATE  SERVICE VERSION
5000/tcp closed upnp
```

hartytp, 2018-11-30:
@wizath any ideas what could be causing this or how we debug?

wizath, 2018-12-03:
I'll add debug commands for SCPI server. Investigating issue right now.

hartytp, 2018-12-03:
:+1: Thanks!

cjbe, 2018-12-03:
@wizath the problem is with the "old" (prototype, silver front panel) Booster. As @ljstephenson mentioned, we can ping the device, and we see DHCP information in the bootlog, so it looks like communication between the ethernet module and the CPU is working.

hartytp, 2018-12-03:
> @wizath the problem is with the "old" (prototype, silver front panel) Booster

Okay, so you are only having issues with the v1.0 prototype box we have, not with any of the v1.1 (black boxes) units?

cjbe, 2018-12-03:
@hartytp we saw an identical problem with one of the v1.1 units as well, however this spontaneously resolved as we were investigating it, so it seems there is some underlying common problem.

hartytp, 2018-12-03:
ack. how odd.

ljstephenson, 2018-12-03:
The issue has now reappeared on the v1.1 unit.

wizath, 2018-12-03:
[eth-scpi.zip](https://github.com/sinara-hw/Booster/files/2639507/eth-scpi.zip)

Could you use this firmware with detailed network logs, and post me an output from _logstash_ command?

cjbe, 2018-12-03:
Using this new firmware on the problematic v1.0 (silver FP) Booster solves the problem:

I can successfully connect on port 5000:
```
$ telnet 10.255.6.217 5000
Trying 10.255.6.217...
Connected to 10.255.6.217.
Escape character is '^]'.
^]

telnet> quit
Connection closed.
```
There is nothing sinister-looking in the log:
```
> logstash
[INFO] device boot
[INFO] SYSCLK frequency: 168000000
[INFO] PCLK1 frequency: 42000000
[INFO] PCLK2 frequency: 84000000
[INFO] ADC: OK | raw: 1970 | VrefInt 1.20 V
[INFO] PGOOD: OK
[INFO] Found mainboard EEPROM, loading values
[INFO] RF Channel at 0 detected, id = 54:10
[INFO] RF Channel at 1 detected, id = 54:10
[INFO] RF Channel at 2 detected, id = 54:10
[INFO] RF Channel at 3 detected, id = 54:10
[INFO] RF Channel at 4 detected, id = 54:10
[INFO] RF Channel at 5 detected, id = 54:10
[INFO] RF Channel at 6 detected, id = 54:10
[INFO] RF Channel at 7 detected, id = 54:10
[INFO] network init done, starting server..
[INFO] TCP Server initialization complete
[INFO] network socket state 13
[INFO] network socket listen OK
[INFO] DHCP IP acquired: 10.255.6.217
[INFO] DHCP lease time: 691200 s
[INFO] network client 10.255.6.191 connected
[INFO] network client disconnected
```

Any ideas why this new firmware solves this problem?

cjbe, 2018-12-03:
@wizath another question: The version string states which hardware version the firmware was built for - should I be using different builds of the firmware on the v1.1 and the v1.0 Booster? If so, could you build one for me for the v1.0 hardware, or tell me how to build it?

wizath, 2018-12-03:
sorry, the version string is misleading. Firmware is intended to be _universal_, with some features disabled while working on older hardware.

hartytp, 2018-12-03:
> with some features disabled while working on older hardware.

Does it automatically detect which hardware version it's running on?

wizath, 2018-12-03:
It detects specified hardware features e.g eeprom memory

hartytp, 2018-12-03:
Nice!

cjbe, 2018-12-05:
After power cycling a v1.1 Booster several times this issue has reoccured. Running 'logstash' seems to cause a watchdog reset (front panel LEDs flash), and gives the following output:

```
> logstashturer id 77
                     [WARN] watchdog reset occurred!
[INFO] device boot
[INFO] SYSCLK frequency: 168000000
[INFO] PCLK1 frequency: 42000000
[INFO] PCLK2 frequency: 84000000
[INFO] ADC: OK | raw: 1978 | VrefInt 1.21 V
[INFO] PGOOD: OK
[INFO] Found mainboard EEPROM, loading values
ADS7924 ID: 25==25 | PASS
                         [INFO] RF Channel at 0 detected, id = 54:10
ADS7924 ID: 25==25 | PASS
                         [INFO] RF Channel at 1 detected, id = 54:10
ADS7924 ID: 25==25 | PASS
                         [INFO] RF Channel at 2 detected, id = 54:10
ADS7924 ID: 25==25 | PASS
                         [INFO] RF Channel at 3 detected, id = 54:10
ADS7924 ID: 25==25 | PASS
                         [INFO] RF Channel at 4 detected, id = 54:10
ADS7924 ID: 25==25 | PASS
                         [INFO] RF Channel at 5 detected, id = 54:10
ADS7924 ID: 25==25 | PASS
                         [INFO] RF Channel at 6 detected, id = 54:10
ADS7924 ID: 25==25 | PASS
                         [INFO] RF Channel at 7 detected, id = 54:10

> [INFO] network init done, starting server..
[INFO] TCP Server initialization complete
                                         [INFO] network socket state 13
[INFO] network socket listen OK
[INFO] DHCP IP acquired: 10.255.6.216
[log] ===== W5500 NET CONF : DHCP =====
[log]   MAC:    00:08:DC:AB:CD:EF
[log]   IP:     10.255.6.216
[log]   GW:     10.255.6.254
[log]   SN:     255.255.255.0
[log] =======================================
[INFO] DHCP lease time: 691200 s
```

```
> version
RFPA v1.20, built Nov 19 2018 13:43:11, for hardware revision: v1.10
```

cjbe, 2018-12-06:
@wizath when this problem occurs, the connection is actively refused by the Booster (i.e. we send a SYN, and it replies RST/ACK). We get the same response trying to connect to any other port, so this suggests the Wiznet chip is not being correctly configured.

We have now seen this same problem occur using the older (26th September) firmware as well - so we currently cannot use any of the Booster.

cjbe, 2018-12-12:
@wizath using your latest firmware (6th December) with enhanced logging I saw this problem again:
```
> logstash
[INFO] device boot
[INFO] SYSCLK frequency: 168000000
[INFO] PCLK1 frequency: 42000000
[INFO] PCLK2 frequency: 84000000
[INFO] ADC: OK | raw: 1977 | VrefInt 1.21 V
[INFO] PGOOD: OK
[INFO] Found mainboard EEPROM, loading values
[INFO] RF Channel at 0 detected, id = 54:10
[INFO] RF Channel at 1 detected, id = 54:10
[INFO] RF Channel at 2 detected, id = 54:10
[INFO] RF Channel at 3 detected, id = 54:10
[INFO] RF Channel at 4 detected, id = 54:10
[INFO] RF Channel at 5 detected, id = 54:10
[INFO] RF Channel at 6 detected, id = 54:10
[INFO] RF Channel at 7 detected, id = 54:10
[INFO] network init done, starting server..
[ERROR] TCP Server initialization fail, status 253
                                                  [INFO] network socket listen OK
[INFO] DHCP IP acquired: 10.255.6.216
[INFO] DHCP lease time: 691200 
>
```

This occurs infrequently - on most restarts the interface works properly.

wizath, 2018-12-13:
Oh, an error. As I'm looking at it it seems as socket initialization is returning error because device IP is not set yet (before DHCP acquires it). I'll fix it shortly.

wizath, 2019-01-31:
Fixed

closed 2019-01-31 by wizath