sinara-hw/Booster #120: Interlock values set over serial interface do not save when the Booster is power cycled

bnichol, 2018-11-30:
As the Scipy interface wasn't working on one of our Boosters I tried to set the interlock values using the USB interface. 
I was able to set the interlock values in this way so that the interlock tripped when desired. However after power cycling the Booster it seem sthe interlock values have not been saved.

hartytp, 2018-11-30:
My preference here would be to delete the interlock setting commands from the VCP interface. IMHO there is no point repeating these commands on the VCP interface.

Let's keep the VCP interface to a minimal debugging interface rather than duplicating the SCPI functionality. Of course, that does require the ethernet interface to work robustly (cf https://github.com/sinara-hw/Booster/issues/119)

wizath, 2018-12-03:
Main use for interlock setting command in VCP is for automated calibration software. I can delete command definition and keep them secret.

hartytp, 2018-12-03:
> Main use for interlock setting command in VCP is for automated calibration software.

@wizath My preference would be to expose the interlock calibration commands on the VCP interface, but not the interlock setting API (even if this is used internally by the calibration routines, it shouldn't need to be accessible via the VCP interface).

hartytp, 2019-01-07:
@wizath let's simplify the VCP interface to avoid duplicating functionality that's also available on the SCPI interface. It should be a minimal interface for debugging only.

The problem with duplication is that if the two interfaces are different (in terms of capability or operation) then it can cause a lot of user confusion.

wizath, 2019-04-09:
VCP Interlock level command output is saved to EEPROM. It uses the same method as SCPI. I don't want to delete it from VCP since it's useful in calibration procedures. I can remove the command declaration from help and docs.

closed 2019-04-09 by wizath