sinara-hw/Booster #121: Reverse power interlock always trips on certain channels

cjbe, 2018-12-05:
With the latest firmware the reverse power interlock always seems to trip on certain channels. This interlock trip is indicated on the front panel LEDs, but not by any of the SCPI status functions.

Explicitly, on one particular v1.1 Booster, channels 1-4 and 7 are always tripped (amber LED).

The SCPI interface does not report any problem:
```
INT:STAT? 1
0
INT:ERR? 1
0
INT:OVER? 1
0
```

Reseting the interlock by `INT:CLEAR 1`, I see the channel status LED flick to green, then back to amber, and I see a message on the UART:
```
[ERROR] Reverse power interlock tripped on channel 1, tx 0.04 rfl 0.04
```
However all the SCPI status commands still indicate that there is no problem.

I am using firmware version "RFPA v1.20, built Nov 19 2018 13:43:11, for hardware revision: v1.10"

wizath, 2018-12-05:
Could you also post output, where you type 'start' into CLI ?

cjbe, 2018-12-05:
```
PGOOD: 1                                                                                                                              │································································
FAN SPEED: 0 %                                                                                                                        │································································
AVG TEMP: 32.38 CURRENT: 33.25                                                                                                        │································································
CHANNELS INFO                                                                                                                         │································································
==============================================================================                                                        │································································
                #0      #1      #2      #3      #4      #5      #6      #7                                                            │································································
DETECTED        1       1       1       1       1       1       1       1                                                             │································································
HWID            7C:60   42:15   5D:41   57:00   2A:E4   61:98   3A:15   8D:F0                                                         │································································
TXPWR [V]       0.07    0.04    0.04    0.04    0.05    0.11    0.06    0.05                                                          │································································
RFLPWR [V]      0.04    0.04    0.05    0.05    0.06    0.08    0.04    0.05                                                          │································································
TXPWR [dB]      -2.65   0.04    -1.14   -1.04   -0.50   -2.49   -1.87   -2.00                                                         │································································
RFLPWR [dB]     0.08    0.02    -0.04   -0.35   -2.13   -1.04   0.78    -0.62                                                         │································································
I30V [A]        0.128   0.219   0.150   0.235   0.211   0.159   0.128   0.152                                                         │································································
I6V0 [A]        0.247   0.245   0.237   0.237   0.246   0.250   0.249   0.247                                                         │································································
IN8V0 [mA]      0.013   0.013   0.016   0.010   0.029   0.005   0.005   0.013                                                         │································································
5V0MP [V]       4.811   4.831   4.843   4.834   4.759   4.789   4.861   4.774                                                         │································································
ON              1       1       1       1       1       1       1       1                                                             │································································
SON             1       0       0       0       0       1       1       0                                                             │································································
IINT            0       0       0       0       0       0       0       0                                                             │································································
OINT            0       0       0       0       0       0       0       0                                                             │································································
SINT            0       1       1       1       1       0       0       1                                                             │································································
OVC             0       0       0       0       0       0       0       0                                                             │································································
ADC1            119     67      70      71      76      181     105     77                                                            │································································
ADC2            68      70      74      80      102     128     71      75                                                            │································································
DAC1            92      94      83      93      95      1500    82      90                                                            │································································
DAC2            250     244     240     261     148     238     253     294                                                           │································································
SCALE1          52      57      56      54      54      49      53      53                                                            │································································
[... cut ...]
HWIO            -13.47  -9.38   -11.14  -20.36  -11.47  -11.65  -17.18  -15.60                                                        │································································
LTEMP           27.00   30.00   30.50   31.50   31.00   31.00   26.00   30.25                                                         │································································
RTEMP           28.50   31.00   31.00   33.25   33.00   33.25   28.50   31.25                                                         │································································
============================================================================== 
```

wizath, 2018-12-05:
This seems like reverse power interlock acting too early while enabling channel. Will post fixed version tomorrow.

cjbe, 2018-12-06:
@wizath any update?

wizath, 2018-12-06:
[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/2655006/eth-scpi.dfu.zip)
Temporarily disabled reverse power interlock. Sorry for delay, I'm a little bit sick.

cjbe, 2018-12-06:
@wizath thanks - I will test this tomorrow.

hartytp, 2019-06-04:
was this fixed by https://github.com/sinara-hw/booster-firmware/commit/340d759e42bf00008eed65d01684ee634d7306a0 ?

hartytp, 2019-06-05:
@wizath ping, is this fixed?

wizath, 2019-06-05:
1. Set fixed value for reverse power e.g. 20 dBm to cut off the channel
2. Set cutoff for difference between TX and RFL power

I'm wondering which one will be better for this task. I'll set a fixed one for now if reverse power will exceed 20 dBm.

hartytp, 2019-06-05:
@wizath good question.

My understanding is that the sole purpose of this interlock is to protect Booster from damage with reflective loads. So the right interlock depends on what the damage mechanism is. AFAICT, damage only occurs because of the total magnitude of the reflected power, not the fraction of reflected power. e.g. this is why all amps can take max reflections with low input power.

So, I think that (1) is appropriate -- set a fixed threshold. Pick a threshold number that is high enough to avoid damage, without making Booster unduly sensitive to load S11. Can we safely go a bit higher than 20dBm? That means that at max power (~37.5dBm) you need a 17.5dB return loss, which seems a bit high. I'd have guessed that +26dBm should be safe, but maybe @gkasprow can advise?

wizath, 2019-06-05:
Ok, I'll leave it at 26 dBm until we decide

hartytp, 2019-06-05:
@wizath apart from finalizing the threshold value, is this fixed?

wizath, 2019-06-05:
Yep

closed 2019-06-05 by hartytp