sinara-hw/Booster #122: Multiple TCP connections are not allowed

cjbe, 2018-12-05:
Currently only one control connection to port 5000 is allowed. When a connection is already open, subsequent connection attempts are refused.

I would expect multiple connections to be allowed - if this is not possible it should be documented.

cjbe, 2019-02-06:
@wizath would you be able to support more than 1 connection? A fixed maximum of 8 would be fine.

All my Booster seem to refuse connections after they have been on for a while - it is unclear if this is something leaving a connection open, or a bug in the firmware. Once this is implemented we can debug this issue better.

wizath, 2019-02-06:
You mean they drop connection or refuse to accept it after idling for a while? Maximum clients I can get is 7 because one socket is reserved for DHCP.

cjbe, 2019-02-06:
My observation was that if I had one connection active, I could not make another (I got connection refused). After I closed the first connection, I could then make another.

At the moment all 3 of my Booster are giving "connection refused" errors trying to connect, so I cannot debug things properly (we are running experiments, so I cannot reboot the devices).

wizath, 2019-02-06:
Yes, that's because only one socket is supported at the moment. I'll implement multiple connection support after we get rid of more concerning issues.

cjbe, 2019-02-06:
Aha! That sounds like a good plan.

hartytp, 2019-02-18:
@wizath how is work on the firmware going? What's the ETA for a version with all known issues fixed? It seems like we're really close to finishing this project off...

wizath, 2019-02-18:
I'm waiting for new hardware version to be produced this week, so I can test all things out (including new bias voltage selection) since only board I have is the revision v1.0 and zero working channels.

hartytp, 2019-02-18:
> since only board I have is the revision v1.0 and zero working channels.

:(

> I'm waiting for new hardware version to be produced this week, so I can test all things out (including new bias voltage selection)

aah, okay, that makes sense. Sounds like a good plan.

cjbe, 2019-03-26:
@wizath it would be really useful to have this feature implemented.

wizath, 2019-03-26:
Well, It's not quite easy to implement multiple connections on the same port according to WIZnet forums. I'll try to figure it out this week

wizath, 2019-03-28:
Ok, got first positive results, will post working version soon

wizath, 2019-03-29:
https://github.com/sinara-hw/Booster/issues/98#issuecomment-478037725 Added support for multiple (4) connections. Please check if this suits your needs. If yes, then close

closed 2019-04-09 by hartytp

hartytp, 2019-04-10:
Confirmed, I can simultaneously use two TCP connections. Thanks!