sinara-hw/Booster #123: bias cal issue

cjbe, 2018-12-10:
@wizath for the Booster channel that was just repaired and returned to us I'm seeing

```
 > status 3
[status] e=0 s=0 r1=69 r2=78 tx=-1.074 rf=-0.392 curr=0.059
```

This is with firmware version
```
> version
RFPA v1.20, built Dec  6 2018 22:39:28, for hardware revision: v1.10
```

So the bias current does not seem to be correctly. The bias current calibration routine seems to be different to the previous firmware. I see

```
> biascal 3 160
[*] 036% [###       ] 58/160
[*] 036% [###       ] 58/160
[*] 036% [###       ] 58/160
[*] 036% [###       ] 58/160
[*] 036% [###       ] 58/160
[*] 036% [###       ] 58/160
[*] 036% [###       ] 59/160
[*] 036% [###       ] 59/160
[*] 036% [###       ] 59/160
[*] 036% [###       ] 59/160
[*] 036% [###       ] 59/160
[*] 036% [###       ] 59/160
[*] 036% [###       ] 59/160
[*] 036% [###       ] 59/160
[*] 036% [###       ] 59/160
[*] 036% [###       ] 59/160
[*] 036% [###       ] 59/160
[*] 035% [###       ] 57/160
[*] 036% [###       ] 59/160
[*] 036% [###       ] 59/160
[*] 036% [###       ] 59/160
[biascal] error, current too low 59 < 160
```

Does this mean I need to adjust the trim pot, or is it a firmware problem?

wizath, 2018-12-10:
You should adjust the trimpot. Biggest issue (that is being addressed in next HW version) is that there is "slight" variation of bias current between mainboard channels, so we can't use one calibration for module without target mainboard.

hartytp, 2018-12-10:
Thanks. Done that but now I'm having trouble getting the supplemental control board back. @gkasprow those SMT board-board connectors are pretty flimsy. AFAICT on this Booster channel the board-board connectors were soldered on a bit crooked so it's now very difficult to get the supplemental board back on. Can we use a better solution for this next time we change the design??

cjbe, 2018-12-10:
Tweaking the pot and reassembling, the biascal scan looks unchanged. 

Upon taking the lid of a (working) neighbouring channel, this channel then also stopped working (bias current went from 160mA -> 57mA and biascal does nothing). Replacing the lid and rerunning a biascal, the maximum achieved current was 130mA.

A few minutes later, all the channels seem to be misbehaving; the I30V current is ~60 mA for all channels:
```
PGOOD: 1                                                                                                                                                                                                                                     │·································
FAN SPEED: 0 %                                                                                                                                                                                                                               │·································
AVG TEMP: 28.18 CURRENT: 28.00                                                                                                                                                                                                               │·································
CHANNELS INFO                                                                                                                                                                                                                                │·································
==============================================================================                                                                                                                                                               │·································
                #0      #1      #2      #3      #4      #5      #6      #7                                                                                                                                                                   │·································
DETECTED        1       1       1       1       1       1       1       1                                                                                                                                                                    │·································
HWID            7C:60   42:15   5D:41   57:00   2A:E4   61:98   3A:15   8D:F0                                                                                                                                                                │·································
TXPWR [V]       0.07    0.04    0.04    0.40    0.04    0.11    0.06    0.04                                                                                                                                                                 │·································
RFLPWR [V]      0.04    0.04    0.04    0.04    0.05    0.07    0.04    0.04                                                                                                                                                                 │·································
TXPWR [dB]      -2.73   -0.09   -1.25   9.83    -0.65   -2.65   -1.98   -2.11                                                                                                                                                                │·································
RFLPWR [dB]     0.00    -0.12   -0.16   11.16   -2.30   -1.22   0.65    -0.74                                                                                                                                                                │·································
I30V [A]        0.077   0.073   0.053   0.057   0.068   0.096   0.072   0.067                                                                                                                                                                │·································
I6V0 [A]        0.037   0.036   0.032   0.032   0.035   0.001   0.023   0.021                                                                                                                                                                │·································
IN8V0 [mA]      0.857   0.808   0.829   0.371   0.868   0.566   0.756   0.857                                                                                                                                                                │·································
5V0MP [V]       4.588   4.637   4.675   4.641   4.446   4.521   4.696   4.527                                                                                                                                                                │·································
ON              0       0       0       0       0       0       0       0                                                                                                                                                                    │·································
SON             0       0       0       0       0       0       0       0                                                                                                                                                                    │·································
[... cut ...]
LTEMP           26.50   26.00   27.25   26.00   26.00   27.00   26.00   28.50                                                                                                                                                                │·································
RTEMP           25.00   26.50   25.00   26.00   27.25   28.00   27.25   26.00                                                                                                                                                                │·································
============================================================================== 
```

hartytp, 2018-12-10:
Oh dear! Looks like another mechanical issue.

When working on Booster before, I found that the mechanical tolerances are so tight that putting the lids on can place a lot of mechanical strain on components, causing issues.

hartytp, 2018-12-11:
After power cycling, the status of all channels is
```
PGOOD: 1
FAN SPEED: 0 %
AVG TEMP: 28.76 CURRENT: 29.00
CHANNELS INFO
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	1	1	1	1	1	
HWID		7C:60	42:15	5D:41	57:00	2A:E4	61:98	3A:15	8D:F0	
TXPWR [V]	0.07	0.04	0.04	0.41	0.05	0.11	0.06	0.05	
RFLPWR [V]	0.04	0.04	0.04	0.05	0.05	0.07	0.04	0.05	
TXPWR [dB]	-2.65	0.02	-1.16	10.00	-0.52	-2.59	-1.91	-2.04	
RFLPWR [dB]	0.08	0.00	-0.06	11.33	-2.15	-1.16	0.73	-0.66	
I30V [A]	0.153	0.148	0.148	0.059	0.175	0.151	0.128	0.139	
I6V0 [A]	0.247	0.245	0.237	0.237	0.245	0.249	0.249	0.247	
IN8V0 [mA]	0.010	0.010	0.016	0.008	0.029	0.005	0.005	0.013	
5V0MP [V]	4.818	4.850	4.864	4.831	4.774	4.791	4.864	4.794	
ON		1	1	1	1	1	1	1	1	
SON		1	1	1	1	1	1	1	1	
IINT		0	0	0	0	0	0	0	0	
OINT		0	0	0	0	0	0	0	0	
SINT		0	0	0	0	0	0	0	0	
OVC		0	0	0	0	0	0	0	0	
ADC1		120	65	69	667	73	176	103	75	
ADC2		68	69	72	78	87	120	70	74	
DAC1		92	94	83	93	95	1500	82	90	
DAC2		250	244	240	261	148	238	253	109	
SCALE1		52	57	56	54	54	49	53	53	
OFFSET1		258	65	134	127	102	303	204	183	
BIASCAL		3870	2130	285	2475	1395	920	3570	540	
HWIS		8.72	8.42	9.36	10.22	9.36	8.38	9.67	9.41	
HWIO		-11.78	-9.93	-14.53	-20.36	-14.53	-9.27	-16.42	-15.60	
LTEMP		26.25	27.00	27.50	24.00	26.00	28.50	25.50	26.50	
RTEMP		27.50	26.00	27.00	24.00	29.00	28.00	25.50	27.50	
==============================================================================
```

Channel 3 has 0V on the 29V rail, measured on the supplemental control board.

hartytp, 2018-12-11:
Putting all channels in standby, I see

```
PGOOD: 1
FAN SPEED: 0 %
AVG TEMP: 28.86 CURRENT: 26.50
CHANNELS INFO
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	1	1	1	1	1	
HWID		7C:60	42:15	5D:41	57:00	2A:E4	61:98	3A:15	8D:F0	
TXPWR [V]	0.07	0.04	0.04	0.40	0.04	0.11	0.06	0.04	
RFLPWR [V]	0.04	0.04	0.04	0.04	0.05	0.07	0.04	0.04	
TXPWR [dB]	-2.79	-0.07	-1.25	9.80	-0.65	-2.61	-2.02	-2.13	
RFLPWR [dB]	-0.06	-0.10	-0.16	11.12	-2.30	-1.18	0.61	-0.76	
I30V [A]	0.078	0.075	0.054	0.059	0.067	0.096	0.072	0.068	
I6V0 [A]	0.038	0.035	0.033	0.025	0.035	0.001	0.024	0.021	
IN8V0 [mA]	0.868	0.816	0.795	0.382	0.881	0.556	0.759	0.865	
5V0MP [V]	4.571	4.670	4.675	4.583	4.438	4.521	4.712	4.515	
ON		0	0	0	0	0	0	0	0	
SON		0	0	0	0	0	0	0	0	
IINT		0	0	0	0	0	0	0	0	
OINT		0	0	0	0	0	0	0	0	
SINT		0	0	0	0	0	0	0	0	
OVC		0	0	0	0	0	0	0	0	
ADC1		114	61	64	656	67	175	97	70	
ADC2		64	64	67	72	83	118	65	70	
DAC1		92	94	83	93	95	1500	82	90	
DAC2		250	244	240	261	148	238	253	109	
SCALE1		52	57	56	54	54	49	53	53	
OFFSET1		258	65	134	127	102	303	204	183	
BIASCAL		3870	2130	285	2475	1395	920	3570	540	
HWIS		8.72	8.42	9.36	10.22	9.36	8.38	9.67	9.41	
HWIO		-11.78	-9.93	-14.53	-20.36	-14.53	-9.27	-16.42	-15.60	
LTEMP		26.00	27.25	26.50	25.00	27.50	28.50	26.25	27.25	
RTEMP		26.50	25.00	26.50	25.25	26.00	26.00	24.00	25.00	
==============================================================================
```

Why are the currents so high when all modules are in standby? Shouldn't the bias current go down to 0? Or, is this just the leakage through the FET even with the bias cut off?

hartytp, 2018-12-11:
@gkasprow what could cause the 29V rail to go to 0V for this channel? I measure less than 1mV on the supplemental control board pin.

hartytp, 2018-12-11:
Putting channel 4 (which works correctly) into the channel 3 slot, I now see

```
PGOOD: 1
FAN SPEED: 0 %
AVG TEMP: 29.65 CURRENT: 28.50
CHANNELS INFO
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	1	0	1	1	1	
HWID		7C:60	42:15	5D:41	2A:E4	00:00	61:98	3A:15	8D:F0	
TXPWR [V]	0.07	0.04	0.04	0.04	0.39	0.11	0.06	0.05	
RFLPWR [V]	0.04	0.04	0.04	0.05	0.59	0.08	0.04	0.05	
TXPWR [dB]	-2.65	0.00	-1.16	-0.59	0.00	-2.53	-1.87	-2.04	
RFLPWR [dB]	0.08	-0.02	-0.06	-2.24	0.00	-1.09	0.78	-0.66	
I30V [A]	0.149	0.145	0.143	0.059	0.000	0.160	0.140	0.135	
I6V0 [A]	0.247	0.245	0.237	0.250	0.000	0.249	0.249	0.247	
IN8V0 [mA]	0.010	0.010	0.013	0.008	0.000	0.005	0.003	0.010	
5V0MP [V]	4.824	4.852	4.867	4.838	0.000	4.799	4.867	4.795	
ON		1	1	1	1	0	1	1	1	
SON		1	1	1	1	0	1	1	1	
IINT		0	0	0	0	0	0	0	0	
OINT		0	0	0	0	0	0	0	0	
SINT		0	0	0	0	0	0	0	0	
OVC		0	0	0	0	0	0	0	0	
ADC1		120	65	69	70	630	179	105	75	
ADC2		67	68	72	88	964	124	70	74	
DAC1		92	94	83	95	0	1500	82	90	
DAC2		250	244	240	148	0	238	253	109	
SCALE1		52	57	56	54	0	49	53	53	
OFFSET1		258	65	134	102	0	303	204	183	
BIASCAL		3870	2130	285	1395	0	920	3570	540	
HWIS		8.72	8.42	9.36	9.36	0.00	8.38	9.67	9.41	
HWIO		-11.78	-9.93	-14.53	-14.53	0.00	-9.27	-16.42	-15.60	
LTEMP		27.50	28.50	27.00	27.50	0.00	28.50	26.00	27.00	
RTEMP		27.25	26.00	28.50	26.00	0.00	28.25	27.00	27.00	
==============================================================================
```

Again, I see 0.00V on the 29V rail. So, it seems that an issue has developed with the main board.

hartytp, 2018-12-11:
@gkasprow any suggestions about how I should debug this?

wizath, 2018-12-11:
You can check if the LED on the main board lights up and measure voltage on (big) transistor T1 near module connector on main board. The LDO could be dead if it's not connection issue.

hartytp, 2018-12-11:
@wizath not trivial to do that AFAICT because I can't see the LED when the module is plugged in, but if the module is not detected then the LDO is disabled.

hartytp, 2018-12-11:
Also, which LED? There are a few LEDs per channel.

hartytp, 2018-12-11:
FWIW, there are no visible signs of magic smoke having escaped so I don't think there has been a catastrophic failure here...

wizath, 2018-12-11:
LED should be right beside this big transistor.

hartytp, 2018-12-11:
@wizath okay, that one (LD1). Fine, but what's the best way of enabling the channel without the module plugged in?  AFAICT, plugging in the supplemental control board only is not enough.

wizath, 2018-12-11:
use `enable` with proper bit shifted number of channel (eg. 32 for 5th channel). It ignores detected channel masking. I left it like that for such situations.

hartytp, 2018-12-11:
@wizath nice!

With `enable 8` and looking at channel 3, I see the green LED next to the 5V LDO turn on, but not the LED next to the power BJT.

Interestingly, when I run `enable 16` the LED near the power BJT turns on very dimly for channel 3 (it turns on at full brightness for channel 4 as expected). Running `enable 8` turns the channel 3 LED off completely. Any idea why this happens?

Anyway, the conclusion here seems to be that the main board is not functioning correctly.

hartytp, 2018-12-11:
With channel 3 enabled, I see:
- 1 V on the power BJT collector
- (assuming I've got the correct pin out for the BJT) I see 0.6V on the base and 1V on the emitter
- 30V on pin 1 of the LDO

hartytp, 2018-12-11:
@gkasprow @wizath let me know if you can think of anything else I should measure

hartytp, 2018-12-11:
@gkasprow I spoke to @gkasprow about this briefly today. The situation seems to be this:
- the version of Booster supplied by Technosystem does not limit the 29V current to a low enough value
- as a result, anything that causes excessive power dissipation on the 29V rail can cause the 29V LDO to die (specifically, the power BJT that's used as an external pass element goes into thermal run away)
- that seems to be what has happened on this channel

So, the question is what caused excessive current consumption? Did this channel ever work? Did it die after running the bias cal routine? 

Also, are you sure about that observation that the 29V current changes as you remove the lids of modules? I had a quick play with the booster unit but didn't see that effect.

cjbe, 2018-12-11:
@hartytp yes, I am sure about the observation of the change in current when removing the lid.

I ran bias cals on ch3 (broken) and ch2 (working) several times, to compare the behaviour. I then removed the lid on ch2 to look at voltages on the test points for a working channel, and observed that the bias cal output had changed.

wizath, 2018-12-12:
> @wizath does this version of the firmware have software current limiting detection? @gkasprow thought that the LDO should only die if it is run at max current for several minutes, but shouldn't the software prevent this from happening?

Protection is enabled when channel is detected and running.

hartytp, 2018-12-12:
@gkasprow @wizath the situation with this board is as follows:
- it had been working reliably for some time. All channels had the bias currents calibrated and were running correctly at 160mA
- at some point one channel stopped working (we do not know exactly when this happened). When we noticed that, we ran the `start` command to see the status of all channels. The bias currents now varied from approx 130mA to 170mA for all channels apart from channel 3, which had 60mA.
- we tried to recalibrate channel 3, but could not increase the bias current (even when we adjusted the potentiometer), after removing the module we discovered that the 29V LDO was dead

So, the question is what is wrong, and what caused this LDO to die.

hartytp, 2018-12-12:
Given the fluctuations in bias current observed by @cjbe I suspect that this issue is related to 
https://github.com/sinara-hw/Booster/issues/115#issue-373442453

There I observed fluctuations in the gain of 1 channel by 10dB. If the bias current is changing by a factor of 2, that could easily cause a huge change in the large-signal gain, so maybe this is all one issue.

cjbe, 2018-12-12:
@gkasprow @wizath I looked at this somewhat closer, and don't understand what is happening; There seems to be something chronic going on.

In all these measurements, ch3 and 4 were removed, and ch2 was misbehaving grossly. All plots are of the reported current on the 28V rail from each channel, and the time axis is roughly in seconds. During these measurements I checked that the reported current was in agreement with the measured voltage across the current shunt, and saw agreement.

First, I turned the unit on, and ch2 ran at normal current (~140mA). After poking the channel lightly the current dropped slowly by a factor 2, and stayed there until powering the device off then on again. Both before and after this current drop, the OUT_BIAS pin on the module stayed at -1.663V, and the 29V rail on the module stayed at ~28.95V.
[image]

After powering off then on again, this current dropout occurs repeatedly without touching the channel. Looking at the measured voltage across the current shunt (which gives a very precise measure of current) one sees the current slow dropping over time, then rapidly accelerating down, then jumping back to nominal. There are also occasional glitches on other channels.
[image]

cjbe, 2018-12-12:
@gkasprow it is completely unclear to me what is causing this.  How do you suggest we try and debug this?

wizath, 2018-12-13:
Could you measure if the OUT_BIAS voltage is stable? Since it is directly responsible for setting that current value.

hartytp, 2018-12-13:
@wizath we did the following:
- connect a high-resolution voltmeter across the sense resistor to measure the 29V current directly
- monitored the reported 29V current from the firmware and checked this agreed with our direct current measurement
- measured the V_BIAS and 29V supply on the supplemental control board
- we observed no variation in the V_BIAS or 29V at the 1mV level, but we saw large changes in the 29V current. So, there was no variation in V_BIAS on the supplemental control board
- this does not rule out an issue like a dodgy pot leading to fluctuations in V_BIAS at the FET
- this also does not rule out any fast oscillations that would not be seen by the DMM.
- one thing I do not understand is that normally a bad connection would lead to V_BIAS dropping, which would increase the bias current, which is the opposite of our observation that the bias current dropped
- also, if the FET were oscillating then I would have expected the bias current to increase as well
- the only thing I can think of is a damaged FET that has a metastable ON resistance, but this seems somewhat unlikely

hartytp, 2019-01-08:
@gkasprow looking at one of our other boosters, I see:

```
PGOOD: 1
FAN SPEED: 30 %
AVG TEMP: 38.00 CURRENT: 37.25
CHANNELS INFO
==============================================================================
                #0      #1      #2      #3      #4      #5      #6      #7
DETECTED        1       1       1       1       1       1       1       1
HWID            60:8B   CE:DC   9E:60   E6:4F   91:0D   B0:96   AF:03   D4:60
TXPWR [V]       0.04    0.04    0.04    0.04    0.05    0.04    0.04    0.04
RFLPWR [V]      0.04    0.04    0.04    0.04    0.04    0.04    0.04    0.14

TXPWR [dB]      -0.93   -0.05   -0.05   -0.13   -2.56   -1.02   -0.58   -0.42
RFLPWR [dB]     -2.55   -0.81   -1.02   -1.31   -3.24   -0.67   -4.13   -6.60
I30V [A]        0.153   0.154   0.150   0.172   0.057   0.136   0.136   0.167
I6V0 [A]        0.031   0.031   0.030   0.034   0.011   0.027   0.027   0.034
IN8V0 [mA]      0.655   0.652   0.639   0.733   0.242   0.574   0.574   0.712
ON              1       1       1       1       1       1       1       1
SON             1       1       1       1       1       1       1       1
IINT            0       0       0       0       0       0       0       0
OINT            0       0       0       0       0       0       0       0
SINT            0       0       0       0       0       0       0       0
OVC             0       0       0       0       0       0       0       0
ADC1            67      73      69      71      82      70      73      73
ADC2            71      70      70      71      69      71      72      223
DAC1            84      96      90      85      89      90      89      1500
DAC2            239     244     229     224     244     254     224     173
SCALE1          55      57      55      54      52      54      55      57
OFFSET1         118     76      72      78      215     125     105     97
BIASCAL         3810    2490    2115    1800    2445    3375    2160    1920
HWIS            9.65    10.28   9.01    9.13    10.01   9.57    8.27    10.21
HWIO            -16.75  -20.23  -13.02  -13.44  -18.73  -16.63  -8.91   -19.74
LTEMP           33.00   34.25   33.00   35.00   33.00   32.00   35.50   35.50
RTEMP           34.25   35.25   33.00   35.00   33.00   34.25   36.25   36.25
==============================================================================
```

The scatter on the bias currents here is very concerning. When I calibrated this booster some time ago, all bias currents were 160mA +-2mA. Now, channel 4 is showing a very low bias current (dead MOSFET?). Even discounting that, the other channels show a large range in bias current.

The device has only changed temperature by 15C since then, so I would not expect such extreme variations in current.

@gkasprow do you have any ideas here? This is causing us real problems, so we need to address it asap!

The most likely causes I can think of would be: damage to the MOSFETs (but, how?); parasitic oscillations (although I can't see how this could lower the bias current); or some mechanical issue like a problem with the biasing potentiometer.

The thing that complicates this all is that the InAmps are working out of specification, so it's hard to know how much we can trust these readings anyway.

hartytp, 2019-01-08:
@gkasprow on channel 4 (low bias current) I measure a gain of 33.6dB (should be 40dB).

We previously observed the gain on some channels fluctuating by 10dB, this seems the be the same issue as the bias current.

hartytp, 2019-01-08:
@gkasprow ping...

gkasprow, 2019-01-08:
The thing that comes to my mind is offset setting range.
If we are working within small fraction of DAC range, offset drift of DAC, MOSFET or 3V supply may cause such issue

hartytp, 2019-01-08:
I considered that. We performed the following measurement:
- measure get gate bias voltage on supplemental control pcb with dvm 
- measure fet bias current using high resolution dmm to make a differential measurement of the voltage on the sense resistor
- we observed a factor of two change in the bias current with no change in the bias voltage at the 1mV level 
- so if there is a fluctuation in the gate voltage then it must be on the rf pcb e.g due to a mechanical problem with the pot. Possible but seems unlikely given that we see similar things on many units

hartytp, 2019-01-08:
Yes, that would explain everything.

Although, I find it a little surprising that we see this issue on so many channels. But, it's possible!

One possibly related observation: on the channel with the destroyed 28V LDO, the supplemental control board was quite difficult to insert and remove. Inspecting the module, it's clear that the headers on the supplemental control board were not soldered in straight, also the sockets on the RF module are not straight ether. This leads to a lot of mechanical stress when the supplemental control board is inserted. When I removed the supplemental control board, two pins came off the header on the RF module (these need resoldering before the module can be used).

hartytp, 2019-01-08:
When you receive the Booster (should be Friday morning), the first thing to do is probably to monitor the bias current (e.g. using the VCP diagnostics) while also recording the bias voltage at the FET pin. This needs to be done with the RF board connected to the supplemental control board/main PCB in the normal way to ensure that everything is mechanically the same as normal operation.

If you see fluctuations in the bias current without changes in the bias voltage then you know this isn't a mechanical issue. If the voltage fluctuates as well then we've found the cause.

If the issue isn't to do with the bias voltage then the next thing to do is probably to sniff around and see if we can find any oscillations.

gkasprow, 2019-01-08:
I will devote the whole Monday to debugging these issues.

hartytp, 2019-01-08:
Thanks Greg!

hartytp, 2019-01-08:
> poor mechanical contact of gate voltage would explain both drifts and occasional MOSFET damage.

Thinking about this more, I'm not sure that's true. Look at the data we have: https://user-images.githubusercontent.com/13227288/49899411-8c09e700-fe53-11e8-8cc2-df79d9b5a508.png

See how almost all of the fluctuations involve the bias current *decreasing*. However, if the gate voltage came disconnected the bias current would increase. 

I guess that a mechanical issue that caused the gate voltage to disappear, so that the bias current became very high might kill the FET. A dead FET might behave like this (although, it's still quite odd behaviour).

Maybe a dodgy pot could also cause a problem, but again it's surprising that the current always decreases.

hartytp, 2019-01-08:
Anyway, it's worth double checking the bias voltage on a few channels to rule out mechanical issues so we can focus on other things.

hartytp, 2019-01-14:
@cjbe Which channel is which in this plot? https://user-images.githubusercontent.com/13227288/49899397-86140600-fe53-11e8-9327-162ddb553bdb.png

Do you remember which channel we were looking at the bias current on?

gkasprow, 2019-01-15:
So the plan for tomorrow for me and @wizath :
- assemble back all 8 channels. Attach the main board as it was
- apply mechanical Dirac Delta signal :) and observe bias fluctuation
- attach 8 scopes to all channel outputs, activate channels and look for any activity on the outputs
- if a certain channel is faulty, remove the lid and tap the trimpot gently to see how it affects bias current
- it no issues are observed, do as above.

hartytp, 2019-01-15:
> @hartytp do you think there is a correlation of the issue with the fact that some modules have trimpot installed? Do some older revisions have such a problem?

I thought all versions had the trim pot. Was it not present on the v1.0 unit we have (the one with the silver chassis). @cjbe do you have any data for that Booster?

hartytp, 2019-01-15:
> I replaced all 28V Isense resistors with 0.1R||1R which gives 91mR. Then I added pullup to rev ID signals that now indicate 1. Other boards that were not modified have ID =0.
We observed 2 suspected channels but so far no current variation. I think that may cause an issue is that the booster was disassembled and not completely assembled. It looks to me like a mechanical problem which may be triggered by some vibration.
> Jakub replaced the LDO transistor because it was broken

Let me check I understand what you've done so far:
- remove all modules and the main board from the chassis
- change the HW revision pin straps on the main board
- change the BJT for one channel (but not the LDO itself?). Which channel was this?
- change the current sense resistors for the 28V rail (but not the 5V rail yet)
- replace only  2 of the RF modules in the chassis (which channels?), being careful to put the channels in the same slot that they were in before
- re-run the bias current calibration routine
- look for glitches in the 29V bias current by probing the ADC input with a scope probe

Is that correct? Is there anything missing from that list?

hartytp, 2019-01-15:
> So the plan for tomorrow for me and @wizath :

That sounds good.

I'd add to that:
- run @cjbe's script to monitor the bias current on all channels for 10 min or so. See if there are any drop outs like the ones we posted above
- try power cycling it/running the bias cal a few times
- double check the gain on all channels is still 40dB (we measure 30dB sometimes)

hartytp, 2019-01-15:
The only thing I can think of is some mechanical issue either due to thermal stress or to the D-type putting pressure on the PCB.

Maybe also some kind of grounding issue, but I can't think of what. It's a real mystery!

gkasprow, 2019-01-15:
I'll add to the list - check 5V rail.

hartytp, 2019-01-15:
Thanks. Either the voltage on the 5V is way out of spec or the ADV readings are very inaccurate. Either way something is not right

gkasprow, 2019-01-15:
In the case of trimpot , there are 3 contacts that can be broken. But true, the most probable one is wiper. And it would lead to current increase and LDO breakdown.

hartytp, 2019-01-15:
There are also the adc glitches to look at but that’s probably less important. First thing is to reproduce the bias current issue.

cjbe, 2019-01-15:
FYI the script I was using to dump the logger current vs time traces is https://hastebin.com/gibopovuta.py
Note that this only samples at ~1 Hz since the `status` command takes a while to run.
To plot the results I used https://hastebin.com/oqacatejeg.js

hartytp, 2019-01-16:
@gkasprow were you able to reproduce this after reassembling the unit?

hartytp, 2019-01-16:
> I replaced all 28V Isense resistors with 0.1R||1R which gives 91mR. 

What about the 6V rail sense resistor? Doesn't that need replacing as well?

gkasprow, 2019-01-16:
We have 500mA range and 200-250mA readout. We can change it but IMHO it is not critical.

closed 2019-01-30 by hartytp