sinara-hw/Booster #124: Too much solder on main board T1

hartytp, 2018-12-11:
The power BJT used for the 29V LDO has too much solder. It's hard to see from this photo, but all around it there are large blobs of solder which come very close to shorting against near by components. 

[image]

gkasprow, 2018-12-12:
It looks like it got so hot that melt the solder joint. Previous release of the main board did not have reduced current limit. When over-current happens, the LDO enters foldback mode, but it still dissipates enough power to melt the tin after a few minutes. Then the BJT gets breakdown and damages the regulator chip and probably power stage.
This could also happen when power stage was replaced without proper calibration or during calibration routine when software over-current protection is disabled. 
With new board release it was fixed.

hartytp, 2018-12-12:
Okay, thanks Greg.

closed 2018-12-12 by hartytp