sinara-hw/Booster #125: Missing output capacitor on 29V LDO

hartytp, 2018-12-11:
@gkasprow not sure how I missed this in previous design reviews, but why is there no output capacitor on the 29V LDO??? I know there is some capacitance on the RF modules and on the supplemental control board, but shouldn't there be 100nF || 10uF right by the power FET? The datasheet indicates that this is a requirement.

```
In the fixed 5 volt version, the only external parts
required are an external pass element, an output capacitor, and a com-
pensation capacitor. On the adjustable version the output voltage can
be set anywhere from 2.5V to 35V with two external resistors.
```

gkasprow, 2018-12-12:
There is already 1uF capacitor after the sense resistor. It is old-style LDO that wouldn't work with large ceramic capacitors connected directly. In our case we have 1uF/20mR equivalent capacitor. On RF board we have another 1uF and 47uF tantalium with 0.1R series FB resistance so it should be fine.

hartytp, 2018-12-12:
Okay, if you don't think this is an issue then it's fine by me.

My concern was because the LDO data sheet recommends 10uF. 

I wonder what the capacitance of the 1uF capacitor is once it's been biassed to 29V. Even if it's a 50V cap, it could end up being quite low.

True there is a 47uF capacitor in the RF modules, but that's quite a long way from the LDO so quite a bit of ESL/ESR is added by the traces.

gkasprow, 2018-12-12:
Anyway they recommend 10uF electrolytic capacitor. At that time X7R caps simply didn't exist:)

gkasprow, 2019-05-26:
added 10u tantalum.

closed 2019-05-26 by gkasprow