sinara-hw/Booster #126: InAmps operating out of specification

hartytp, 2018-12-12:
@gkasprow unless I'm misunderstanding things, there seems to be a major issue with the InAmps used to measure currents on Booster. According to the [data sheet](http://www.ti.com/lit/ds/symlink/ina196a-q1.pdf) the InAmps are only specified to operate with a differential input of >20mV, equivalent to 1A across the 20mOhm sense resistor.

Bearing in mind that the maximum current we want to measure is <200mA, this is an issue.

This is potentially a major issue, since these InAmps are used to calibrate the FET bias current, so any errors here could lead to damage of the amplifier modules.

hartytp, 2018-12-12:
This also explains why we measure 60mA of 28V current even when the amplifier modules are in standby mode (we used a high-precision multimeter to measure the differential voltage across the sense resistor and confirmed that it was <<1mV, even thought the firmware reported 50mV).

hartytp, 2018-12-12:
I wonder if this explains some of our exploding FETs.

hartytp, 2018-12-12:
aaah, as @cjbe points out, this might well also explain why each amplifier has to be calibrated in the slot is used in. The issue may not be variations in the biasing, but rather errors in the current measurements.

hartytp, 2018-12-12:
See section 7.4.2, which explains that these InAmps do not function as specified when Vsense is below 20mV

hartytp, 2018-12-12:
7.4.2.3 describes the conditions we operate under (Low Vsense Case 3)

gkasprow, 2018-12-12:
True, in addition we use half of the ADC range. WE should increase resistor values in currently produced revision.

gkasprow, 2018-12-12:
We should switch to INA195A or INA198A that would utilise nearly full ADC range + increase resistors.

hartytp, 2018-12-12:
If we increase the sense resistor, should we put it inside the LDO FB loop to mitigate the voltage drop?

In the long run, we should use an InAmp that is designed for lower sense voltages.

gkasprow, 2018-12-12:
no, we will use higher gain amplifier.

gkasprow, 2018-12-12:
we can lower the ADC range, use 100V/V amplifier to maintain 20mV voltage drop.

hartytp, 2018-12-12:
Why do you need a higher-gain amplifier? The main issue here is the input sense voltage being too low, not the output voltage being too low. The input sense requirements are the same for all chips in this family AFAICT.

gkasprow, 2018-12-12:
I need higher gain amplifier to keep 20mV voltage drop on resistors and use full ADC range.
Now we have 50V/V and use only half of ADC range because at such gain the amplifier works up to 1V at its output.
So when we switch to 100V/V, we keep 20mV input range and increase twice the output range. Then we increase input resistors to use full amplifier and ADC range. The amp has anyway +/- 0.5mA of offset so accuracy won't be much better.
But do we care about absolute accuracy so much?

hartytp, 2018-12-12:
> The amp has anyway +/- 0.5mA of offset so accuracy won't be much better.
> But do we care about absolute accuracy so much?

@gkasprow I'm not so worried about small errors here due to not using the ADC's full range, since we do not need this to be an ultra-precise measurement.

What concerns me is that these InAmps have very nasty non-linear behavior when operated with Vsense < 20mV. So, we must either use a larger sense resistor or use an InAmp designed for lower sense voltages.

If we do use a larger sense resistor, are we concerned with increasing the impedance of the power supply rails?

gkasprow, 2018-12-12:
We would increase the resistors twice or three times. So  the impedance would still be low. For 28V would be still  <0.06Ohm

hartytp, 2018-12-12:
@gkasprow correct me if I'm wrong, but for the 6V rail the current is only 30mA. So, to get 20mV minimum voltage we will need a 1 ohm sense resistor. Are you happy to have such a large sense resistor on a power supply rail?

gkasprow, 2018-12-12:
These RF stages don't output very  power so I don't expect large variation of supply current. We can change feedback loop in next revision.

hartytp, 2019-01-07:
NB this will require a change to the firmware to support the new current sense resistors.

hartytp, 2019-01-08:
@gkasprow I regard this issue as **critical** to fix on all existing Boosters, since with the current sense resistors the bias calibration is inaccurate and varies depending on which socket the RF module is plugged into.

Please could you change the sense resistors on the Booster I posted back to you? Once we confirm the new sense resistors work, and once we have resolved https://github.com/sinara-hw/Booster/issues/123, I will change the sense resistors on all other Booster units we have.

gkasprow, 2019-01-08:
OK, in this way firmware will be the same for all boxes.

hartytp, 2019-01-31:
This is fixed now, right? Can we close?