sinara-hw/Booster #127: Remove bias voltage current draw monitoring

hartytp, 2018-12-13:
Is there any reason to bother monitoring the current consumption on the Vbias rail?

We do not really care about the current draw on this rail (unless the FET gate shorts, it's basically 0mA). And, in any case, the InAmps can't measure down to nearly 0mA, so the reading is very inaccurate.

Let's just scrap it to reduce the design complexity and save cost + space(we can leave the 4R7 resistor to help filtering/EMI).

To do:
- Remove this from the firmware (@wizath)
- Remove InAmp and related components from the design
- adjust power budget

hartytp, 2018-12-13:
This does free up another input to the analogue mux, so we can consider whether there are any other analogue signals we want to monitor to help debugging etc.

wizath, 2019-01-31:
Removed

closed 2019-01-31 by wizath