sinara-hw/Booster #130: VCP noise

hartytp, 2019-01-08:
Currently, the VCP has a lot of diagnostic noise, that can't be suppressed (power calculations, temperatures etc). That's a bit annoying since it interferes with scraping the outputs (e.g. to log bias currents). IMO the VCP should not print random debug info, but should only print info when requested (with the exception of reporting serious device errors).

wizath, 2019-01-09:
I'll change the default log level to put messages only in the logstash. If you want to scrape data from VCP commands I can send you dedicated python driver that disables several features while exchanging data.

hartytp, 2019-01-09:
Thanks.

A lot of this will be less important once the SCPI interface is working and documented, since that's our preferred way of getting information out of Booster anyway. The SCPI interface should provide access to all device status info, so that the user never needs to touch the VCP interface apart from during calibration (which should only be done by the factory anyway).

hartytp, 2019-01-30:
@wizath now you're done fixing the broken Booster unit (thanks again for that!) please could you have a look at the remaining software issues? We really want to get this project finished asap and have a final version of the firwmare with all known issues fixed.

If there is any reason that you're not able to start working on this right away, please let me know (email is fine if easier).

wizath, 2019-01-30:
Yes, I've fixed most of the issues listed apart from new temperature management, but I had no time to test them.

hartytp, 2019-01-30:
Amazing, thanks! Are you able to start testing now, and post a new firmware version soon?

wizath, 2019-01-30:
Yes, I'll try to post working version before weekend.

wizath, 2019-01-31:
Removed unnecessary printouts from VCP. All important logs are now stored in _logstash_

closed 2019-01-31 by wizath