sinara-hw/Booster #132: USB console stops responding

cjbe, 2019-01-09:
After being on for a few weeks, a Booster stopped responding to commands or outputting logs on the USB serial port. The ethernet SCPI interface still worked. After unplugging and replugging the USB port  the device successfully enumerated, but still did not respond.
So far this has only occurred once.

Firmware version:
```
> version
RFPA v1.20, built Dec  6 2018 22:39:28, for hardware revision: v1.10
```

wizath, 2019-01-09:
I'll try to run data gathering on vcp for couple of days to see if the problem reoccurs.

hartytp, 2019-01-09:
If the VCP interface dies, will it be reported in the logstash? If so, can we expose the lagtash via the SCPI interface as well? That way we might get some better diagnostics to help debug this.

wizath, 2019-01-09:
It can be done

cjbe, 2019-02-06:
@wizath any progress in fixing this?

wizath, 2019-02-06:
Reproduced issue, set debugger for catching it again

cjbe, 2019-03-11:
@wizath with the firmware you posted in https://github.com/sinara-hw/Booster/issues/147#issuecomment-470537145 all the VCP issues seem to be fixed. 

(I will give it a while longer before I close it, just in case this is still lurking and I just have not seen it yet)

cjbe, 2019-03-19:
I have not seen this reoccur!

closed 2019-03-19 by cjbe

hartytp, 2019-03-19:
@wizath just to check, did you disable the CPU watchdog to fix this? Does that mean we're running without a watchdog to reboot the CPU in case of firmware crashes?

wizath, 2019-03-19:
Oh forgot bout it, I'll add feeding the dog in interlock task since its most crucial and it should work fine.