sinara-hw/Booster #133: Booster channel error LED on but no reported errors

cjbe, 2019-01-09:
One of the channels (ch5) on one of my Technosystems Boosters seems to have died. The channel was previously operating normally, and at some point went into the error state (red channel LED on). Neither the input or output of the channel were connected at the time the channel went faulty, but it might have become fault after a power cycle (the device was power cycled several times between testing the channel and noticing it was faulty).

The status table for the Booster is below:
```
PGOOD: 1
FAN SPEED: 0 %
AVG TEMP: 54.00 CURRENT: 54.00
CHANNELS INFO
==============================================================================
                #0      #1      #2      #3      #4      #5      #6      #7
DETECTED        1       1       1       1       1       1       1       1
HWID            AB:A8   8E:3E   B2:F1   51:A8   57:71   44:DC   60:92   75:57
TXPWR [V]       0.78    1.01    0.15    0.05    1.14    0.07    1.09    0.04
RFLPWR [V]      0.31    0.39    0.05    0.08    0.49    0.11    0.48    0.04
TXPWR [dB]      19.49   28.18   -1.10   -0.83   31.49   -2.31   30.22   -0.98
RFLPWR [dB]     23.59   32.60   3.49    -1.08   36.59   -2.70   36.65   -0.38
I30V [A]        0.150   0.181   0.085   0.072   0.256   0.070   0.224   0.136
I6V0 [A]        0.241   0.234   0.242   0.235   0.238   0.016   0.223   0.222
IN8V0 [mA]      0.039   0.018   0.013   0.021   0.008   0.109   0.008   0.010
5V0MP [V]       4.681   4.684   4.702   4.718   4.632   4.299   4.625   4.686
ON              1       1       1       1       1       0       1       1
SON             1       1       1       1       1       0       1       0
IINT            0       0       0       0       0       0       0       0
OINT            0       0       0       0       0       0       0       1
SINT            0       0       0       0       0       0       0       0
OVC             0       0       0       0       0       0       0       0
ADC1            1276    1655    240     75      1863    113     1787    73
ADC2            505     631     75      132     798     185     784     71
DAC1            94      94      94      89      95      103     97      97
DAC2            250     235     254     266     240     240     231     131
SCALE1          51      55      52      53      51      54      55      56
OFFSET1         282     105     297     119     257     238     125     128
BIASCAL         1305    2265    1920    1125    3120    2850    1245    420
HWIS            8.60    9.05    8.96    10.12   8.68    9.00    8.28    8.55
HWIO            -11.45  -13.28  -13.23  -19.55  -12.92  -12.73  -8.69   -10.45
LTEMP           51.00   54.25   51.00   50.50   52.25   45.00   51.00   50.00
RTEMP           53.00   54.00   53.25   50.25   53.50   46.00   52.00   50.00
==============================================================================
```

wizath, 2019-01-09:
Hmm, could you post the _logstash_ output?

cjbe, 2019-01-09:
Additionally, there seems to be no way of determining what the source of error, as none of the interlock or status functions report an off-nominal condition.

cjbe, 2019-01-09:
```
> logstash
[INFO] PGOOD: OK
[INFO] Found mainboard EEPROM, loading values
[INFO] RF Channel at 0 detected, id = 54:10
[INFO] [tempmgt] Temp 41.739925 -> Fan 33
[INFO] [tempmgt] Temp 39.739830 -> Fan 31
[INFO] [tempmgt] Temp 37.739651 -> Fan 29
[INFO] [tempmgt] Temp 35.739217 -> Fan 27
[INFO] [tempmgt] Temp 35.739456 -> Fan 27
>
```

cjbe, 2019-01-09:
It is also interesting to note that in the status table in the top post that all the channels are ~50 deg C yet the fan speed seems to be 0%

cjbe, 2019-01-10:
Today, the error LED for ch5 stays off for the first 15s after power-cycling the Booster, then it turns on, and stays on permanently.

wizath, 2019-01-10:
Is there any new error on log output? That error occurs when channels are enabled?

cjbe, 2019-01-10:
Aha - this did cause a loggable error this time. It seems like there is an over-current on ch5:
```
> logstash
[INFO] device boot
[INFO] SYSCLK frequency: 168000000
[INFO] PCLK1 frequency: 42000000
[INFO] PCLK2 frequency: 84000000
[INFO] ADC: OK | raw: 1976 | VrefInt 1.21 V
[INFO] PGOOD: OK
[INFO] Found mainboard EEPROM, loading values
[INFO] RF Channel at 0 detected, id = 54:10
[INFO] RF Channel at 1 detected, id = 54:10
[INFO] RF Channel at 2 detected, id = 54:10
[INFO] RF Channel at 3 detected, id = 54:10
[INFO] RF Channel at 4 detected, id = 54:10
[INFO] RF Channel at 5 detected, id = 54:10
[INFO] RF Channel at 6 detected, id = 54:10
[INFO] RF Channel at 7 detected, id = 54:10
[INFO] network init done, starting server..
[INFO] TCP Server initialization complete
[INFO] network socket state 13
[INFO] network socket listen OK
[INFO] DHCP IP acquired: 10.255.6.220
[INFO] DHCP lease time: 691200 s
[ERROR] OVC on channel 5, current 0.59
>
```

I don't understand why this does not set the "OVC" flag in the below table:
```
PGOOD: 1
FAN SPEED: 26 %
AVG TEMP: 33.25 CURRENT: 33.25
CHANNELS INFO
==============================================================================
                #0      #1      #2      #3      #4      #5      #6      #7
DETECTED        1       1       1       1       1       1       1       1
HWID            AB:A8   8E:3E   B2:F1   51:A8   57:71   44:DC   60:92   75:57
TXPWR [V]       0.90    0.87    0.10    0.04    1.21    0.06    1.09    0.04
RFLPWR [V]      0.32    0.20    0.04    0.05    0.54    0.09    0.46    0.04
TXPWR [dB]      23.27   23.85   -2.65   -0.89   33.94   -2.46   30.18   -0.98
RFLPWR [dB]     27.53   27.65   1.84    -1.14   39.14   -2.87   36.61   -0.38
I30V [A]        0.161   0.167   0.141   0.146   0.303   0.063   0.214   0.140
I6V0 [A]        0.242   0.236   0.244   0.237   0.238   0.018   0.225   0.223
IN8V0 [mA]      0.036   0.008   0.008   0.016   0.000   0.055   0.000   0.003
5V0MP [V]       4.745   4.771   4.774   4.762   4.701   4.515   4.705   4.750
ON              1       1       1       1       1       0       1       1
SON             1       1       1       1       1       0       1       0
IINT            0       0       0       0       0       0       0       0
OINT            0       0       0       0       0       0       0       1
SINT            0       0       0       0       0       0       0       0
OVC             0       0       0       0       0       0       0       0
ADC1            1469    1417    159     72      1988    105     1785    73
ADC2            531     330     72      74      890     140     756     68
DAC1            94      94      94      89      95      103     97      97
DAC2            250     235     254     266     240     240     231     131
SCALE1          51      55      52      53      51      54      55      56
OFFSET1         282     105     297     119     257     238     125     128
BIASCAL         1305    2265    1920    1125    3120    2850    1245    420
HWIS            8.60    9.05    8.96    10.12   8.68    9.00    8.28    8.55
HWIO            -11.45  -13.28  -13.23  -19.55  -12.92  -12.73  -8.69   -10.45
LTEMP           30.00   30.00   31.50   31.50   31.50   27.00   31.25   31.50
RTEMP           31.00   33.25   32.25   30.00   32.25   27.25   32.25   32.25
==============================================================================
```

wizath, 2019-01-31:
Changed setting OVC to 1 after ADC alert occurred.

There are 3 ways to get a RED LED (error condition):

1. Channel is not detected
2. Channel got temperature error (logged)
3. Channel got overcurrent error (logged)

closed 2019-01-31 by wizath

hartytp, 2019-01-31:
Sounds good. Please make sure this is documented in the wiki (or wherever we’re keeping docs these days)