sinara-hw/Booster #134: IC10 output capacitance

hartytp, 2019-01-10:
MCP1525T-I/TT needs a 1uF load capacitor for stability. Summing up all the capacitances on that rail it just comes to one uF. In practice (accounting for bias voltage, etc) we're probably not supplying enough capacitance to keep that reference happy.

hartytp, 2019-01-10:
Also, wouldn't one normally put 1uF at the microprocessor ADC Vref input as well?

hartytp, 2019-01-15:
@gkasprow what do you think about this?

gkasprow, 2019-01-15:
Let's simply put 1uF and forget about it.

gkasprow, 2019-05-26:
done

closed 2019-05-26 by gkasprow

hartytp, 2019-05-27:
@gkasprow are we confident that this ref can use a ceramic capacitor? The datasheet is clear that at least 1uF - 10uF is required for stability, but does not specify that it can be ceramic. Should we use tantalum/damping resistor to be safe, or are you happy with ceramic?

gkasprow, 2019-05-27:
This is a modern IC and they usually assume that ceramic capacitors can be used.

hartytp, 2019-05-27:
okay, good. The data sheet looked a bit old-school, which concerned me ;)

gkasprow, 2019-05-27:
Microchip produces analog chips for 15 years or so.
Old dinosaurs from 30..40 years ago had issues with ceramic caps. I.e. LM317 + 10uF ceramic at the output makes a generator.

gkasprow, 2019-05-27:
Anyway, if it oscillates we will see the effects immediately.