sinara-hw/Booster #135: Large per-channel variation in reported +5V MP rail

cjbe, 2019-01-10:
Looking at two Boosters, both show that the +5V management power rail is significantly below 5V. The regulator tolerance is <1%, so I am worried that this is something sinister:

Data from Booster "lab1_2", varies from 4.27V to 4.77V (4.27V is from ch5, the faulty channel from #133):
[image]

Data from Booster "lab1_3", varies from 4.49V to 4.78V:
[image]

There also seems to be some drop-outs in the measurements - I will discuss these in a separate issue.

hartytp, 2019-01-15:
@gkasprow did you have a chance to look at this and see why the 5V MP is so far out of spec?

gkasprow, 2019-01-15:
5V MP has PTC. Its resistance depends on load current so I'd look for it first

gkasprow, 2019-01-16:
the voltage on the LDO is 5.00V. The ADC input has 1.98V. Reference is 2.49V. And with these values the ADC reads 1.8V which then is multiplied by 2.5 .

gkasprow, 2019-01-16:
It looks like we are switching ADC channels too fast. And the C48 should be 220pF to provide enough charge for SAR ADC. The radical solution would be adding a buffer between Mux and ADC. Funny thing is that I used this circuit in many places before and nobody complained about ADC accuracy.

gkasprow, 2019-01-16:
Sorry, false alert.
I measured on the channel witout RF module and was looking at the channel where module was inserted :)
The 0.5V drop is on the PTC fuse.
The current flowing through the fuse is 17mA.
The producer specifies 2 resistances of the  fuse 3 and 60Ohms.
Real R values I measured are between 12 and 22 Ohms. So this explains a lot.
We will mount another fuses wich much lower resistance in the new revision.

gkasprow, 2019-01-16:
It does not change much if I add C48 capacitor or not. But let's mount 220p in rev1.3.

hartytp, 2019-01-16:
Okay, so we only have 4V5 on the 5VMP rail due to the resistance of the fuse. Is that likely to cause problems anywhere? Do we need to worry about that for the current design?

hartytp, 2019-01-16:
Okay, so not a crucial issue, we can fix later.

So, it looks like the issues we see are all minor apart from the bias current fluctuations...

hartytp, 2019-01-30:
@wizath Am I right in thinking that you fixed this issue by changing the ADC sampling time? If so, please can you post a trace showing what the measurement should look like?

Thanks!

hartytp, 2019-03-06:
This seems fixed on the v1.3 hardware

closed 2019-03-07 by hartytp