sinara-hw/Booster #136: Drop-outs / glitches on channel monitoring ADC

cjbe, 2019-01-10:
The per-channel monitor ADC seems to have significant glitches / drop-outs. Logging the values output by the VCP monitor task, I see frequent glitches on I30V, I6V, IN8V, and 5VMP rails. As these glitches are so large, yet we do not seem to see gross misbehaviour of the RF, I would guess that this is a problem with the ADC / readout / code, rather than on all the power rails themselves:

[image]
[image]
[image]
[image]

wizath, 2019-01-15:
I plugged scope to lines _meas_I_P29V_ and _meas_I_P6V_ and saw these spikes happening sometimes.

[image]

hartytp, 2019-01-15:
@wizath thanks for looking into this.

> I plugged scope to lines meas_I_P29V and meas_I_P6V and saw these spikes happening sometimes.

That's the input to the ADC/the filtered InAmp output, right?

For the 29V rail, the conversion is I=V/(20mOhm * 50V/V) = 1 A/V
For the 5V rail, the conversion is I=V/(100mOhm * 50V/V) = 0.2 A/V

Which channel is which? Assuming channel 1 (yellow) is 29V?

The DC current draw for the 29V rail is something like 160mA, so the DC voltage should be 160mV. The pk-pk there looks to be something like 4V, which means that the voltage at that node must be going negative, which isn't possible for a unipolar InAmp. So, is this this just pickup in the measurement setup?

wizath, 2019-01-15:
Hm, possible. I'm looking for an issue causing these glitches and can't find anything that would cause them.

hartytp, 2019-01-15:
ack. If I'm right that the glitches you showed there would correspond to a negative InAmp output then we can be sure that they're not generated by the InAmp. Whether they're an artefact of your measurement, or a genuine signal at the ADC is harder to say....

Out of curiosity, do you know what the source is? Looks like a SMPs. The glitches look identical on both channels so almost certainly from a common noise source.

gkasprow, 2019-01-15:
These spikes look like EMI from outside. Just connect robe and switch on/switch off the light in the lab :)

gkasprow, 2019-01-15:
I replaced all 28V Isense resistors with 0.1R||1R which gives 91mR. Then I added pullup to rev ID signals that now indicate 1. Other boards that were not modified have ID =0.
We observed 2 suspected channels but so far no current variation. I think that may cause an issue is that the booster was disassembled and not completely assembled. It looks to me like a mechanical problem which may be triggered by some vibration.

gkasprow, 2019-01-15:
@hartytp do you think there is a correlation of the issue with the fact that some modules have trimpot installed? Do some older revisions have such a problem?

gkasprow, 2019-01-15:
Jakub replaced the LDO transistor because it was broken.

gkasprow, 2019-01-15:
D you remember an issue with gain suddenly dropping in early versions? It was caused by too low inductor in output circuit. I did not understand it.

gkasprow, 2019-01-15:
missing trimpot contact would explain why current drops down. Poor contact could be caused by PCB washing process.

wizath, 2019-01-17:
I raised mainboard in order to give connectors more space to plug-in, then left booster running for some measurements. I logged three parameters - current, measured tx power and actual output power on scope. I can see small variations in current, but they are negligible (1-2 mA). But apart from that one channel got sudden drop in 28V current along with output power.

**Current plot (measured on Booster)**
[image]

**TX Power plot (measured on Booster)**
[image]

**Output plot (measured on scope)**
[image]

gkasprow, 2019-01-17:
I measured the temperature of the PA IC with thermal camera at 30dBm power and it is almost 80 degrees. The dice temperature is much higher. @wizath reduced the current to less than 70mA and we will leave it working to see if that helps.

gkasprow, 2019-01-17:
with reduced idle current and same output power the temperature reaches 55 C

hartytp, 2019-01-17:
@gkasprow hmmm...If I've read it correctly, the data sheet gives R_{theta,jc} as 15C/W. Assume approx 50% efficiency, so 5W dissipated in the chip. So, for an 80C case, the junction would be at something like 160C. Well within the absolute maximum ratings for the FET, but quite hot.

Worth noting that if one heats the FET while holding the gate voltage constant then the bias current decreases. Over 80C or two, there will be a significant change in the bias current, which could cause some instability.

gkasprow, 2019-01-17:
This is under assumption of ideal contact of the PCB with heat sink.

hartytp, 2019-01-17:
@gkasprow true, but MTBF is still 1 million hours with junction temperature of 200C, so I don't think we're likely to cause permanent damage to the FETs like this. Of course, they will not operate as specified once they get so hot, so other odd things may happen.

hartytp, 2019-01-17:
@gkasprow did you discuss with Maciej what bias current the design should use? Now we've fixed the current sense circuit, so the bias calibration should be more reliable, let's pick the correct bias current and check the circuit operates correctly at that current.

hartytp, 2019-01-17:
@wizath thanks for posting that data!

So, the glitch you saw was not as long or as frequent as the ones we see, but it is there. The bias current drops by nearly a factor of two and the output power drops at the same time (the scope reports only a small decrease in the output power, but maybe that is due to filtering?).

So, what do we do about this? How do we debug it?

gkasprow, 2019-01-17:
True, but we don't know what happens to the FET structure when it gets 2.5A  of current for some time. And this happens when missing gate voltage occurs. 
We set 70mA current and left the Booster overnight with scopes connected. Take into account that now we have correctly plugged D-SUB connector.

gkasprow, 2019-01-17:
Maciej was not available today - he has some champaign in DESY. Will try to reach him tomorrow.

hartytp, 2019-01-17:
> Maciej was not available today - he has some champaign in DESY

:)

gkasprow, 2019-01-17:
I mean campaign :)

hartytp, 2019-01-17:
> True, but we don't know what happens to the FET structure when it gets 2.5A of current for some time. And this happens when missing gate voltage occurs.

We don't know that ever happened do we? It's possible, but only a theory at this stage, right?

Also, if this issue is just due to FET damage then you should observe the same thing we did. However, you don't see as many glitches as we saw. That makes me think there is some other issue.

> We set 70mA current and left the Booster overnight with scopes connected. Take into account that now we have correctly plugged D-SUB connector

Okay, well if you don't see any glitches then it's probably time to send this Booster unit back to us. We'll install it in our lab and continue testing. If we don't see any more glitches then we'll know that the problem was either mechanical or instability due to operation with excessive bias current.

hartytp, 2019-01-18:
@gkasprow what did you find over night? Were you able to reproduce the bias current drop outs with the new bias current?

gkasprow, 2019-01-18:
We didn't observe such events anymore. We will leave it over the weekend and send it back on Monday.

hartytp, 2019-01-18:
Okay good. If you don’t see any glitches over the weekend, please can you send the Sayma back to us on Monday? Please overnight it so we have it on Tuesday. (If you need to invoice us for expedited shipping that’s fine)

hartytp, 2019-01-18:
Ps please make sure you check the gain and saturation power for all channels before shipping

hartytp, 2019-01-18:
The conclusion seems to be that this was a mechanical issue and either due to bugs in the v1.2 mechanics or due to the FETs running at the wrong bias current.

hartytp, 2019-01-18:
Let’s focus on getting the firmware finished next week

wizath, 2019-01-18:
We also need to fix damaged channels

hartytp, 2019-01-18:
Aah ok. Which channels are damaged?

hartytp, 2019-01-18:
Well, if possible, it would be really helpful to get it fixed and back in the post on Monday otherwise Tuesday is ok.

Thanks again for all the help with this!

gkasprow, 2019-01-18:
One of the LDO died.

wizath, 2019-01-18:
Well, one channel (kinda) died because of power loss. After setting lower bias current, it seems that they are more stable. But I was certain that channel 0 was damaged and then it started to work fine... But still there are minor glitches with measurements (probably parsing errors?) since they happened on single chart.

[image]

klickverbot, 2019-01-18:
> But I was certain that channel 0 was damaged and then it started to work fine... 

So does this point to another latent mechanical problem? Seems unlikely that the FET would suddenly recover…

gkasprow, 2019-01-18:
@wizath did you replace broken LDO and power transistor?

gkasprow, 2019-01-18:
we should definitely look at the power supply of channel 0

hartytp, 2019-01-20:
@gkasprow While I really want this Booster back asap, I'd also really like to avoid yet another cycle of sending it back to us, only for the issue to reappear and we have to send it back to you again. So, let's do our best to make sure that the issue really is gone this time.

1. It seems that the fixes you made to the mechanics did not help this issue. That's not surprising given our observation that all the correct voltages were present on the supplemental control board (we even checked the local ground potential relative to the chassis).
2. I agree with @klickverbot that this does not seem to be FET death, since the FETs start working again properly. It's also odd that we see exactly the same behaviour on many channels
3. All I can think of is a mechanical issue *within* the RF modules or a parasitic oscillation. Either of these might be fixed by reducing the bias current, or that might not help
4. This data https://user-images.githubusercontent.com/11598020/51409128-37980c80-1b61-11e9-8bc9-70326fc5e766.png is kind of scary! Those are large drop outs on channel 0. Even though they stopped at some point, there is clearly an issue there and this channel is *not* behaving correctly.

gkasprow, 2019-01-21:
The glitches are still occurring on 2 channels. Maciej said that they could be caused by overheated FETs. That's why I replaced one.

wizath, 2019-01-22:
[image]

Ran tests on channel with replaced FET and bias current set to 50mA. Despite variations due to temperature changes, it is behaving correctly I think. 

I'll rewrite temperature management since it broke (again) during tests. If any channel is enabled, they won't turn off, just step down to 20% speed to get rid of temperature drifts.

hartytp, 2019-01-22:
@wizath nice! That looks really good.

It's nice to see that all the glitches in the diagnostics are gone!

What's the plan now? Measure the bias current on all channels over night to see if they all behave correctly and, if there are no more glitches, send the unit back to us first thing tomorrow?

hartytp, 2019-01-22:
FWIW, I've never seen the bias current so stable on any of our Booster units, so that's a really good sign. Now that we can trust the current diagnostics we can fix these issues more easily, since we can make sure our experiment constantly logs the bias current on all channels and reports any glitches straight away.

gkasprow, 2019-01-22:
I have just two space PA chips and rest can desolder from old PA boards.

gkasprow, 2019-01-23:
Yes, it was a lot of work. It seems to work fine, Jakub is testing right now.

hartytp, 2019-01-23:
Great, thank you both!

Let's hope that after fixing the mechanics and bias current that it's all stable now.

To check, the plan is now to monitor the bias current on all channels over night and look for glitches?

Also, did you test the small signal gain and P1dB after replacing the FETs?

gkasprow, 2019-01-23:
Yes, Jakub was doing some tests. @wizath any conclusion?

wizath, 2019-01-23:
[image]

With all channels set to 50 mA's idle current, they seems fine for me (except for one channel that needs FET replacement). Need to recalibrate them, get one channel fixed from Maciej and we can ship it back to you. Beside that, at current state they heat up much less than before, fans don't even kick in. Will plot characteristics tomorrow.

hartytp, 2019-01-24:
That looks great! Did you check Gain and psat?

I just hope that we did find the cause of all this and don’t end up with another batch of dead FETs. But now you’ve got the diagnostics working so well at least we will know quickly if a problem arises.

hartytp, 2019-01-24:
If possible can you try to overnight this to us today so we get it on Friday?

wizath, 2019-01-25:
I'm still lacking one working channel. Maciej said that he'll get me one on monday, then I'll check it and we'll post it back to you.

hartytp, 2019-01-25:
@wizath thanks for the update. If we don't find any other issues, please can we aim to ship this back on Monday?

Before then, could you confirm what the small signal gain, P1db and Psat with the new bias current is, please? (A S11 small signal trace would be great).

Also, please could you log temp, bias current and output power of all working channels over the weekend, please?

hartytp, 2019-01-28:
@wizath did you manage to install the last channel? Do you still think we'll be able to finish testing by the end of day?

Also, how did the logging over the weekend go?

hartytp, 2019-01-29:
@wizath how did the testing go? Did all channels operate correctly without any instability?

Are you ready to ship it back to us today?

wizath, 2019-01-29:
I finished testing and re-calibrating entire unit. Apart from some channels having a little lower gain, they seems to work fine. Courier is set to tomorrow morning to take the package.

hartytp, 2019-01-29:
@wizath Great, thanks!

Do you have any logs you can post showing the stability of the gain/bias current of all channels?

Also, did you measure the gain and P1dB/Psat of any channels? If so, can you post the data you have please?

Any info you have would be helpful to give us something to compare our observations with in case we suspect there is an issue. (essentially, it tells us what "normal" device operation looks like).

hartytp, 2019-01-30:
ping @wizath could you post any data you have to give us an indication of what level of performance to expect from a working Booster?

wizath, 2019-01-30:
[image]

There are clearly some measurement glitches occurring, but it seems that they don't affect output power (since there is no drop in TX power as before). I added error counters for each I2C transactions and will try to catch that (again). 

Apart from that, the lines look straight.

hartytp, 2019-01-30:
@wizath thanks for posting the data! That looks much better.

Did you record the small-signal gain/P1db/Psat for each channel? If so can you post it, please?

closed 2019-03-07 by hartytp