sinara-hw/Booster #137: non-contacting DSUB connectors

gkasprow, 2019-01-16:
@hartytp stupid thing, but it may be possible that many of the issues could be related to the wrongly positioned mainboard shelf. Please use calipers and measure RF module to mainboard distance. It should be 16.2-16.5mm. If it is 17 or more, the contact cannot be maintained and this would explain many issues with both bias current, I2C, LEDs.
There are 2 solutions. Radical and easy.
Easy is to lift the shelf to the extent. You need 2.5mm hex key and pliers. Some screws have easy access, some need pliers.
In rev1.3 the screws will be mounted below the shelf so access would be very convenient.
The radical solution is to replace all mainboard standoffs. There should be 10mm ones, now are 8mm ones. When Technosystem redesigned the board, they didn't notice that shelf regulation range changed when they used much thicker bottom and upper walls.
In a moment I will post a photo which shows how to measure module position.

gkasprow, 2019-01-16:
Such a situation does not maintain contact
[image]
such mounting height does
[image]
and this is a way how to check correct position
[image]
it should be 16-16.5mm

cjbe, 2019-01-16:
This would explain quite a few of our observations!
A lot of the problems seem to be worse with the lid removed from the Booster (hence we worried about oscillation) but whenever we had the lid off we had the Booster sitting on a table rather than supported in a rack, hence changing the mechanical alignment somewhat.

gkasprow, 2019-01-16:
My professor used to say that practical electronics is a branch of science that studies missing connections.

gkasprow, 2019-01-16:
When U lift the shelf, it may happen that power supply connector conflicts with upper lid. One can change it to right angle one. It was such one in the prototype but was changed to straight one by Technosystem.

gkasprow, 2019-01-16:
I asked to assemble the power connector reversed so RA plug could be used. But did not change that in the documentation so next board batch had this connector assembled in wrong way which prohibits usage of RA plug... And they could install the mainboard at the lower position to fit the connector.

gkasprow, 2019-01-16:
In rev1.3 the power entry connector is correctly rotated.

hartytp, 2019-01-16:
@gkasprow Good catch!

While this could explain some things, I do not think it could explain the bias current drop outs. We measured the 29V rail and the bias voltage rail on the supplemental control board. We did not see any variations in these voltages during the bias current drop outs.

hartytp, 2019-01-16:
@cjbe can you check the v1.0 design (silver box) and see if that has bias current issues?

cjbe, 2019-01-16:
Aha - lab1_2 (silver box, v1.0) does not show any drop-outs. lab1_3 (black box, v1.1) does show drop-outs (and lab1_1 is the black box that we sent back to @gkasprow that also shows drop-outs).

hartytp, 2019-01-16:
@gkasprow is there any changed between the silver and black booster units that could cause this?

gkasprow, 2019-01-16:
@hartytp with missing GND contact the ground would be closed by external cabling, causing PA GND potential to be higher than chassis GND. This would increase the effective negative voltage between PA GND and MSOFE gate voltage decreasing the bias current.

gkasprow, 2019-01-16:
we have only 2 GND pins placed on the same connector edge.

hartytp, 2019-01-16:
> @hartytp with missing GND contact the ground would be closed by external cabling, causing PA GND potential to be higher than chassis GND. This would increase the effective negative voltage between PA GND and MSOFE gate voltage decreasing the bias current.

hmmm...I don't think that's the issue, but it's possible.

Were you able to reproduce the glitches we saw? If so, this should be easy to confirm.

gkasprow, 2019-01-16:
It looked good but Jakub just reported that gain on one channel dropped for a moment. So the issue is there. But at lease we managed to exclude mechanics and recreate in the lab.

gkasprow, 2019-01-16:
To replay your questions - there were a lot of changes in the design but not many that affect the power stage. We added trimpot, changed the LDO range and that's all that could be related to the power stage.

hartytp, 2019-01-16:
> Jakub just reported that gain on one channel dropped for a moment. So the issue is still there. But at least we managed to exclude mechanics and recreate in the lab.

Okay, that's good! At least you can reproduce the issue now

gkasprow, 2019-01-16:
We leave it overnight with scopes connected to various points. The statistics are poor so far.

gkasprow, 2019-01-16:
The script is also working looking for terminal output.

gkasprow, 2019-01-16:
@hartytp the more I think about it, the more I'm getting convinced that these bias current issues are caused by damaged MOSFETs. I cannot find the issue number on github where you send me back the booster because the DC/DC converter was broken and I had to replace all power stages. They behaved in very similar way - loss of gain, inability to set-up stable bias current.
The scenario is simple - one of the module does not receive bias voltage while the 28V rail is on. The MOSFET turns fully on and gets 2.5A gets beating which damages it partially. In some cases it causes LDO overheating. 
Do you remember where this issue was placed?

hartytp, 2019-01-16:
@gkasprow I think this issue was even present on the old v1.0 hardware (silver box) at some point. See here 
https://github.com/sinara-hw/Booster/issues/2#issuecomment-383538536

when we first received it there was one channel with the wrong gain by 10dB.

hartytp, 2019-01-16:
We also observed this: https://github.com/sinara-hw/Booster/issues/36

hartytp, 2019-01-16:
Here we saw oscillations on one channel, followed by catastrophic failure: https://github.com/sinara-hw/Booster/issues/50

We assumed this was to do with the coax cable going too close to the FET, but maybe there was some other cause?

hartytp, 2019-01-16:
Also here we see a large variation in the gain of some channels when I first receiver this Booster unit: https://github.com/sinara-hw/Booster/issues/35

So, either the calibration was not done correctly, or this issue occurred there as well.

hartytp, 2019-01-16:
It's possible that the FETs are dying. It seems very odd that the damage is so intermittent. You have played with the Booster we sent you for quite a while and it seems to work, even though you suspect the FETs to be damaged. It's also not clear how the FETs are being damaged in the first place. Power sequencing issue? Firmware bug?

hartytp, 2019-01-16:
@gkasprow am I going mad? The output FET is a [NPTB00004A](https://cdn.macom.com/datasheets/NPTB00004A.pdf) which is specified to operate at Idq of 50mA. Why have we calibrated them to operate at 160mA? Can they actually withstand that level of quiescent current?

hartytp, 2019-01-16:
Why was this bias current chosen? Also, given the InAmp issues, is it possible that we're running far too much currents through these FETs?

Finally, are you confident that the correct turn on/turn off sequencing is followed in the firmware?

hartytp, 2019-01-16:
The other thing I'm curious about is the 28V FET biasing. Normally when I've designed PAs it's necessary to have some RC damping in the drain decoupling. This is done on the data sheet design (see R2), but not in Booster. Are we sure that we're happy with the PA design from a stability point of view?

gkasprow, 2019-01-16:
Turn on/turn off sequence is ensured by hardware sequencer. There is other HW protection mechanism that ensures start of sequencing when all voltages are present.
We have tantalum 47uF capacitor with 0.3R ESR on upper board so it is equivalent to damping circuit.
The datasheet specifies the Idq up to 100mA.
I don't know precisely why we had chosen such current. It was after series of tests because it gave best performance. At such current we are dissipating nearly 5W in this poor chip and this may stress it too much.

hartytp, 2019-01-17:
@gkasprow according to the ADC when we run Booster at 5W of RF the bias current is in excess of 250mA.

I wonder if what happened was that when we 50mA was set the chip didn't work well because of the ADC reading being incorrect. To compensate, a higher current was chosen.

hartytp, 2019-01-17:
> Turn on/turn off sequence is ensured by hardware sequencer. 

Okay. The data sheet recommends first increasing the bias voltage to quite a large value (pinch off). Then turning on the 28V supply, then reducing the bias voltage to achieve the desired bias current. That's not quite what we do is it?

> We have tantalum 47uF capacitor with 0.3R ESR on upper board so it is equivalent to damping circuit.

Okay. So long as this has been thought about and simulated, that's fine then.

gkasprow, 2019-01-17:
I think this is exactly what we do.
Am I right @wizath ?

hartytp, 2019-01-29:
@gkasprow please can you confirm that this issue is fixed in the new design?

gkasprow, 2019-01-29:
Yes, v1.3 solves it completely. Let's wait for the HW to arrive.

gkasprow, 2019-01-29:
new mechanics has a completely different fixing mechanism. Each module is individually screwed to the additional mounting plate.

hartytp, 2019-01-29:
Cool so it’s just a matter of fixing existing units to prevent failure.

hartytp, 2019-01-29:
@gkasprow please can we make sure that when the new units arrive they are properly tested to prevent this kind of issue.

Also was the new mechanics uploaded to github yet?

gkasprow, 2019-01-29:
Not yet, I want to make sure everything fits nicely. But will do.

hartytp, 2019-03-06:
@gkasprow Did you have a chance to check this on the new V1.3 mechanics, or should I look at this when we open the Booster up to tighten the SMA screws? If so, what should I check?

gkasprow, 2019-03-06:
I didn't check it. I asked the Technosystem to pay attention during assembly.
@wizath did you look if the modules are plugged deep enough to the mainboard sockets?
Just to make sure...

hartytp, 2019-03-06:
Okay, thanks. When I open the box up, I'll check for this clearance... 
https://github.com/sinara-hw/Booster/issues/137#issuecomment-454726446

hartytp, 2019-03-08:
@gkasprow https://github.com/sinara-hw/Booster/issues/137#issuecomment-454726446

I measure 17mm in the TechnoSystem v1.3 hardware. Is that okay, or should I report the issue?

gkasprow, 2019-03-08:
Good question. There were issues with DSUB connectors availability and replacement ones were installed which could result with a slightly different height. We will check it with the second, identical unit that @wizath is working on right now. @wizath please remind me on Tuesday to check if the connectors are correctly matched.

hartytp, 2019-03-08:
It seems fine, but worth double checking. I'll reopen this as a reminder. Please close when you're happy that the new mechanics are fine.

hartytp, 2019-03-19:
@gkasprow can you confirm whether the 17mm used in the current TechnoSystem mechanics is okay? Please close this issue if it's fine.

hartytp, 2019-03-19:
@gkasprow do you think you can have a look at this tomorrow so we can close this issue? Thanks!

gkasprow, 2019-03-19:
Yes, I even added it to the calendar to not forget it.

hartytp, 2019-03-19:
;)

Thanks Greg!

Sorry we've been chasing on Booster so much recently. It's so close that I really just want to finish it on and move on to other things.

gkasprow, 2019-03-19:
BTW, tomorrow @marmeladapk and @wizath will get Master of Science degree. So it's going to be a great day for them.

hartytp, 2019-03-20:
Whooo!! 🍾

gkasprow, 2019-03-20:
@hartytp I've checked the new hardware and the DSUB connectors are properly matched. They use slightly higher replacement so 17mm is still fine.

hartytp, 2019-03-20:
Great, thanks.

closed 2019-03-20 by hartytp