sinara-hw/Booster #139: Full characterisation at lower power stage bias current

klickverbot, 2019-01-18:
In https://github.com/sinara-hw/Booster/issues/136, it was discovered that the output PA bias current was set higher than seems optimal, and reduced accordingly.

What is the plan for characterising gain/P1dB/IP3/… and their channel-to-channel variations with the new bias settings?

(Apologies if this has already been discussed in a private forum.)

klickverbot, 2019-01-18:
(I.e. are we doing this once we have the unit back?)

hartytp, 2019-01-20:
@klickverbot The plan is for @gkasprow/@wizath to do basic characterisation/sanity checking before returning the unit to us. This means measuring the small-signal gain and P1dB/Psat on all channels. We'll do any other testing we want done (based on feedback from @cjbe who is in a real hurry to get this booster back into the lab).

klickverbot, 2019-01-21:
Ack.  See also: #140.

closed 2019-01-29 by hartytp