sinara-hw/Booster #140: confirm bias current

hartytp, 2019-01-21:
Before we decide to fix the bias current at 50mA we must understand why it was set to 160mA in the first place. Was this a deliberate decision e.g. to improve device linearity? Does reducing the bias current actually help to fix the issues we've seen with Booster (and, if so, why?)? And, are there any downsides to reducing the bias current.

@gkasprow can you discuss this with Maciej please?

gkasprow, 2019-01-21:
I discussed it with Maciej and he frankly said that the bias current was chosen arbitrarily and he can not really say what was the rationale behind it. He simply got satisfactory results and didn't optimize it.

gkasprow, 2019-01-21:
I've just replaced the power stage with a new one and now Jakub is testing it. We additionally are observing bias voltage and 28V rail with the scope.

gkasprow, 2019-01-21:
We are testing it already more than 1.5h and so far it's stable. We only see 0.2dB drift caused by temperature. Jakub will plot results in a moment.

hartytp, 2019-01-21:
@gkasprow Interesting...

So, what do you think then? FET damage due to excessive temperature? Seems surprising since I don't think we ever exceeded 200C junction temperature.

If so, can you try logging all channels over night, please?

If there are no more glitches at the lower current, and if the small-signal gain and Psat still look okay then please send us the unit back in the morning.

hartytp, 2019-01-21:
What still bothers me about all of this is that I don't feel like we understand the underlying issue. Did the FET really die, or did poking the circuit to replace the FET just make the issue go away for a while? If the FET did die, what caused it? Since we don't have a particularly plausible explanation for the symptoms we're seeing, I'm cautious in believing that we've really solved the problem...

gkasprow, 2019-01-21:
This particular FET got serious beat from dying LDO. It could get several tens of Amps in peak. Tomorrow morning I will replace another channel FET and leave them for a few hours.
Remember that the FET dissipated nearly 5W of power during idle mode and probably twice more with full output power. Its structure could get a very high temperature gradient and simply crack.

hartytp, 2019-01-21:
@gkasprow yes...but why did the LDO die?

Afaict the power dissipation in the FET does not increase with RF appled. With 5W of RF it’s about 300mA *30V=10W but half goes to the load so still same power dissipated in the FET

hartytp, 2019-01-21:
I still don’t understand what’s causing the FETs to draw so much current that the LDOs die.

gkasprow, 2019-01-21:
The mechanical issue which causes the lack of FET  gate voltage is enough to trigger LDO short circuit. This causes >10W dissipation of LDO transistor which triggers overheating, melting the tin, thermal breakdown, damage of LDO controller current sense resistor and large current surge on the FET. Some channels managed to survive, some not, but many FETs could be affected.

gkasprow, 2019-01-21:
You observed solder balls around the LDO transistor. Moreover, we found one of the LDO channels dead. So something triggered such an event. It could be a mechanical issue or FET overheating, I have no proof which one actually happened. The plot of the output power/supply current resembles shot noise, but in a very slow time scale. Could be caused by thermal effects in already damaged FET.

gkasprow, 2019-01-21:
For this reason, we connected probes to every point of the power stage we could and are waiting for the event to occur.

hartytp, 2019-01-22:
Good plan. Okay let’s see what the log says today. Also worth logging while power cycling a few times, running bias cal etc

closed 2019-01-29 by hartytp