sinara-hw/Booster #146: Scatter of 30V rail current`

cjbe, 2019-02-04:
@gkasprow @wizath on the repaired Booster we have received back there is significant scatter on the 30V rail: the current varies from 53mA to 85mA. I thought you adjusted these to all be 50 mA? Is this level of scatter something you would expect?

```
PGOOD: 1
FAN SPEED: 0 %
AVG TEMP: 25.98 CURRENT: 28.25
CHANNELS INFO
==============================================================================
                #0      #1      #2      #3      #4      #5      #6      #7
DETECTED        1       1       1       1       1       1       1       1
HWID            7C:60   3A:15   42:15   7C:5D   2A:E4   5D:41   61:98   8D:F0
TXPWR [V]       0.08    0.04    0.04    0.04    0.06    0.04    0.10    0.05
RFLPWR [V]      0.04    0.04    0.04    0.05    0.04    0.05    0.09    0.04
TXPWR [dB]      -1.61   -0.16   0.05    -0.50   -1.80   -0.06   -1.98   -1.08
RFLPWR [dB]     1.22    -0.04   0.02    -1.65   0.61    -1.71   -0.45   -0.20
I30V [A]        0.053   0.085   0.054   0.057   0.065   0.069   0.074   0.053
I6V0 [A]        0.247   0.240   0.242   0.240   0.242   0.252   0.250   0.246
5V0MP [V]       4.875   4.884   4.860   4.889   4.797   4.771   4.886   4.841
ON              1       1       1       1       1       1       1       1
SON             1       1       1       1       1       1       1       1
IINT            0       0       0       0       0       0       0       0
OINT            0       0       0       0       0       0       0       0
SINT            0       0       0       0       0       0       0       0
OVC             0       0       0       0       0       0       0       0
ADC1            126     68      67      70      98      71      168     73
ADC2            68      73      69      82      70      82      141     71
DAC1            85      74      86      85      77      88      85      85
DAC2            248     261     263     275     280     270     265     247
SCALE1          51      56      56      58      51      53      48      53
OFFSET1         208     76      64      99      189     75      263     130
BIASCAL         1422    1180    720     1314    2340    606     1560    2014
HWIS            11.55   10.68   8.60    13.20   11.11   9.02    9.50    11.57
HWIO            -27.69  -23.46  -11.92  -38.17  -26.64  -14.51  -17.03  -27.73
LTEMP           26.00   27.50   27.00   25.00   26.50   27.25   28.25   27.50
RTEMP           26.00   26.00   27.00   27.50   26.50   28.25   27.00   27.25
==============================================================================
```

cjbe, 2019-02-04:
Here is the `logstash` output:
```
> logstash
[INFO] hardware revision 1
[INFO] device boot
[INFO] SYSCLK frequency: 168000000
[INFO] PCLK1 frequency: 42000000
[INFO] PCLK2 frequency: 84000000
[INFO] ADC: OK | raw: 1980 | VrefInt 1.21 V
[INFO] PGOOD: OK
[INFO] Found mainboard EEPROM, loading values
[INFO] RF Channel at 0 detected, id = 7C:60
[INFO] RF Channel at 1 detected, id = 3A:15
[INFO] RF Channel at 2 detected, id = 42:15
[INFO] RF Channel at 3 detected, id = 7C:5D
[INFO] RF Channel at 4 detected, id = 2A:E4
[INFO] RF Channel at 5 detected, id = 5D:41
[INFO] RF Channel at 6 detected, id = 61:98
[INFO] RF Channel at 7 detected, id = 8D:F0
[INFO] [tempmgt] Temp 34.236306 -> Fan 26
```

cjbe, 2019-02-05:
Leaving this logging overnight with ~28.5 dBm output from every channel, I see some channels behave stably, but some (ch1, yellow, in particular) show a drift in bias current:
[image]
The transmitted power shows similar misbehaviour, again especially for ch1:
[image]
The temperature appears stable throughout all of this:
[image]

@gkasprow @wizath any suggestions?

hartytp, 2019-02-05:
@wizath please can you have a look at this asap.

- are the bias currents reported above (without RF power applied) consistent with the values you expect? If not, is it possible that the channels were permuted in the chassis before shipping?
- are the other observations above consistent with your expectations? The drop in bias current and output power on the yellow channel seems particularly disturbing as it could indicate that the old issue is still present.
- how do you suggest we debug this? Should we ship the booster back to you asap for debugging?

wizath, 2019-02-05:
Yes that one channel seems to be faulty. I didn't replace every channels FET, because I had only a few replacements.  Apart from that, they seem to be stable. I'll try to get working channel module from Maciej and send it to you as replacement?

wizath, 2019-02-05:
1. Yes I've left them overnight with about 28-30 dBm output power and the chart's were ok.
2. As I re-calibrated each one of them, they seemed to be working fine (reaching proper gain etc)
3. Bias current is another story. I double checked and bias voltage is set as described in calibration values, but varying from power-cycles current may be off by some amount and I don't really know why - maybe we should opt for fast bias current cal after powering-up the unit, since this behavior is quite unpredictable.

hartytp, 2019-02-05:
> 1. Yes I've left them overnight with about 28-30 dBm output power and the chart's were ok.
> 2. As I re-calibrated each one of them, they seemed to be working fine (reaching proper gain etc)

That's interesting. So these channels did not seem to be misbehaving when you used them at WUT, but they are instantly misbehaving here. That's concerning.

> 3. Bias current is another story. I double checked and bias voltage is set as described in calibration values, but varying from power-cycles current may be off by some amount and I don't really know [...]

I'm not sure I understand you. Is your observation that once each Booster is powered on the bias current is stable, but that there appear to be variations in the bias current between power cycles? That's quite concerning. Do you have any idea what could cause that?

hartytp, 2019-02-05:
> I'll try to get working channel module from Maciej and send it to you as replacement?

I'm still concerned that there is an underlying issue that we haven't understood/fixed yet. We can't keep piling up the dead FETs indefinitely.

wizath, 2019-02-05:
Could you tell me which channel number is orange and yellow? There may be couple of possible reasons.

- One channel was resurrected by me (by replacing power detector) to complete the unit, since Maciej was not really available to provide me a working one
- I've only replaced about 4 FETs since that was all I had
- But again, they seemed to be working fine before shipping

I have some channels that drop gain from time to time so it's up to Maciej to find the issue

cjbe, 2019-02-05:
ch1: yellow
ch6: orange
ch7: red

cjbe, 2019-02-05:
@wizath so far based on both the transmitted power diagnostic and the 30V rail current diagnostic all channels look good apart from 1 and 6.

FYI after I received the Booster I measured the gains for every channel at -30dBm and -10dBm input power - these all seem very consistent, despite the large scatter in bias current.

When you tuned the bias current of each channel did you try and set them to 50mA, or did you set them to give a fixed gain? 
[image]

[image]

wizath, 2019-02-05:
I've re-calibrated them couple of times to set them at 50mA's

hartytp, 2019-02-05:
> I have some channels that drop gain from time to time so it's up to Maciej to find the issue

Aah, great! I didn't realize that Maciej had a non-working channel. Since he can now reproduce the issue hopefully he can solve it :)

hartytp, 2019-02-05:
@wizath I still don't fully understand your observations about the bias current changes. Which of these do you mean:
1. The bias current varies after calibration, even without power cycling the unit
2. the bias current is stable after calibration so long as the unit is not power cycled. after power cycling, the bias currents start up to a different value, but are stable at that value until the next cycle

wizath, 2019-02-05:
Second one, if you set current to fixed value, it can vary after power-cycling the device, even if the bias voltage is set to desired level.

cjbe, 2019-02-05:
@wizath can you send us the measured bias currents of this Booster before you shipped it to us? If there were all calibrated to be 50 mA, but have changed to between 53 mA and 85 mA from shipping this is very worrying

hartytp, 2019-02-05:
> Second one, if you set current to fixed value, it can vary after power-cycling the device, even if the bias voltage is set to desired level.

That's pretty terrifying. Did you check that the bias voltage does not change between power cycles? Otherwise, what I think you're telling me is that these amplifiers exhibit metastable behavior; even with the same temperature and bias voltage, they can draw different bias currents. If so, that would explain why they sometimes jump between different bias currents after power cycling.

wizath, 2019-02-05:
I've already checked that, but will make sure that it's set to proper value again to exclude any mistakes.

hartytp, 2019-02-05:
@wizath If we're going to track down these horrible gain fluctuations then we have to be really critical in our approach to testing/debugging. If the amplifiers were clearly exhibiting inexplicable meta-stable behavior across power cycles, why did you ship it back to us? A working FET amplifier will have a very repeatable gain under constant bias and temperature. So, if the bias is not repeatable then the only conclusion is that there is something fundamentally wrong with the amplifier.

hartytp, 2019-02-05:
Anyway, the positive from all of this is that we now do clearly have a well-defined problem to investigate in terms of amplifier bias current instability. It's clear that we haven't solved this issue, and more work is needed on the RF modules to understand it.

hartytp, 2019-02-05:
FWIW, this feels very reminiscent of https://github.com/sinara-hw/Booster/issues/10 so I wonder if we never really fixed the root cause of that issue after all?

wizath, 2019-02-05:
Mostly because everyone was so eager to send it back after seeing the first results, and at first I though that issue was not so important as it is now.

gkasprow, 2019-02-05:
Is there a chance that trimpot changed its position during transport?

This would explain such difference in the bias currents.

hartytp, 2019-02-05:
@gkasprow that wouldn't explain @wizath's observation that, even after replacing the FETs and calibrating the bias voltage, the bias currents still varied between power cycles.

gkasprow, 2019-02-05:
Let's measure FET gate voltage after several power cycles. This could be caused by some software issues.

hartytp, 2019-02-05:
Do you mean looking at the steady-state FET gate voltage, or looking for transients during turn on/turn off?

Where will you probe this? On the supplemental control board, or on the RF PCB (at the FET pin)?

Chris and I carefully measured the FET bias voltage on the supplemental control board for a malfunctioning channel. We observed the bias voltage being very stable, while the bias current and gain varied significantly. We measured the bias voltage both relative to the main PCB ground and relative the supplemental control PCB ground to rule out any grounding issues. So -- barring a failure in the pot/solder joints -- I do not believe that these instabilities are related to changes in the bias voltage.

I believe that @wizath made similar measurements and also concluded that variations in the steady-state bias voltage are not the cause of the gain/current instabilities that we've observed.

gkasprow, 2019-02-05:
Did you measure it after several power cycles or observed long term stability?

cjbe, 2019-02-05:
This afternoon I power-cycled the Booster twice, once leaving it off for a minute, and once leaving it off for 20 minutes. I then turned it back on an applied RF at the input as before. Both times all the measured bias currents came back to within ~1 mA of where they were before.

Obviously this is a small sample - @wizath what changes exactly did you see when you power-cycled it? How much did the current change from the nominal 50 mA value? Did it jump and stay steady?

hartytp, 2019-02-05:
> Did you measure it after several power cycles or observed long term stability?

IIRC, we looked across a few power cycles and at long-term stability. The voltage seemed stable but not the current/gain. Although, of course, that was with a unit that may have had a damaged FET.

I also wonder if there is any possibility that the channels were swapped around after calibration but before Booster was shipped to us. When we received it, all the front panel screws were missing and the modules were unsupported.

hartytp, 2019-02-18:
@wizath @gkasprow good news: apart from the two channels that misbehaved when the Booster arrived with us, all other channels still seem to work correctly after a week of continuous usage.

Can you post us two working channels to replace the damaged ones so we can test a full Booster chassis?

hartytp, 2019-03-01:
Ping @gkasprow @wizath are you able to send us two working channels to replace the two that were DOA in this unit?

gkasprow, 2019-03-01:
We have 2 new channels from current production. @wizath are we able to prepare them?

gkasprow, 2019-03-01:
The channels need to have a rev1.2 control board and a potentiometer added to be compatible with rev1.2 main board.  Jakub is still fighting with software fixes caused by different calibration and interlock values and ranges caused by new revision hardware.

hartytp, 2019-03-04:
@gkasprow understood. When do you think the channels will be ready?

gkasprow, 2019-03-05:
At the moment we don't have any rev1.2 mainboard + chassis to test them. @wizath has something to do in another project that has the highest priority so we can have a look at it next week.

hartytp, 2019-03-06:
Closing this issue as, so far, it doesn't seem to be present in the v1.3 hardware

hartytp, 2019-03-06:
@gkasprow one question: it seems that the biggest problem with the v1.1 and v1.2 hardware was thermal runaway with the 28V LDO BJT. Are you absolutely sure that with the new current limit in v1.3 the BJT can never go into thermal runaway (even at max ambient temperature etc)? Should we consider using a transistor with a higher current rating/internal thermal limiting/etc to make absolutely sure?

gkasprow, 2019-03-06:
The new current limit and foldback reduce the BJT max power to a safe limit. It's roughly 2W and this can be dissipated.

hartytp, 2019-03-06:
Great!

closed 2019-03-07 by hartytp