sinara-hw/Booster #147: Access diagnostics via SCPI interface

cjbe, 2019-02-06:
Can we add some commands to access the diagnostics over the SCPI (ethernet) interface? I would like to be able to read out all the diagnostics I can get from the VCP status table via the SCPI interface - I don't have a strong opinion on how the interface should look.

hartytp, 2019-03-01:
@wizath are you able to work again on the Booster firmware now that the new hw is here? We still don't have a version with reliable diagnostics (both the SCPI and VCP crash after logging for a few days), which would really help us to debug potential hw issues.

wizath, 2019-03-04:
Hi,

Sorry for such delay, I was a little overwhelmed. ATM I've got one v1.3 board - finished checking new hardware interlocks and bias current calibration. Need to rewrite most of the calibration functions to include new hardware revision. 

Also, disabled watchdog to catch that nasty bug that causes VCP to crash and left it running, will find out about the cause tomorrow. Also, what kind of diagnostics would you like via SCPI? Like one command to return all of the crucial statistics from selected channel? I've updated SCPI command definitions to include most of VCP status page data:
https://github.com/sinara-hw/Booster/wiki/Remote-programming-interface

hartytp, 2019-03-06:
> Also, disabled watchdog to catch that nasty bug that causes VCP to crash and left it running, will find out about the cause tomorrow. 

Did you have any luck tracking this down?

> Also, what kind of diagnostics would you like via SCPI? Like one command to return all of the crucial statistics from selected channel?

Yes, a single command to get all diagnostic info for a channel in one go would be great.

hartytp, 2019-03-06:
> I've updated SCPI command definitions to include most of VCP status page data:
https://github.com/sinara-hw/Booster/wiki/Remote-programming-interface

Great, thanks!

To check, can you confirm that all commands in the SCPI/VCP interface are listed there, and that the description is correct (e.g. right arguments, units documents etc).

Also, can you confirm that all commands do proper input validation (can't set out of range currents for the bias etc)?

wizath, 2019-03-06:
Ok, I think I found issue with SCPI crashing. DHCP client is returning error while setting DHCP after some time and that crashes whole Ethernet connection.

My VCP was running for 3 days with constant communication and did not break. Maybe that's related to disabled watchdog. I;m gonna give watchdog more time before reset and feed it in the interlock task since that's most crucial task.

hartytp, 2019-03-06:
Nice! Good catch.

hartytp, 2019-03-07:
@wizath can we have a version of the FW with the DHCP fix (and ideally a fan speed fix as well)?

Generally, it would be really helpful to get incremental firmware releases if possible rather than waiting for a release that fixes many issues at once.

wizath, 2019-03-07:
Pushed DHCP fix to github branch v1.3, with disabled watchdog - hopefully that fixes both communication interfaces crashes

hartytp, 2019-03-07:
Does that also fix the fans?

wizath, 2019-03-07:
Pushed fans update. When enabled they will run at 30%

hartytp, 2019-03-07:
Thanks. Did you post binaries?

wizath, 2019-03-07:
Oh, snap. Forgot
[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/2941564/eth-scpi.dfu.zip)

hartytp, 2019-03-07:
thanks! Okay, cool, so just the interlock and then we have all basic functionality working on v1.3. Getting there!

wizath, 2019-03-07:
Sorry, but we're debugging odd current measurements on one board, so I'm a little busy

hartytp, 2019-03-07:
ack, thanks for the heads up.

wizath, 2019-04-15:
Added diagnostic command - [docs](https://github.com/sinara-hw/Booster/wiki/Remote-programming-interface).

closed 2019-04-15 by wizath

hartytp, 2019-04-15:
awesome! Thanks