sinara-hw/Booster #148: production feedback

gkasprow, 2019-02-07:
- [ ] C19 has 100pF in comment field but 100nF in PN.

gkasprow, 2019-02-19:
- [ ] L2, L3 Mouser 810-MLZ2012N1R0L or  Mouser 710-74479773210

gkasprow, 2019-02-19:
- [ ] PA_OUT : C3 footprint is 0603, PN is 0402

gkasprow, 2019-02-19:
- [ ] C5=CD4013BM96G4
U2=ADL5536ARKZ-R7, comment was ADL5535

gkasprow, 2019-02-19:
not only

hartytp, 2019-02-19:
Could you make a note of which PCB each change applies to, please?

gkasprow, 2019-02-19:
done

hartytp, 2019-02-19:
>     * PA_OUT: L2, L3 Mouser 810-MLZ2012N1R0L or  Mouser 710-74479773210

What's the motivation for this change?

gkasprow, 2019-02-19:
There is an error in the Mouser catalog and the coil we use has much lower current than specified. They specify it as 1A but in fact, it is 370mA.
Since it is an air coil, it cannot be saturated, but it happened twice that it was damaged during LDO breakdown. It could also be overheated causing Q degradation or rapid inductance changes due to shortcircuit. It could be responsible to some extent for strange behavior we observed.
Maciej recommended replacement.

hartytp, 2019-02-19:
good catch!

gkasprow, 2019-02-19:
- [ ] MAIN: IC7 in P&P is on top layer instead of the bottom.

gkasprow, 2019-02-21:
- [ ] wrong PN  for 1nF value: C19, C22, C26, C27
- [ ] wrong PN for 100nF value: C3
- [ ] wrong PN for 46R4 : R13 i R16
- [ ] wrong PN for 71R1 : R10
- [ ] wrong PN for 2n2: C10, C11, C12, C15, C16, C35

closed 2019-05-26 by gkasprow