sinara-hw/Booster #151: v1.3 testing

hartytp, 2019-03-06:
From an initial play, overall I really like the new v1.3 mechanics. All the concerns I had before seem to be fixed.

The only gripe I have right now is that the recess around the micro USB connector wasn't large enough to fit any connectors I could find in our lab, so I had to take a scalpel to one to get it to fit.

hartytp, 2019-03-06:
@wizath the scatter and drift on the bias currents looks very good for v1.3 so far. The channel-channel scatter is 48mA to 53mA, which I can imagine is due to temperature etc. The drift is quite low too. I also haven't seen any nasty drop outs yet.

We'll start stress-testing now.

hartytp, 2019-03-06:
Small-signal gain:

[image]

hartytp, 2019-03-06:
Small-signal cross-talk with channel 1 as the aggressor:

[image]

hartytp, 2019-03-06:
The channel 5 cross-talk measurement looks a little odd, will double check that one.

Otherwise, that all looks really good. Until around 300MHz, we're limited by the measurement noise floor.

hartytp, 2019-03-06:
[v1.3.zip](https://github.com/sinara-hw/Booster/files/2937314/v1.3.zip)

Raw data and analysis scripts (with notes on the analysis method)

hartytp, 2019-03-06:
aah, for channel 5 I had the measurement bandwidth up, so the measurement noise floor was higher. Ignore that channel, it should be exactly the same as the others...

hartytp, 2019-03-06:
[image]

Here is the bias current trace for the last half hour (no RF applied). We'll keep logging over night (assuming the VCP doesn't crash). This looks significantly better than anything I saw with the older units, so hopefully things will be stable...

cjbe, 2019-03-07:
The bias currents at idle still look very stable overnight (the initial transient is from the device cooling down after a load test). This looks far better than anything we have seen with the earlier versions!

[image]

cjbe, 2019-03-08:
@wizath with the firmware you posted in https://github.com/sinara-hw/Booster/issues/147#issuecomment-470537145 I see an odd problem that did not occur with the firmware that was on the device when I received it.

I apply RF to all 8 channels that is just below the output power interlock threshold. This causes all the interlocks to trip (expected), but also triggers an 'OVC' error on 3 channels. Do you know what is going on here?
```
[ERROR] Interlock tripped on channel 0, i=0 o=1
[ERROR] Interlock tripped on channel 1, i=0 o=1
[ERROR] Interlock tripped on channel 2, i=1 o=1
[ERROR] Interlock tripped on channel 3, i=0 o=1
[ERROR] Interlock tripped on channel 4, i=0 o=1
[ERROR] Interlock tripped on channel 5, i=0 o=1
[ERROR] Interlock tripped on channel 6, i=0 o=1
[ERROR] Interlock tripped on channel 7, i=0 o=1
[ERROR] OVC on channel 4, current 0.00
[ERROR] OVC on channel 0, current 0.00
[ERROR] Interlock tripped on channel 1, i=0 o=1
[ERROR] Interlock tripped on channel 2, i=1 o=1
[ERROR] Interlock tripped on channel 3, i=0 o=1
[ERROR] Interlock tripped on channel 5, i=0 o=1
[ERROR] Interlock tripped on channel 6, i=0 o=1
[ERROR] Interlock tripped on channel 7, i=0 o=1
[WARN] DHCP failed after 33 retries, retrying
[ERROR] OVC on channel 7, current 0.06
```

wizath, 2019-03-08:
Interesting, checking diff

cjbe, 2019-03-11:
@wizath any idea what is causing this problem? Have you seen any errors like this?

wizath, 2019-03-11:
That's related to weird working of interlocks, I'm debugging it right now, since they go off way earlier than set threshold

wizath, 2019-03-11:
Found some issues, working of fixes

hartytp, 2019-03-11:
Great, thanks!

wizath, 2019-03-11:
Turns out that comparator in ADL5904 works in logarithmic scale, but power detectors scale linearly that's why interlocks went crazy.

hartytp, 2019-03-11:
@wizath My intention with the new design is that we should *not* need separate calibrations for the interlock and power detector. I hope that we can use exactly the same calibration both for measuring the power and for setting the interlock.

My assumption here is that the signal from the power detector (20mV/dB) is large enough that we can ignore offsets etc in the ADCs + DACs.

Let's start with this simple approach and only use a second calibration table for the interlock if absolutely necessary.

wizath, 2019-03-11:
Yes, that's correct - since we're using common source, we could calibrate it once and use in two ways - interlocks and power measurements.

hartytp, 2019-03-11:
:+1: we just need to double check that there isn't some offset/gain error somewhere that effects the calibration. But, I would expect it to work without issue.

wizath, 2019-03-11:
Tested new interlock accuracy.  Error would be less in reality because that value is the last recorded before interlock going off with step of 0.1 dBm of input power.

| Interlock setpoint [dBm] | Output min [dBm] | Output avg [dBm] | Output max [dBm] | Error avg [dBm] |
|-------------------:|------------|------------|------------|-----------|
| 25                 | 24.18      | 24.20      | 24.24      | 0.80      |
| 26                 | 25.31      | 25.33      | 25.36      | 0.67      |
| 27                 | 26.35      | 26.37      | 26.41      | 0.63      |
| 28                 | 27.43      | 27.43      | 27.44      | 0.57      |
| 29                 | 28.46      | 28.49      | 28.50      | 0.51      |
| 30                 | 29.24      | 29.26      | 29.27      | 0.74      |
| 31                 | 30.25      | 30.27      | 30.30      | 0.73      |
| 32                 | 31.30      | 31.33      | 31.35      | 0.67      |
| 33                 | 32.38      | 32.39      | 32.40      | 0.61      |
| 34                 | 33.35      | 33.38      | 33.46      | 0.62      |
| 35                 | 34.41      | 34.42      | 34.43      | 0.58      |
| 36                 | 35.40      | 35.42      | 35.45      | 0.58      |

hartytp, 2019-03-11:
Wow, nice! Was that using the same calibration you'd previously set up for the power measurements?

What's interesting is that the error is nearly constant. Could that be because you calibrated the power measurement a while ago and are using slightly different cabling in the interlock measurement? In any case, 0.5dB is pretty good for RF (e.g. most 30dB attenuators that I would use for this measurement are only specified to 1dB).

hartytp, 2019-03-11:
Also, if there is a fixed offset between the power measurements and the interlock tripping, is that the same on all channels or just some random offset (e.g. a DAC offset)?

hartytp, 2019-03-11:
In any case, can you post firmware with this fix included? Thanks!

wizath, 2019-03-12:
I'll check the input interlock quickly and post you update

hartytp, 2019-03-12:
Great thanks

cjbe, 2019-03-12:
@wizath poke! It would be very useful to have this updated firmware so we can finish testing the v1.3 Booster. We need to get this testing finished and get this in use in the lab ASAP - the current delay is hurting us.

wizath, 2019-03-12:
Sorry, I was fixing new issue (https://github.com/sinara-hw/Booster/issues/158) with interlocks. I'm attaching new firmware that's dedicated to v1.3.

You need to re-calibrate the interlocks - use something like:
clear interlock using `clearcal <ch_num> 4095`,
set output power to 0 dB then type `intcal <ch_num> 1 0.0` in VCP console,
then set output power to 36 dB then type `intcal <ch_num> 2 36.0`

cjbe, 2019-03-12:
@wizath why does this have a separate calibration table? I thought we had agreed that we should be able to use the same calibration for both the output power ADC and the output power interlock?

hartytp, 2019-03-19:
This issue's got a bit long. Let's break it down to smaller issues for further tests

closed 2019-03-19 by hartytp