sinara-hw/Booster #152: v1.3 recess around micro USB not big enough to fit most connectors

hartytp, 2019-03-06:


gkasprow, 2019-03-06:
was it better in the previous mechanic's version? I have no control over Technosystem mechanics.

hartytp, 2019-03-06:
@gkasprow I had fewer problems with the previous version. However, as I said, I believe that the design was probably the same and the only difference was how tightly the screws were done up.

> I have no control over Technosystem mechanics.

How should we make sure this is fixed for the next revision? Should I email TechnoSystem directly, or will they see this issue?

gkasprow, 2019-03-06:
I think we have to notify them by email.

hartytp, 2019-03-06:
Okay, thanks!

Assuming this issue doesn't apply to Creoetch mechanics, I'll just write to Technosystems and close the issue.

closed 2019-03-07 by hartytp