sinara-hw/Booster #153: Input SMA D coutouts

cjbe, 2019-03-06:
@gkasprow are there definitely D-shaped cutouts for the input SMAs in booster v1.3? One of the smas on Booster front panel came loose when I did up the cable...

cjbe, 2019-03-06:
@gkasprow after disconnecting and reconnecting the inputs finger tight a few times 5 out of the 8 input SMAs are loose.

hartytp, 2019-03-06:
:) I think this must have slipped through the net of mechanical fixes in the previous version.

gkasprow, 2019-03-06:
is the newest release affected as well?

hartytp, 2019-03-06:
Yes, the v1.3 unit we just received is the worst yet in terms of the SMAs. Just measuring the S21 and cross-talk 3 of the SMA channels started rotating. We're going to have to take it to pieces to tighten the SMAs up more.

hartytp, 2019-03-06:
> Yes, the v1.3 unit we just received is the worst yet in terms of the SMAs

(i.e. I think it's the same mechanics as the previous units, but probably not tightened up quite as much).

closed 2019-03-07 by hartytp