sinara-hw/Booster #154: interlock status

cjbe, 2019-03-06:
@wizath currently the input interlocks trip at about -15dBm input power (+25dBm output), so we can't do anything with the Booster. Any updates on getting the interlock programming working with the v1.3 hardware?

wizath, 2019-03-06:
I'll send you fix tomorrow, since I'll get more RF channels. You can use command `clearcal <ch_num> <value>` as quickfix for setting interlock values for both interlocks in order to test them.

hartytp, 2019-03-08:
@wizath we need the interlock to make use of booster in our lab. Do you have any update on when the new interlock might be ready?

Thanks again for all the work on this, and apologies for chasing

wizath, 2019-03-08:
I'm testing them at the moment, but they behave a little odd - need to figure that out.

cjbe, 2019-03-11:
@wizath any progress on getting this interlock working?

closed 2019-03-19 by hartytp