sinara-hw/Booster #156: Use MAC address from EEPROM

cjbe, 2019-03-06:
The v1.3 Booster we have received seems to have a non-unique MAC address (0008dcabcdef) - shouldn't there be a unique MAC address in the EEPROM?

wizath, 2019-03-06:
It should, I'll check it tomorrow

wizath, 2019-03-06:
Ahh, last issue was that mainboard EEPROM was not equipped with UUID feature. But anyway if nothing is saved in EEPROM, it should use CPU UUID for setting MAC address.

Could you check if MAC address is equal to bytes [6:11] from UUID? VCP command is `devid`

cjbe, 2019-03-06:
```
> devid
[devid] 002600213436470832383539
```
and
```
> ifconfig
[log] ===== W5500 NET CONF : Static =====
[log]   MAC:    00:08:DC:AB:CD:EF
[log]   IP:     192.168.1.10
[log]   GW:     192.168.1.1
```

wizath, 2019-03-06:
Ok, found it. That's related to DHCP client bug that fails after some time and is programmed to set default settings with this MAC address. I'll change it to try indefinitely when error occurs and it should fix it.

wizath, 2019-03-06:
btw, in v1.3 new EEPROM was added with UUID-64 so i'll add support for reading that into MAC address.

cjbe, 2019-03-06:
Awesome - thanks!

wizath, 2019-03-29:
https://github.com/sinara-hw/Booster/issues/98#issuecomment-478037725
Fixed. Could you check if this is working on your device, since there is an issue with EEPROMS (https://github.com/sinara-hw/Booster/issues/187)

hartytp, 2019-04-09:
What do you want me to check?

wizath, 2019-04-09:
You can try to set custom MAC address then reboot and check if it was properly saved.

hartytp, 2019-04-09:
Should using the default MAC address from the EEPROM work now?

wizath, 2019-04-09:
It depends on hardware. If there are 2 EEPROMs mounted on mainboard there could be a problem.

hartytp, 2019-04-09:
How do I find out?

wizath, 2019-04-09:
Give me default MAC address that's displayed

hartytp, 2019-04-09:
how do I query that?

wizath, 2019-04-09:
ifconfig

hartytp, 2019-04-10:
ifconfig
[log] ===== W5500 NET CONF : DHCP =====
[log]	MAC:	80:1F:12:56:19:BC
[log]	IP:	0.0.0.0
[log]	GW:	0.0.0.0
[log]	SN:	0.0.0.0
[log] =======================================

hartytp, 2019-04-10:
Is that what you expect?

wizath, 2019-04-10:
Looks good. Can you change MAC address (using `macaddr` command like `macaddr 11:22:33:44:55:66`) then reboot and check if it is saved properly. Then revert using `macaddr default`.

hartytp, 2019-04-10:
@wizath okay, the mac address is saved across reboots.

However...
```
> macconfig default
[macconfig] Default MAC address set
> 
> ifconfig
[log] ===== W5500 NET CONF : DHCP =====
[log]	MAC:	36:34:38:38:32:32
[log]	IP:	0.0.0.0
[log]	GW:	0.0.0.0
[log]	SN:	0.0.0.0
[log] =======================================
> 
```

when I reset it to the default address I get something odd...

hartytp, 2019-04-10:
also, cosmetic issue but `MAC address 80 set` is an odd comment. Shouldn't it print the whole MAC address there?

```
> macconfig 80:1F:12:56:19:BC
[macconfig] MAC address 80 set
> 
> ifconfig
[log] ===== W5500 NET CONF : DHCP =====
[log]	MAC:	80:1F:12:56:19:BC
[log]	IP:	0.0.0.0
[log]	GW:	0.0.0.0
[log]	SN:	0.0.0.0
[log] =======================================
> 
> 
```

wizath, 2019-04-10:
I'll add it to my list

hartytp, 2019-04-10:
Since `80:1F:12:56:19:BC` does not correspond to any vendor, I'd guess that it's something random that someone at Creotech picked...

wizath, 2019-04-10:
Hmm, maybe there is an issue with that dual EEPROM. But I can't confirm it right now.

gkasprow, 2019-04-10:
Isn't it MAC address taken from EEPROM EUI ?

wizath, 2019-04-10:
> Isn't it MAC address taken from EEPROM EUI ?
It is

hartytp, 2019-04-10:
Okay, so @gkasprow the big question is how do we get @wizath more hardware to test???

hartytp, 2019-04-13:
closing as duplicate of https://github.com/sinara-hw/Booster/issues/187

closed 2019-04-13 by hartytp