sinara-hw/Booster #157: measurement glitches

hartytp, 2019-03-07:
I have now seen one event where the 28V bias current reported on one channel decreased by a factor of 2 for one sample. I'd guess this is just a measurement error, so it looks like the measurement glitch issue is very nearly but not totally solved now...

hartytp, 2019-04-09:
@wizath did you manage to reproduce this?

hartytp, 2019-06-03:
@wizath were you able to reproduce these glitches?

wizath, 2019-06-05:
Nope, but I left it running for some time, to observe if any I2C error occur and came to conclusion that these glitches are not related to communication errors.

hartytp, 2019-06-05:
Do you mean that you didn't see any glitches on any of the voltages/currents we monitor? Or that you did see glitches, but you think they're an analog issue (EMI or actual glitches in the voltages/currents we're looking at)?

wizath, 2019-06-05:
Second option I suppose

hartytp, 2019-06-05:
How big are the residual glitches you're seeing, and which lines are they on? Do you have a log trace I can see? Did you probe the signal line on a scope in persist mode to check whether these glitches are present on the signal or not?

Also, have you ruled out things like ADC configuration errors (IIRC previously wasn't part of the problem due to the timing of the ADC samples or something like that?)

hartytp, 2019-06-05:
@wizath cool, so this is the last known software issue :)

hartytp, 2019-06-06:
@wizath did you have a chance to probe the signals with a scope and see if the glitches are present on the signals, or just some EMI issue? Can you post a debug log from booster and a scope trace that we can compare?

hartytp, 2019-06-06:
@wizath did you have any luck looking at this? This is the last non-cosmetic issue with Booster, so I'd really like to get it nailed down asap.

hartytp, 2019-06-09:
@wizath can you look at this on Monday? I need to get Booster wrapped up soon for my sanity (guessing you feel the same way). Let’s get these final issues done and move on to other things :)

hartytp, 2019-06-13:
@wizath did you manage to get a trace showing these glitches?

wizath, 2019-06-13:
Left booster running without RF for a night - here's my observations:

Apart from one channel that dropped 5 mA in bias current there are no obvious glitches in current reading (maybe these little ones on green and purple?)

[image]

As for power readings, there is a one noisier channel, but apart from that, also not really any spikes
[image]

hartytp, 2019-06-13:
Not sure why that one channel is more noisy than the others, but there is nothing there that concerns me at all.

That's really wonderful, THANK YOU!

closed 2019-06-13 by hartytp