sinara-hw/Booster #158: v1.3 Interlock issue

wizath, 2019-03-12:
As I was testing new hardware interlocks I came up with interesting issue. When you ramp up input power and it's crosses with set-point, the interlock goes off. But in a case where you reset interlock without touching input power, device resumes it's operation.

But that's the problem - current interlock will react only on rising edge when power will cross the set-point, but won't react on continued difference between levels because input from comparator goes to the flip-flop which outputs the interlock signal.

There is a need to append an OR-gate at the flip-flop output in order to include the signal from comparator even after the flip-flop is set by rising edge. I'll fix it within software, but that fix is needed to be included in supplementary board next revision.

@update
Software fix will trip in about 60ms. Since there is a backup solution, guarding for resetting interlocked channel while power still is exceeding interlock value I think that'll do?

[image]

hartytp, 2019-03-12:
@wizath good catch.

Okay, yes, for now let's implement a software fix (don't allow the user to reset the interlocks if the signal level is still too high). We can add an OR gate to existing boards and to the next revision easily.

@gkasprow what do you think about this?

hartytp, 2019-03-12:
> Software fix will trip in about 60ms. Since there is a backup solution, guarding for resetting interlocked channel while power still is exceeding interlock value I think that'll do?

Might be better to fix by not allowing the user to reset the interlock while the power is high.

wizath, 2019-03-12:
> Might be better to fix by not allowing the user to reset the interlock while the power is high.

You can't measure output power when interlock tripped, since the output power detector is located after the RF switch. We can also use the input power detector, but that's another topic.

gkasprow, 2019-03-12:
There is already 0R resistor on the DFF output. So we can mount 1k one and add Schottky diode between CLK input and this resistor and make a poor-man pseudo-OR gate :)

gkasprow, 2019-03-12:
But if we can live with this bug and software fix, I'd leave it to the next revision

hartytp, 2019-03-12:
Let’s fix it by adding the diodes

hartytp, 2019-03-12:
@gkasprow Let me check I understand this
- currently, the interlock latch is only sensitive to rising edges on the comparator output
- when the interlock triggers, the comparator output goes low (RF switch open)
- when the interlock is reset the comparator output goes high again (RF switch closes)
- currently, there is a hw delay set up to make sure that the DFF reset occurs some time after the hw switch
- as a result, the comparator output has already gone high by the time the DFF reset triggers, so the interlock does not re-trigger (it doesn't see a rising edge on the data pin)
- if we add an OR gate at the DFF output then I guess we get a situation where when the interlock is reset the switch opens and closes repeatedly until the DFF latches it (given that the power detector has a slightly slow response time compared with the RF switch, I'm not sure what the dynamics of this will actually look like). This seems like a slightly nasty solution.

Correct me if I'm wrong here, but my understanding of the problem here is that the real issue is the ordering of the interlock reset and the switch opening. Ideally, we want the DFF reset to occur *before* the switch re-opens. In other words, if we just DNP C20, won't that fix the problem.

In the previous hardware revision, we needed C20 to avoid the interlock re-triggering due to fast (~ns) transients when the RF switch opens. However, in this hardware revision, (a) these transients are much smaller since we fixed the switch control voltages (b) the interlock response time is slower (microseconds) and so should not re-trigger due to these fast transients. So, I think we can scrap C20.

Do you agree with all of that? If so @wizath can you try removing the software interlock you added and remove C20 on the supplemental control board and see if that fixes the issue.

gkasprow, 2019-03-12:
@hartytp we discussed today an idea to scrap C20, but AFAICT there will be some transients. However, this needs checking if they are big enough to trigger the protection.

hartytp, 2019-03-12:
In particular, if we go for the OR solution then I think we need to be careful to make sure that the DFF triggers faster than the OR gate, otherwise we could find some situation where the system just oscillates without the interlock ever latching.

hartytp, 2019-03-12:
@gkasprow I think this is something we can test quickly: remove C20 and see what the lowest interlock threshold we can set is that doesn't re-trigger when the switches are opened.

hartytp, 2019-03-12:
We measured the raw switch transients on an eval board here 
https://github.com/sinara-hw/Booster/issues/93#issuecomment-423493610 In this data we have approx -20dBm going into the switch at 200MHz. That's equivalent to -14dBm at Booster's input due to the input attenuators, so it's what we would get with an output power of approx +26dBm (after Booster's gain). So, all things considered, this is quite a low input power, so a good test for glitches.

You can see that with the 560pF filter on the switch pins, the transients becomes smaller is definitely still there. 

We then tried putting the eval board at the input to Booster to see what happens at Booster's output once the transients have been shaped by Booster's gain. Interestingly, the transients were no longer visible. I believe the reason for this is that Booster has a high-pass response with a sharp cut-off at around 50MHz. With the RC filter on the switch pins, the transients become too slow to pass through Booster. 

So, this may work.

Anyway, we did not check that this still works at higher powers once Booster is starting to saturate, but it's quick to test and would be the simplest solution.

hartytp, 2019-03-12:
Also, however we fix this, we do need this fixed (including on the Creotech units). This is a critical piece of functionality for us (prevents users destroying dozens of AOMs with a simple programming error) so we do not want to rely on firmware interlocks alone.

hartytp, 2019-03-12:
FWIW the other aspect of this is that in the previous version, these glitches caused real issues for the interlock because the interlock response time was so fast (10s of ns IIRC). This also caused issues with things like the Urukul switches, which show similar glitches.

Part of the motivation of using the external comparator interlock is that we can make the response time a bit lower (a few us in the current hw IIRC). Even with the 560pF capacitors, the glitch time is only a couple of hundred ns, so the glitches will be greatly reduced by the interlock response time. So, hopefully, the hysteresis we see due to these glitches will be quite small...

hartytp, 2019-03-13:
@wizath did removing C20 work?

wizath, 2019-03-13:
Yes and no. Interlock will stay enabled if you don't touch input power, but if you increase it, it'll reset and continue working.

hartytp, 2019-03-13:
? I don't follow. Without C20, I expect the following to happen:
- interlock trips
- RF switch opens
- comparator output goes low
- interlock reset goes high
- DFF resets
- RF switch closes
- copmarator output goes high
- DFF latches

What error are you seeing with C20 removed?

hartytp, 2019-03-13:
@wizath what's the sate of this? I don't think I understand your previous post about the issue.

hartytp, 2019-03-14:
ping @wizath can you explain a bit more what the problem is here?

wizath, 2019-03-14:
My observations after removing C20 and disabling software protection:

- enable channel
- ramp up power to trigger the interlock, works fine
- try to clear interlock, it works just like the old version, channel is instantly interlocked again
- (by curiosity) set input power a little bit higher while interlock is on
- interlock reset works fine, channel continues to work without any interlock

I had a little time yesterday to debug this, will post detailed measurements tomorrow

hartytp, 2019-03-14:
hmm..odd.
From your measurements, can you tell me which of the following assumptions is not correct:
- before resetting the interlock, the DFF input (comparator output) is LOW
- the DFF is reset before the RF switch opens
- the DFF input goes high again after the switch opens

AFAICT at least one of those must be false. If we know which then this should be easy to fix.

hartytp, 2019-03-14:
We need to fix this asap, so any data you have on this would be a huge help

wizath, 2019-03-15:
Legend:
yellow - flip-flop output
green - comparator output
blue & pink - voltage setpoint & voltage from power readout

1) Interlock going off first time while increasing power
[image]

2) Resetting interlock while not touching input power
[image]

3) Increasing input power then clearing the interlock
[image]

Seems that flip-flop don't register comparator event even with C20 removed. Comparator works fine, so adding OR gate should work just fine.

[image]
Also there, more detailed interlock reset procedure with constant input power
yellow - output power
blue - comparator output
pink - flip flop output
green - reset line

gkasprow, 2019-03-15:
These oscillations look very suspicious.
If the oscillations are higher than a certain frequency, the DFF output state may be random.

The DFF needs a steep signal edge on its clock input. When the edge is too slow, then its output will go high when the clock is high, then will get back to the low state when the clock is low. I observed once such behavior. 
What surprises me is the fact that the DFF output toggles more than 1us after the leading edge of the comparator output! Maybe the DFF is lacking its power supply and gets supplied by the signal ?

gkasprow, 2019-03-15:
It's quite possible that the CD4013 performance at 3V3 is so poor that it causes the issues. Worth checking with SN74HC4013 or something with better performance.

hartytp, 2019-03-15:
> These oscillations look very suspicious.

@wizath am I right in thinking that the oscillations on the comparator output are due to the fact that the power here is right on the interlock threshold, so the comparator toggles due to power detector noise? i.e. if you used  power far above the interlock threshold then the oscillations would disappear? @gkasprow is it worth adding a small amount of hysteresis (positive feedback or use a comparator with Schmidt trigger) to avoid this?

@wizath my guess is that the oscillations we see on the other lines are a grounding/measurement issue with the scope probe -- the other signals probably don't oscillate like that, but there is a common-impedance coupling through the scope grounding or something like that?

hartytp, 2019-03-15:
@gkasprow AFAICT the real issue here is that the DFF reset line takes over 2us to transition from high to low! That explains everything we see, I think. If that was fast, I don't think we'd see any problems. Why is it so slow?

hartytp, 2019-03-15:
I think it's important that we check these oscillations are not a problem. @wizath on Monday, please can you repeat this measurement with an input power 2dB above the threshold and verify that there is no oscillation?

hartytp, 2019-03-15:
@gkasprow @wizath there seems to be a lot of cross-talk in these measurements between the digital and analog signals. I expect that this is just a measurement artefact, but it could also be very poor grounding on the supplemental control PCB. Please can you repeat the measurements with better grounding and check that there aren't ground issues. If there are ground issues then all kinds of things could go wrong (e.g. the FET bias voltage could jump when the interlock resets).

hartytp, 2019-03-15:
@wizath is the green (DFF reset line) the output of IC1A (SN74LVC32AD)? That should have a rise time of ns, so why on earth is it taking us to rise???

hartytp, 2019-03-15:
@gkasprow what voltage is IC1 supplied from?

hartytp, 2019-03-15:
Okay, so I think the slow rise/fall time on that green trace is the real issue here. AFAICT, it's probably related to the incorrect supply voltage on that chip.

@wizath If I'm right about that, please can you
1. Replace IC1 and IC5 with new components (since they're probably damaged)
2. Cut the trace and supply IC1 from 3V3
3. Repeat the measurement (still with C20 removed)

Thanks!

hartytp, 2019-03-15:
If I'm not right about the incorrect power supply then we still need to investigate this, since that component is clearly not behaving as expected so there is something wrong with the circuit...

hartytp, 2019-03-15:
Hmmm I don't think the power detector inputs can take 5V either, so that might also damage the detectors...

gkasprow, 2019-03-15:
IC1 and IC2 are supplied from 3V3 already.
IC7 is supplied from 5V to deliver correct  control levels to the RF switch
the  IC1A  rise time is soo poor because on the power stage board the R2 is mounted and C18 (47nF) shorts the signal to GND.  Just remove either R2 or C18.

hartytp, 2019-03-15:
Aah, good catch @gkasprow! What is the point of C47 anyway? It seems odd to have such a large capacitor on the reset line.

hartytp, 2019-03-15:
Hopefully if you remove R2 and replace C18 with 0R (to keep the reset line in a well-defined state) the interlock will work properly. @wizath can you try this and repeat your measurement on Monday?

Thanks!

wizath, 2019-03-18:
[image]

Blue & pink - comparator inputs
Yellow - flip-flop output
green - comparator output

With increased power:
[image]

Only the IN_PWR voltage changes, circuit works fine

[image]

As you can see, it takes about 1.6 us to trip the interlock again.

After removing C18 (or whatever C28 is on shematics) from PA_Stage board, interlock works just fine with or without increasing input power

wizath, 2019-03-18:
But...

hartytp, 2019-03-18:
Why is is hard to reset the interlock with c20 removed? I don’t follow.

hartytp, 2019-03-18:
Is blue the set point voltage and pink the power detector output?

hartytp, 2019-03-18:
> But... since reset signal is so fast, it's hard to reset interlock if you lower input power just a bit (1-2 dBm) and it's super annoying

Is the issue here that the interlock is re-triggering from overshoot in the circuit when you turn it on? If that's the problem them we can fix it easily by increasing C27/C26 a bit to increase the power detector time constant. Increasing that time constant to ~10us to avoid triggering on small fast transients could be a good idea.

wizath, 2019-03-18:
I'll try replacing C27 to smth like 50 nF and check.

wizath, 2019-03-18:
[image]

With 50 nF, reset sequence works like that.
blue - setpoint voltage, pink - VRMS
yellow - FR_RST signal
green - comparator output (noisy because is connected by long wire)

with interlock triggering at -10dBm input power, it can clear it at -11.4 dBm~.

hartytp, 2019-03-18:
@wizath can you post an image with the DFF output as well?

Anyway, that seems to work then. Good job!

hartytp, 2019-03-18:
The one thing that I still don't understand is these oscillations we see on all the traces. When I looked at this just now both the comparator input and output seemed to have huge oscillations. I don't think they are just pickup (it's over 2V in your plot). Any idea what's oscillating?

hartytp, 2019-03-18:
@cjbe @wizath I wonder if the comparator oscillations we see are due to the OpAmp buffers ringing. For 100pF load capacitance the OpAmps need a 100Ohm isolation resistor (R17 on the PA board). IIRC those OpAmps drive the microprocessor ADCs so could be quite a bit of capacitance.

wizath, 2019-03-18:
[image]

Well, about that oscillations, It get's worse when power level is close to the interlock set-point voltage. I wonder how worse it is due to the long cable that I used to output the comparator signal outside the board since it's enclosed in the case.

hartytp, 2019-03-18:
@wizath after fixing the comparator oscillations and the reset capacitors, we now see this

[image]

- Yellow is comparator input
- Blue is comparator output/DFF input
- Purple is DFF output

So, that works. However, https://github.com/sinara-hw/Booster/issues/174 seems to happen very often when the interlock trips. Any ideas why??

hartytp, 2019-03-18:
> Well, about that oscillations, It get's worse when power level is close to the interlock set-point voltage. I wonder how worse it is due to the long cable that I used to output the comparator signal outside the board since it's enclosed in the case.

I think this is because when the two inputs are nearly equal the comparator acts like a very high gain linear amplifier so is prone to oscillations (it's a very fast comparator). NB there is still a fair bit of RF on the comparator input left from the power detector, which can also help to knock the comparator from one state to another.

The positive feedback fixes this. A comparator with Schmidt trigger would also do the job.

wizath, 2019-03-19:
> @wizath after fixing the comparator oscillations and the reset capacitors, we now see this

What changes you applied to the circuit?

hartytp, 2019-03-19:
> @hartytp What changes you applied to the circuit?
1. Added positive feedback across the comparator to suppress oscillations (100R and 1nF to give a 1 MHz pole IIRC)
2. Removed the capacitor on the DFF reset
3. Cut the traces that connect the supplemental control board reset line to the power detectors. Shorted the power detector control lines to GND. It should also be sufficient to just DNP the large capacitors on those lines.

After doing that, everything worked as expected.

hartytp, 2019-03-19:
Closing this issue, since the hardware fixes required to make the interlock work are documented in other issues.

closed 2019-03-19 by hartytp

hartytp, 2019-03-19:
@wizath apologies, there was one other hardware modification I forgot to mention: I also increased the power deterctor RMS averaging capacitor to 10nF.

gkasprow, 2019-03-19:
Cannot we simply add OR gate in next revision of HW ?

hartytp, 2019-03-19:
> Cannot we simply add OR gate in next revision of HW ?

@gkasprow No, I don't think we can.

Firstly, because the DFF is quite slow if we just add an OR gate at the DFF output then the OR will trigger before the DFF does, so the output will just oscillate forever without the DFF triggering.

Secondly, even with the OR gate, I think we would need to fix the comparator oscillations. Currently there is too much coupling between the fast comparator output and the power detector (and all other traces on the supplemental control board). These oscillations cause a lot of hysteresis and other nasty effects. So, whatever we do, we must fix this. The positive feedback I added (1nF + 100R) is a simple way of doing it.

Once we've added the positive feedback, the only other part of my fix is DNPing a few capacitors. That seems much simpler than adding an OR gate and will work more reliably.

...so...why would you want to use an OR gate instead?

Anyway, since this issue pretty much breaks the interlock in the current v1.3 design, I'm afraid it must be fixed on all units. We've already patched all 8 channels on the box Technosystem sent us and they now work excellently (we'll post data soon).

gkasprow, 2019-03-19:
OK, since this works and you are happy, we will apply the same fixes.

hartytp, 2019-03-19:
So, the fix is:
1. Add positive feedback from comparator outputs to their inputs (100R + 1nF)
2. DNP C20 that adds the delay on the DFF reset (we want the DFFs to reset before the RF switch opens)
3. Change the RMS averaging capacitor on all power detectors to 10nF.
4. DNP PA R2/R7 and replace C28/C18 with 0R (same for preamp board) to tie the power detector latch reset low

Isn't that simpler than adding an OR gate? Anyway, it's now tested and works, and it would take more time to test out a new solution.

@gkasprow also note that increasing the RMS averaging capacitor is important anyway. Otherwise, we had problems using Booster with Urukul since fast transients on the Urukul RF switch tripped the interlock. 10nF gives a 10us response time which is fast enough to protect our loads, but slow enough to avoid triggering on short/harmless spikes.

hartytp, 2019-03-19:
> @gkasprow also note that increasing the RMS averaging capacitor is important anyway. Otherwise, we had problems using Booster with Urukul since fast transients on the Urukul RF switch tripped the [...]

i.e. even with the OR gates, we would need to do (1) and (3) so adding (2) and (4) seems simpler and more robust than the OR gate.

cjbe, 2019-03-22:
With these hardware mods the interlock works very very nicely.
With the output interlock set at +30 dBm, a pulse of >4us duration at +31 dBm will trip the interlock, and with pulses at +29 dBm I have seen no interlock trips in 200k samples

wizath, 2019-03-22:
Great, I'll let Technosystem to modify rest of the channels and will prepare second unit for shipping

[1 more reply]