sinara-hw/Booster #159: Disconnected channels throwing OVC errors

cjbe, 2019-03-12:
Occasionally disconnected / unused channels show OVC errors. I saw the below errors while scanning the power on channel 0 - channels 5-7 have no input RF and have their outputs terminated. The OVC errors occurred at the same time as the interlock tripped on channel 0.
```
[ERROR] OVC on channel 5, current 0.02
[ERROR] OVC on channel 6, current 0.02
[ERROR] OVC on channel 7, current 0.02
[ERROR] Interlock tripped on channel 0, i=0 o=1
```
(As an aside, it would be useful to have timestamps on these log entries, e.g. [103.234s] [ERROR] ...)

hartytp, 2019-03-15:
I'll be really interested to see what the cause of this is. Is it a false error? Or, are these channels actually seeing an over-current condition. I wonder if this is related to the pile of dead FETs.

wizath, 2019-03-15:
We tested OVC limits yesterday, they work fine and cut channel if they exceed 500mA's. I've reproduced this issue but had no time to look at it closely.

hartytp, 2019-03-15:
Great thanks. :)

cjbe, 2019-03-18:
This seems to have the same root cause as #174

closed 2019-03-18 by cjbe