sinara-hw/Booster #161: SCPI interlock level commands do not work

cjbe, 2019-03-14:
Via the SCPI interface:
Running `int:pow 1,33.0`, then `int:pow? 1` returns `-inf`.
Using the serial interface to set the interlock:
```
> intv 1 33
[intv] Interlock value for 33.00 = 3498
```
Then I get via the SCPI interface:
```
int:pow? 1
1163
```

This looks like the SCPI interface not using the correct units.

I calibrated the interlock as described https://github.com/sinara-hw/Booster/issues/151#issuecomment-472062801 :
The calibration (after taking quite a long time) find DAC values of 631 for 0 dBm output and 3792 for 36 dBm output - these seem sensible.

hartytp, 2019-03-14:
Yes, although I still don't see why we need a separate calibration for the SCPI interface anyway...

wizath, 2019-03-14:
Ack. Could you try intv command from vcp? Probably there is still old method in scpi command. I'll fix it tommorow

cjbe, 2019-03-14:
@wizath yes - the intv command seems to set the interlock power correctly

wizath, 2019-03-14:
> Yes, although I still don't see why we need a separate calibration for the SCPI interface anyway...

That's not the separate calibration method, just need to unify calculation method between commands and hardware revisions. I'll join calibration methods for TX power readout and interlock calibration shortly.

hartytp, 2019-03-14:
Okay, thanks for the explanation. That makes sense.

wizath, 2019-03-19:
Fixed https://github.com/sinara-hw/Booster/issues/98#issuecomment-474332047

hartytp, 2019-03-19:
Great, thanks!

closed 2019-03-19 by hartytp