sinara-hw/Booster #164: comparator oscillations

hartytp, 2019-03-15:
https://github.com/sinara-hw/Booster/issues/158#issuecomment-473439371

- [ ] verify that oscillations vanish when a larger input signal is used
- [ ] consider using a comparator with hysteresis in the next revision

hartytp, 2019-03-18:
I believe that these oscillations are real and are responsible for the remaining odd effects we see in the interlock. Looks like the DFF is too slow and the comparator is too fast and missing hysteresis .

hartytp, 2019-03-18:
@gkasprow looking on a scope it seems that there is some parasitic coupling between the comparator output and its input (the power detector output). It oscillates by a few volts at a high speed which is too slow for the DFF to latch.

hartytp, 2019-03-18:
@gkasprow what's your preferred solution to this?

hartytp, 2019-03-18:
How about AC couple some positive feedback across the comparator? Maybe 1nF + 100R? Or just switch to a slower comparator + faster DFF. We'll add the hysteresis RCs now and see if it works...

gkasprow, 2019-03-18:
DFF won't work at all with a slow comparator. Some hysteresis is a good option. Anyway, only first transition matters so we don't really care about the oscillations.

hartytp, 2019-03-18:
We added 100R + 1nF positive feedback and the interlock now seems to work!

hartytp, 2019-03-18:
NB though, since we don't have a separate buffer for the comparator, this adds a large spike onto the signal measured by the ADC whenever the comparator switches.

hartytp, 2019-03-18:
Might be nice to either use a different comparator with more inbuilt hysteresis, or to add a 1k isolation resistor between the power detector and the comparator input to reduce the amount of coupling between the positive feedback and the power detector output.

hartytp, 2019-03-18:
@gkasprow one final thing about this: I am still a bit surprised that the comparator oscillations couple onto the other signals at such a high level. 

- [ ] Please can you double check that there isn't an issue with grounding/decoupling on this PCB?

gkasprow, 2019-05-26:
we have a solid GND layer.
Hysteresis solved the oscillations issue.
I added also 0R resistor at the comparator input to play with hysteresis

closed 2019-05-26 by gkasprow