sinara-hw/Booster #165: IC1 supply voltage

hartytp, 2019-03-15:
@gkasprow I don't have access to altium right now, so I may be wrong about this, but AFAICT IC1 is supplied from 5V0 even though it's only specified to operate up to 3.6V That's still below its absolute max rating, but doesn't seem like a good idea.

More importantly, that means that we are driving the DFF RESET input with 5V even though it is only supplied from 3V3. Since it's not a 5V tolerant input, this is likely to lead to component damage.

I suspect that this explains the issues we've seen with the interlock...

gkasprow, 2019-03-15:
IC1 is supplied from 3V3. At least on PCB and schematics.

hartytp, 2019-03-15:
Okay, good. I must have followed the traces on the PDF incorrectly then.

hartytp, 2019-03-15:
Thanks for confirming!

closed 2019-03-15 by hartytp