sinara-hw/Booster #173: interlock calibration

hartytp, 2019-03-18:
@wizath after looking at the schematic a bit more carefully I think it probably is worth keeping separate calibration tables for the interlock and power detectors. Otherwise there is quite a large error due to the DAC.

Obviously, we could use a more accurate DAC, but since we're calibrating things anyway it doesn't seem worth it.

What would be really nice is if there was a single command to calibrate both the power detector and interlock in one go.

cjbe, 2019-03-21:
Calibrating the power meter using the built-in calibration command, and manually calibrating the interlock (cf #177) everything seems to work very nicely:
```
Testing channel 0 ...
Gain:
39.6 dB (at P_out = 20.0 dBm)
38.7 dB (at P_out = 30.0 dBm)
38.3 dB (at P_out = 35.0 dBm)
Internal power meter:
f=100 MHz: error -0.4 dB at P_out = 20.0 dBm
f=100 MHz: error 0.0 dB at P_out = 30.0 dBm
f=100 MHz: error 0.1 dB at P_out = 35.0 dBm
f=250 MHz: error 0.1 dB at P_out = 20.0 dBm
f=250 MHz: error 0.3 dB at P_out = 30.0 dBm
f=250 MHz: error 0.6 dB at P_out = 35.0 dBm
Interlock set at 25.0 dBm (200 MHz) -> trips at 25.1 dBm (stddev/min/max = 0.04, 25.04, 25.14 dBm)
Interlock set at 30.0 dBm (200 MHz) -> trips at 30.1 dBm (stddev/min/max = 0.10, 29.89, 30.15 dBm)
Interlock set at 35.0 dBm (200 MHz) -> trips at 34.9 dBm (stddev/min/max = 0.05, 34.78, 34.93 dBm)
```

hartytp, 2019-03-21:
🎆

cjbe, 2019-03-22:
Using the (now fixed, #177) built-in output interlock calibration the performance is slightly worse than this, and looks like an offset problem:
```
Interlock set at 25.0 dBm (200 MHz) -> trips at 24.7 dBm (stddev/min/max = 0.02, 24.66, 24.70 dBm) 
Interlock set at 30.0 dBm (200 MHz) -> trips at 29.6 dBm (stddev/min/max = 0.01, 29.61, 29.65 dBm) 
Interlock set at 35.0 dBm (200 MHz) -> trips at 34.4 dBm (stddev/min/max = 0.07, 34.27, 34.44 dBm) 
``` 
I am calibrating at 25 dBm and 35 dBm.

@wizath are you compensating for the quantization effects in the calibration? In my calibration test code I found I needed to correct for the small quantization error in the gradient term by adjusting the offset to give me no error at ~ 30 dBm.

That is, if our calibration is `y = ax + b`, and then we quantise `a` (it is an int around 80) the errors will increase as `x` goes away from zero. If we adjust b so that this equation gives us the correct result with the quantised `a` around `x = +30 dBm` then we can significantly reduce this effect in the regime we care.

hartytp, 2019-03-22:
If you're doing fixed precision maths, maybe it's better to just multiply all the coefficients by, say, 1000 and only do the division at the last minute.

hartytp, 2019-03-22:
(Not sure if you're using 8 bit or 16 bit numbers, but presumably we can just make everything 32 bit without issue?)

hartytp, 2019-04-02:
@wizath did you manage to implement a more accurate calibration algorithm? Also, out of curiosity, why didn't you just use float32 for everything? It's not like speed/memory are an issue here and it would have avoided this kind of issue completely.

Other than that, what's the plan for Booster firmware? now that most things work, I'd really like to try to finish this project off once and for all.

wizath, 2019-04-03:
I'm using float32 for calibration values

wizath, 2019-04-04:
Oh, snap. I only casted uint16 calculations to float32. Fixed

hartytp, 2019-04-04:
@wizath glad you tracked down the issue. AFAICT even though the calibration constants are stored as float32 internally, the VCP interface only seems to accept ints.

When you've fixed this, can you re-check the interlock accuracy and close this issue?

wizath, 2019-04-04:
Which command on VCP only accept ints?

hartytp, 2019-04-04:
Unless I'm misremembering the interlock calibration command only accepts ints (or, at least, it converts floats to ints internally).

hartytp, 2019-04-04:
Also applies to the power and bias current calibrations I think, but it's been some time so I might be wrong about that.

hartytp, 2019-04-08:
@wizath ping. What's the status on booster fw?

wizath, 2019-04-09:
> @wizath ping. What's the status on booster fw?

* I identified non-float values and fixed them
* Don't have any Booster atm to fix https://github.com/sinara-hw/Booster/issues/186
* https://github.com/sinara-hw/Booster/issues/147 I missed that you want all in one diagnostic command, will add
* https://github.com/sinara-hw/Booster/issues/122 Pushed new feature a while ago
* https://github.com/sinara-hw/Booster/issues/52 There was an issue with LED's but I fixed that
* https://github.com/sinara-hw/Booster/issues/47 I'll test that when old Booster v1.0 arrives since nothing changed in fan control

hartytp, 2019-04-09:
Great, thanks!

> I identified non-float values and fixed them

Does that fix this issue? If so, can we close?

hartytp, 2019-04-09:
>Don't have any Booster atm to fix #186

:( when are you due to get another Booster?

hartytp, 2019-04-09:
I can probably post you one of our v1.2 Boosters with some dead channels if that helps?

hartytp, 2019-04-09:
What about #120 #121 #180?

If you can't do any more development for now, please could you finish the documentation issues #178, #71  and also finish writing up firmware build instructions and merge the firmware with the master branch?

wizath, 2019-04-09:
Ok, will add docs

hartytp, 2019-04-09:
Thanks, that's really helpful. Last time I tried to test the SCPI/VCP commands I had issues because the description on the wiki was wrong, so I didn't know how the commands were supposed to behave.

hartytp, 2019-04-10:
@wizath ran a test on the amplifier:
```
Testing channel 0 ...
Gain:
40.6 dB (at P_out = 20.0 dBm)
39.8 dB (at P_out = 30.0 dBm)
39.2 dB (at P_out = 35.0 dBm)
Internal power meter:
f=100 MHz: error 0.8 dB at P_out = 20.0 dBm
f=100 MHz: error 1.1 dB at P_out = 30.0 dBm
!!! Significant error
f=100 MHz: error 1.2 dB at P_out = 35.0 dBm
!!! Significant error
f=250 MHz: error 1.8 dB at P_out = 20.0 dBm
!!! Significant error
f=250 MHz: error 1.5 dB at P_out = 30.0 dBm
!!! Significant error
f=250 MHz: error 1.5 dB at P_out = 35.0 dBm
!!! Significant error
..........
..........
..........
..........
..........
Interlock set at 25.0 dBm (200 MHz) -> trips at 24.1 dBm (stddev/min/max = 0.05, 24.06, 24.19 dBm)
..........
..........
..........
..........
..........
Interlock set at 30.0 dBm (200 MHz) -> trips at 28.9 dBm (stddev/min/max = 0.02, 28.87, 28.94 dBm)
!!! Significant error
..........
..........
..........
..........
..........
Interlock set at 35.0 dBm (200 MHz) -> trips at 33.8 dBm (stddev/min/max = 0.05, 33.68, 33.82 dBm)
!!! Significant error
```

Looks like the interlock calibration is about 1dB off.

@gkasprow @wizath  Do you know what conditions Creotech used to calibrate the interlock? What powers did they calibrate at? Were they careful to normalise cable lengths and use decently calibrated instruments?

@wizath please can you update the section on the Wiki that covers interlock/power detector calibration so I can recalibrate our device? NB to include https://github.com/sinara-hw/Booster/issues/178

hartytp, 2019-04-10:
NB we should also make sure to specify in the docs which reference frequency is to be used. I'd like calibration to be as reproducible as possible!

hartytp, 2019-04-13:
Actually, I'm probably being a little unfair here. The mean error in the measurements I posted above is only approx 1dB. However, the accuracy of the 30dB attenuator I used is only around 1dB. So, this is error is actually close to the accuracy of my measurement.

Nonetheless I think it's really important that we agree on a calibration procedure (e.g. what power levels to use, what frequency, what level of accuracy to aim for, etc) and make sure that Creotech/TechnoSystem stick to the procedure so that the accuracy it produces is predictable and reproducible between units.

closed 2019-04-13 by hartytp