sinara-hw/Booster #174: OVC when interlock trips

hartytp, 2019-03-18:
@wizath currently when the interlock trips we often see an OVC error on the channel it tripped on.

This is currently the biggest issue we have with Booster. Would you mind looking at it as top priority?

Thanks!

wizath, 2019-03-18:
Wait, what? You see OVC error on channel where interlock was tripped? Or neighboring channels? Since we're close to resolving that interlock reset issue, I'll look for the cause of that problem next.

hartytp, 2019-03-18:
Yes, we see an OVC error on the channel where the interlock tripped (not on neighbouring channels).

hartytp, 2019-03-18:
Aah, I hadn't realised that when the interlock trips it cuts the power to the FET.

hartytp, 2019-03-18:
Ouch, this looks nasty. Sequence of events:
- channel running happily with no RF applied. Output of InAmp (measured on R45) is approx 240mV
- Repeat the following a few times: apply RF to trip the interlock; turn RF off; reset interlock; repeat
- Red LED comes on
- Output of InAmp is now 450mV

hartytp, 2019-03-18:
So, this may well the the cause of our mysterious FET death...

hartytp, 2019-03-18:
I don't think this is an issue with the InAmp, since they are specified to have a max output of 300mV when the differential signal is 0V. So, it does appear that we've caught some glitch that is turning the FET current consumption to max.

hartytp, 2019-03-18:
hmm out bias is -4.9V and V29V is 280mV which seems sensible.

hartytp, 2019-03-18:
@wizath what actually happens when the interlock trips. AFAICT before the interlock trips, Vbias is 1.5V after it trips it goes to 3.4V and when this fault condition occurs it goes to -4.9V

hartytp, 2019-03-18:
AAh, okay, I can also reproduce this by putting the Booster unit in and out of standby mode multiple times.

What I saw just now was this:
1. All channels green
2. Turn standby mode on/off about 10 times
3. Half the channels go red
4. turn standby mode on/off one more time
5. all channels red

So, I think this must be some sequencing thing with the 28V LDO or something like that...

hartytp, 2019-03-18:
So, I think the interlock now works correctly, but we do have a nasty power supply bug. Still, progress :)

wizath, 2019-03-18:
Which device? The v1.3 one?

hartytp, 2019-03-18:
Yes. Scary

wizath, 2019-03-18:
This could be related to sequencing, since it happens while toggling standby button. Will look closely tomorrow.

hartytp, 2019-03-18:
I think something like that. I haven't tracked it down yet, but I'd really appreciate it if you could have a look.

hartytp, 2019-03-18:
@wizath I think this is a false alarm (and this isn't related to the dead FETs).

Here is what I think is happening:
1. In normal operation the common-mode voltage at the InAmps is 30V. So when the current is off we are in "case 3" in the data sheet (section 7.4.2.3). This is quite inacurate, but not as bad as...
2. When the power supply is cut, the common-mode voltage is now below 3V3 (the InAmp supply) so we move into "case 2" which is the least accurate mode of operation for the InAmp. The output offset in this regime can be very large.

We used a DMM to verify that when we see 450mV at the InAmp output there is no voltage across the sense resistor. So, this is just a measurement error.

So, I think the fix here is easy: just make sure that you never sample the InAmp output when there is no common-mode voltage applied.

Other than that everythign checks out :)

hartytp, 2019-03-18:
If you can fix this issue then I think we'll be able to complete our testing.

hartytp, 2019-03-18:
@gkasprow can you add a note on the schematic that the inamp has large errors when the common-mode or differential mode voltage is too low?

hartytp, 2019-03-18:
@wizath also NB for the bias current calibration we need to be careful not to take readings when the current is too low, or there will also be large errors.

The rule for this (to be added to the schamatic) is: never sample the InAmp output when the differential voltage is less than 20mV (5mA for 90mOhm sense resistor) or when the common-mode voltage is less than 3V3.

hartytp, 2019-03-19:
@wizath hmm...actually, that could be a red herring. Thinking about this more, it's likely that the high voltage I measured on the InAmp output was a result of the OVC tripping (so that the power supply was cut) and not the cause of it.

Still, I guess this must be some sequencing issue (like the CPU sampling the current during the initial spike when the power supply tuns on and charges up capacitors). Are you okay to look at this as top priority?

wizath, 2019-03-19:
Ok, reproduced issue with rapid toggling standby button

wizath, 2019-03-19:
OK, I think I found the issue with standby button OVC errors.

Since there are some spikes in the enable sequence, ADC sometimes generates false-positives sampling that spike. I changed measurement type of ADC to include a small pause between each measurement.

Also in current OVC procedure there must be 64 events to trigger the OVC disable, but that counter is never cleared, so consistent one-time alerts from ADC will build up and eventually cause false OVC error. Will change that to clear counter after like 100ish tries without alert (that's like 1 second without any ADC alert status).

hartytp, 2019-03-19:
AAh, okay, that makes sense. When you're ready, can you post fw with a fix. Thanks!

wizath, 2019-03-19:
https://github.com/sinara-hw/Booster/issues/98#issuecomment-474302824

Posted, please check if this solves your issue

cjbe, 2019-03-19:
Thanks! I'll test that now.

In the mean time, can you have a look at https://github.com/sinara-hw/Booster/issues/177 please?

hartytp, 2019-03-19:
@wizath AFAICT this is fixed. To do a thorough test, I need https://github.com/sinara-hw/Booster/issues/161 and #177 to be fixed. Then I can set up an automated test using Urukul + Kasli to stress Booster. Can you make those top priority? Thanks!!

closed 2019-03-19 by hartytp