sinara-hw/Booster #177: Output interlock calibration does not work

cjbe, 2019-03-18:
@wizath the output power interlock calibration does not seem to work properly. I am trying to calibrate the low-power point following [your instructions](https://github.com/sinara-hw/Booster/issues/151#issuecomment-472017483)

The calibration always seems to return the same value of 79 even for output powers as different as 0 and 30 dBm:

For +0 dBm output power:
```
intcal 0 1 0.0
[intcal] Calibrating output interlock ch 0
[ointcal] trying value 0, status 1
[ointcal] done, interlock value = 0
[intcal] Calibration step = 100 completed = 100
[ointcal] trying value 83, status 1
[ointcal] done, interlock value = 83
[intcal] Calibration step = 10 completed = 83
[ointcal] trying value 79, status 1
[ointcal] done, interlock value = 79
```
For +30 dBm output power:
```
intcal 0 1 30.0
[intcal] Calibrating output interlock ch 0
[ointcal] trying value 0, status 1
[ointcal] done, interlock value = 0
[intcal] Calibration step = 100 completed = 100
[ointcal] trying value 83, status 1
[ointcal] done, interlock value = 83
[intcal] Calibration step = 10 completed = 83
[ointcal] trying value 79, status 1
[ointcal] done, interlock value = 79
```

cjbe, 2019-03-19:
@wizath for the high-power calibration power point you seem to be doing a large scan. It is not clear to me how precise this is. 

Assuming the interlock now works reliably, might it not be easier to use a bisection algorithm? I.e. sample at dac_low=0 -> interlock trips and dac_high=4095 ->no trip, then sampling at 1/2*(dac_low+dac_high), and if it trips here use this value for dac_low, or if it does not trip use this value for dac_high. Then you can just repeat this until dac_high-dac_low is small enough. 

Example trace for the current calibration:
```
intcal 0 2 35
[intcal] Calibrating output interlock ch 0
[ointcal] trying value 0, status 0
[ointcal] trying value 100, status 0
[ointcal] trying value 200, status 0
[ointcal] trying value 300, status 0
[ointcal] trying value 400, status 0
[ointcal] trying value 500, status 0
[ointcal] trying value 600, status 0
[ointcal] trying value 700, status 0
[ointcal] trying value 800, status 0
[ointcal] trying value 900, status 0
[ointcal] trying value 1000, status 0
[ointcal] trying value 1100, status 0
[ointcal] trying value 1200, status 0
[ointcal] trying value 1300, status 0
[ointcal] trying value 1400, status 0
[ointcal] trying value 1500, status 0
[ointcal] trying value 1600, status 0
[ointcal] trying value 1700, status 0
[ointcal] trying value 1800, status 0
[ointcal] trying value 1900, status 0
[ointcal] trying value 2000, status 0
[ointcal] trying value 2100, status 0
[ointcal] trying value 2200, status 0
[ointcal] trying value 2300, status 0
[ointcal] trying value 2400, status 0
[ointcal] trying value 2500, status 0
[ointcal] trying value 2600, status 0
[ointcal] trying value 2700, status 0
[ointcal] trying value 2800, status 0
[ointcal] trying value 2900, status 0
[ointcal] trying value 3000, status 0
[ointcal] trying value 3100, status 0
[ointcal] trying value 3200, status 0
[ointcal] trying value 3300, status 0
[ointcal] trying value 3400, status 1
[ointcal] done, interlock value = 3400
[intcal] Calibration step = 100 completed = 3400
[ointcal] trying value 2833, status 1
[ointcal] done, interlock value = 2833
[intcal] Calibration step = 10 completed = 2833
[ointcal] trying value 2698, status 1
[ointcal] done, interlock value = 2698
[intcal] Calibration step = 1 completed = 2698
[intcal] done, got 2-point cal val 2698 pwr 35
[intcal] done, got power factors a=74.00 b=79.00
```

wizath, 2019-03-19:
Yes, I would like to tweak these little things, but with endless hardware tweaks to do, they go down deeper in priority

hartytp, 2019-03-19:
Understood. Let's fix the OVC issue and then return to this.

wizath, 2019-03-19:
Yuh, that's related to resetting the interlock while power is being applied constantly and the limit is lowered down by a small number.

hartytp, 2019-03-19:
> Yuh, that's related to resetting the interlock while power is being applied constantly and the limit is lowered down by a small number.

I think this was an artefact of the comparator oscillations and other interlock bugs. I don't think the hysteresis is significant any more.

e.g. using the current FW:
- set input RF power to -17.5dBm: interlock never trips
- set input RF to -17.4dBm: interlock trips after a while (presumably it's right on the threshold so it only takes a small amount of noise to trip)
- set input RF to -17.3dBm: interlock always trips
- at -17.3 dBm I can't clear the interlock (retrips straight away)
- at -17.4 dBm sometimes I can trip it, sometimes I can't
- at -17.5 dBm I can clear it very reliably

So, there is now approx 0.1dBm of noise in the interlock threshold but no discernible hysteresis.

So, we should be able to use a much simpler calibration algorithm as @cjbe suggests.

wizath, 2019-03-19:
wait, first I need to change the hardware to match your fixes.

cjbe, 2019-03-19:
@wizath with the modded hardware and a careful calibration the interlock seems to behave very nicely - I see ~0.05 dB standard deviation on the trip point, and an accuracy of ~0.1 dBm.

E.g. 24 dBm set -> average trip point 24.02 dBm, 30 dBm -> 29.91 dBm, 36 dBm -> 35.94 dBm.
So it seems like once we have a good calibration routine all the interlock problems are solved.

hartytp, 2019-03-19:
Whooooo!

hartytp, 2019-03-19:
@wizath any luck getting the new calibration routine working?

cjbe, 2019-03-19:
@wizath this is now the only thing holding back our acceptance tests of the Booster - if you cannot get the calibration working ASAP, could you add a VCP command that allows us to set the offset and slope of the interlock calibration so we can set this manually and finish our tests?

cjbe, 2019-03-19:
If I understand correctly we should be able to set the a & b calibration parameters by using the VCP commands to write to the channel EEPROM...

wizath, 2019-03-19:
[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/2985068/eth-scpi.dfu.zip)

Command is `intparams <ch> <a> <b>`. Sorry, I've got my thesis defense tomorrow and I'm a little busy. Will fix that calibration tomorrow / thursday. Also this version got reenabled watchdog with feeding inside interlock task so it shouldn't crash VCP anymore.

cjbe, 2019-03-19:
Amazing - thanks!

hartytp, 2019-03-19:
> Sorry, I've got my thesis defense tomorrow and I'm a little busy. 

I didn't realise! Thanks so much for finding the time to help us out with this. Have fun with the defence.

cjbe, 2019-03-21:
Okay, very nearly there. What's the status of the interlock calibration? Did you have a chance to implement the new bisection algorithm?

wizath, 2019-03-22:
Fixed https://github.com/sinara-hw/Booster/issues/98#issuecomment-475637956

closed 2019-03-22 by wizath